raspberrypi/picamera2 #642: Is there a way to a function for each frame acquire[HOW-TO]

tf4ong, 2023-03-29:
Hi, 

I would like to perform strobing with the picamera. After the quiting of each frame, I would like to call a function that turns puts a gpio on high one odd frames, and an another gpio on on even frames. Is there an easy to do this with start video recording?

davidplowman, 2023-03-29:
Hi, the first thing that comes to mind is that receiving frames from the camera is not a real-time event. That is, although the camera outputs frames to memory at very well-defined intervals, after that everything depends on queues, how busy the system is, keeping up "on average" and so on. So even if you were to do this kind thing down in the C++ layer, you'd find the timings variable.

Furthermore, I'm guessing you're using one of the rolling shutter sensors? In this case it's very common to be exposing both the "current" and "next" frames at the same time, so there's no sense in which you can do something for an "odd" frame but not an "even" one.

Maybe it's possible to make something work if you have slow framerates but the exposure times are not too long? Maybe you could explain something more about what you're trying to achieve and that might help us to figure out if there's a way forward. Thanks!

tf4ong, 2023-03-29:
Right now, I am using the global shutter camera. So, I won't have an issue like with the rolling shutter camera.

So I want to strob light on object. On odd frames, I want to look at an object with blue light. On even frames, I want to look at an object under green light. I was hopping to control two LEDs with two gpios pins. So on receiving each frame, I would call function to on/off the gpio pins.

davidplowman, 2023-03-30:
So I believe the GS cam does have a trigger line which would allow you to turn a light on or off and then capture a frame, unfortunately I don't think this is supported in the driver at present. Probably better to ask on the Camera Forum for information about which sensors might have this kind of support as it's not really something I can tell you much about.

tf4ong, 2023-04-02:
Thanks, I will be closing this. Just to note, I tried the pins and vtr seems promising as it is outputting a high (1.78V) when the camera is started using libcamera