raspberrypi/picamera2 #643: [HOW-TO] Detect motion at low res and then switch Camera to high res for recording?

Dweller, 2023-03-29:
I was wondering if somebody could help me I'm an experience developer but not with picamera.

What I was wondering is it possible motion from taken an occasional low res 640x400 every second or so and then on motion detect switch to hi res 1080p for recording the stream.  To complicate things I've two cameras on a Waveshare Baseboard with a CM4 module and if motion is detected on either camera I'd like to start recording on both of the connected Raspberry pi V2.1 cameras.

Examples of switching the camera modes would be helpful..

In addition I would like to run this on a CM4 for allowing for two cameras running at the same time and motion from either would trigger both to start recording would be perfect.

davidplowman, 2023-03-29:
There are many ways to switch camera modes. In principle they're all the same, you have to create a the new mode that you want, stop the camera, configure it with the new mode and then start it again. Or you can simply call the `switch_mode` function that does this for you. [Here](https://github.com/raspberrypi/picamera2/blob/main/examples/switch_mode.py), for example.

In your case I wonder if you could configure the camera to output two streams. One would be your high resolution stream that you would record, the other would be a downscaled one where you can detect motion. That way you wouldn't need to change modes at all. You could start and stop the encoder when you want, and look for motion in the downscaled image whenever you want. One catch is that the downscaled images always come out in a YUV (rather than RGB) format, but it's not too hard to detect motion there either.

tvoverbeek, 2023-03-29:
See the capture_circular example in the examples directory, which does what @davidplowman describes above (2 streams with the lores stream used for motion detection. Motion detection only used the Y (luminance) part of the YUV image.
For 2 cameras you will need 2 separate Picamera2 sessions

Dweller, 2023-03-30:
Great suggestions and thanks the pointing me at the directions,  I'll have a play with mode switching with a single camera before I try with both from the same process.

Thanks again.

closed 2023-03-30 by Dweller