raspberrypi/picamera2 #644: [BUG] Horizontal banding artifacts

GuidoBartoli, 2023-03-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
I made a script to set up the camera, capture and save frames continuously with the following configuration: 
```
Buffer: 6
Queue: True
Resolution: 2304 x 1296
Flipping: "both"
Exposure mode: "short
Exposure metering: "center"
Histogram constraint: "normal"
Exposure compensation: 0
White balance: "auto"
Auto-focus: "manual"
Lens position: 0
```
I am using this custom exposure mode to ensure that the maximum shutter time is 1/400s:
```
{
    "shutter": [ 100, 600, 1000, 1500, 2000, 2500 ],
    "gain": [ 1.0, 1.0, 2.0, 2.0, 3.0, 3.0 ]
}
```

Since I needed associated metadata, frames are captured using requests inside the loop:
```
request = self.capture_request()
frame, metadata = request.make_array("main"), request.get_metadata()
request.release()
``` 

**Expected behaviour**
The acquired images should be free of visual artifacts.

**Console Output, Screenshots**
Most of the frames have the artifacts that you can see in the following images (the camera is pointed at the floor and frames tiles).
[image]
[image]
[image]

**Hardware :**
*Board*: Raspberry Pi 4 (4 GB)
*Camera*: Camera Module 3 Wide NoIR
*OS*: Raspberry Pi OS Lite (Bullseye)
*Python*: v3.9.2

davidplowman, 2023-03-29:
Hi, it would be good to try and find out a bit more about where this is coming from. I notice slightly dark halos around the artifacts which somewhat suggests it's in the raw sensor data that is being fed to the ISP. Are you able to capture a DNG file where the JPEG has the problem, and we can check if it's in there too?

Also just wanted to check how long your ribbon cable is. Though bad connections often lead to a complete loss of sync and images go magenta, which would suggest that's not the problem. Have you noticed anything that correlates at all with the problem, such as the exposure times, or anything else?

GuidoBartoli, 2023-03-30:
Thanks for answering, I share some additional findings about this issue.
- I had forgotten that this Raspberry Pi 4 is overclocked to 2.0 Ghz so I thought that was the cause, so I reset the ARM's default frequency, but the problem unfortunately remains
- I added a function to overlay metadata on the image, this is another one where artifacts are visible (_T = duration time, S = shutter time, A = analog gain, D = digital gain, L = measured lux_):
[image]
- I captured also the RAW image (you can download it from [here](https://www.dropbox.com/s/5tcwxx92ca1uube/raw.dng?dl=0)) with the same parameters adding `request.save_dng("raw.dng")` into my function and converted to 24bit with RawTherapee, but the artifacts are there too:
[image]
- Here is the really very strange thing: I tried capturing a single image with the same setup using the command-line utility (`libcamera-jpeg -o foo.jpg --width 2304 --height 1296 --rotation 180 --shutter 2500 --analoggain 3 --autofocus-mode manual --lens-position 0 -t 1000`). I realize that this is a single capture unlike the other one which is in a capture cycle with video configuration, however I expected it to come out similar since the parameters are the same and instead it came out much darker, but without artifacts as you can see below:
[image]
- This is a picture of the Raspberry inside its case with the CSI connector coming to the Camera module mounted in front of it:
[image]

GuidoBartoli, 2023-03-30:
The power supply is not a switching one, because the Raspberry is powered by [this Powerbank](https://it.rs-online.com/web/p/power-bank/2361455) which outputs 2.1A, however now I am testing with the official power supply connected to the wall, I will post the result here as soon as possible.

davidplowman, 2023-03-30:
Hi, thanks for the raw file and information. It's interesting to note that the raw data shows the same problem so it's almost certainly some kind of transmission issue between the sensor and the Pi. I'd certainly be double-checking the cables, maybe try swapping them out, or looking for interference. I suppose a broken camera is possible though it feels less likely to me. As you say, double checking the power supply too sounds like a good move. Sorry not have anything more concrete to suggest.

GuidoBartoli, 2023-05-09:
Thanks for coming back on this.

Unfortunately the issue is still present even with the official Raspberry Pi 4 power supply, so I think it is still related to the short custom exposures I explained above, because the more I increase the exposures, the less the number of artifact lines present in the image.

davidplowman, 2023-05-09:
Have you been able to try another pi/cable/camera  (depending on whether any of these are available to you)?

ryangrg, 2023-07-19:
I had this problem using a raspberry pi 4B and an arduocam HQ with switching IR filter. I unplugged and carefully plugged both ends of the connector ribbon back in and problem went away.

GuidoBartoli, 2024-04-27:
> @GuidoBartoli what you seeing here is interference on signal, some information is here [raspberrypi/libcamera#43](https://github.com/raspberrypi/libcamera/issues/43)

Ok, thanks for the information!

closed 2024-04-27 by GuidoBartoli