raspberrypi/picamera2 #645: Setting non-standard shape for "lores" not giving expected width [BUG]

Simzie, 2023-03-31:
I'm capturing a video stream for later assessment and inference, so have set the output to be square, which is working great for the "main" stream (might be slightly different as seen below but at least square), however the "lores" stream which I'm assessing and displaying is coming out with a non-expected shape, specifically a higher width than anticipated. Is this to be expected and I just have to deal with it or is this a bug? I'm using 64bit lite OS.

**To Reproduce**
`import cv2
from picamera2 import Picamera2

# Camera Settings
main_size = 789
lores_size = 456

picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={'size': (main_size, main_size), 'format': 'RGB888'}, lores={'size': (lores_size,lores_size)}) 
picam2.configure(video_config)
picam2.start()

main_img = picam2.capture_array('main')
main_shape = main_img.shape
print(f'Main shape is {main_shape}')

lores_img = picam2.capture_array('lores')
lores_img = cv2.cvtColor(lores_img, cv2.COLOR_YUV420p2BGR)
lores_shape = lores_img.shape
print(f'Lores shape is {lores_shape}')

picam2.stop()
picam2.close()`

**Expected behaviour**
For the "lores" output to match the size set

**Console Output, Screenshots**
`[28:06:09.408565728] [12477]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4224-42eae600
[28:06:09.506968020] [12481]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx519@1a to Unicam device /dev/media3 and ISP device /dev/media0
[28:06:09.524938957] [12477]  INFO Camera camera.cpp:1028 configuring streams: (0) 788x788-RGB888 (1) 456x456-YUV420
[28:06:09.525731145] [12481]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected sensor format: 1280x720-SRGGB10_1X10 - Selected unicam format: 1280x720-pRAA
Main shape is (788, 788, 3)
Lores shape is (456, 512, 3)
`

**Hardware :**
Raspberry Pi Zero 2 W

**Additional context**
aarch64
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian

davidplowman, 2023-03-31:
Hi, thanks for the question. I think all that's happening is that you're getting tripped up by alignment problems.

The catch is that the image processing hardware can't write out rows of image data that are only 456 bytes wide. It can only start at 64-byte boundaries, so 456 gets rounded up to 512 by adding 56 bytes of padding to the end of each row.

Now, because YUV420 has a different line width in the Y and UV planes, it's rather expensive remove that padding if users want to be able to call cvtColor to turn it into RGB. Instead it actually makes more sense for users to remove the padding themselves at the end. So it should be sufficient if you add something like this after the conversion to RGB:
```
lores_img = lores_img[:, :456, :]
```
The extra bit of the image that you're discarding should be empty, which you should be able to verify by displaying it. Hope that helps!

Simzie, 2023-03-31:
Thanks @davidplowman, I was pretty sure there was a reasonable answer but thought it best to ask. That would explain why it was displaying as a solid green band on the screen.  Thanks for the tip!

Pepslee, 2024-12-20:
I have the same problem with width=704.   From the doc, YUV420 has optimal alignment = 64
704/64 = 11 ,  so 704 must be optimal, but picamera2 gives me  768 (704 with padded 64 pixels)

Pepslee, 2025-01-02:
Hi @davidplowman , I use cm5 and [picamera-manual guide ](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). 
But 704 is also 64  aligned, is not it ?
I use YUV420 format

davidplowman, 2025-01-02:
Indeed, and I think the manual might want updating. I think Picamera2 just follows what the kernel driver tells it, so it may be that on a Pi 5 the Y alignment needs to be 128. But I'll check that and submit a doc update if it needs changing. Thanks for pointing this out!

In case it wasn't clear, note that after you've configured the camera, `picam2.camera_configuration()` should always report the correct stride, whatever the platform.