raspberrypi/picamera2 #648: [HOW-TO] record video in h264 and stream video in mjpeg

SureshkumarSenthilkumar, 2023-04-04:
I'm currently utilising a Raspberry Pi camera 2 to download and stream videos. I was successful doing so using the mjpeg and jpeg formats, but when I changed the encoder to h264, the video was not streaming. Could you possibly help me resolve this?

davidplowman, 2023-04-04:
Hi, we're going to be posting an update shortly which makes it much easier to record multiple streams. In the meantime you still have to encode the second stream "by hand" (as shown [here](https://github.com/raspberrypi/picamera2/blob/main/examples/dual_encode.py)).

If you are able to create a small self-contained test case that demonstrates the problem you are having, then we may be able to help, but otherwise it's rather difficult to know what you've been trying to do. Are you able to provide some more information? Thanks.

SureshkumarSenthilkumar, 2023-04-04:
I want to record and stream high-quality video, and I already have a working JPEG and MJPEG encoder code for both encoding and streaming. However, recording video in those encoders takes up a lot of space, so I tried using an H264 encoder instead. However, when I tried using an H264 encoder to stream and record, streaming did not work, and I've read that h264 is not recommended for streaming. Is it feasible to record in H264 and stream in mjpeg or jpeg at the same time?

davidplowman, 2023-04-04:
I'm still not clear what "streaming did not work" means. Are you saying that the software produced an error, or that the software ran but video did not get encoded, or that the video encoding all works fine but you don't have a client that can play the video? Or are you trying to stream the mjpeg and record h.264 locally but saving the local file fails? What are you streaming to - a video player, a web page, something else? Are you streaming any particular format, and over what kind of socket?

I'm afraid there's not really enough detail here for me to know what to suggest. There should be no problem saving h.264 files, though streaming it can be more difficult because you have to find a client that can play it. If you could provide a simple self-contained example, that might help. Thank you.

SureshkumarSenthilkumar, 2023-04-04:
Can't we access the streamed h264 encoder output in chrome or firefox browser?

davidplowman, 2023-04-04:
Web browsers don't natively support h.264 streams. In fact, I've always found streaming anything other than MJPEG to a web page to be surprisingly difficult. There are a number of third party frameworks and libraries that can help you to implement an h.264 client, but it's not really something I know about. Sorry not to be more help.

SureshkumarSenthilkumar, 2023-04-04:
[image]

SureshkumarSenthilkumar, 2023-04-04:
[image]

closed 2023-04-04 by SureshkumarSenthilkumar