raspberrypi/picamera2 #652: [HOW-TO] Tune brightness and contrast?

marghelo94, 2023-04-06:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the request here.

chrisruk, 2023-04-06:
Do you mean you feel the image you've captured is under or over exposed?

You could try manually setting the exposure time, such as in -
https://github.com/raspberrypi/picamera2/blob/main/examples/exposure_fixed.py

marghelo94, 2023-04-06:
Thank you very much! This example was helpfull.
Is there a way to auto-regulate the exposure time according to the particular context? I am interested in taking videos...