raspberrypi/picamera2 #654: [OTHER] Performance are very poor on Qt Application

bud-mo, 2023-04-07:
Hello, I am building an application (hopefully will be an ehnanced photoboot, another one) with this setup:
- RPi 4 B (8GB)
- Raspberry Pi Touch Display
- Raspberry Pi Camera Module 3 (Tested also with Camera Module 2 with the same results)
- Waveshare SIM7600E-H 4G HAT
- HDMI Display (1600x1900, in "Production" will be used a 1366x768 monitor and the windows are both fullscreen on the two monitors)

While project is growing (with a slideshow of pictures), the performances of the `QGlPicamera2` are very poor.
It appears very laggy and drops a lot of frames (I will attach a video as soon as possible).

I tried to check on (almost all) examples and reduced the size, but without any improvement, maybe I am missing something.

Here there is demo of the project, where I removed firebase and modem (managed with QSerial) connections and reduced to not attach the SVG or other resources like fonts. Secondary screen and modem are not required for the demo.

```python
from libcamera import controls
from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2
from PyQt5.QtCore import pyqtSignal, QObject, QSize, Qt
from PyQt5.QtGui import QFont
from PyQt5.QtWidgets import QApplication, QDesktopWidget, QHBoxLayout, QLabel, QMainWindow, QVBoxLayout, QWidget

import sys

class CameraWidget(QWidget):
    def __init__(self):
        super().__init__()

        layout = QVBoxLayout()
        layout.setContentsMargins(0, 0, 0, 0)
        layout.setSpacing(0)
        
        self.picam2 = Picamera2()
        # Adjust the preview size to match the sensor aspect ratio.
        preview_width = 800
        preview_height = self.picam2.sensor_resolution[1] * 800 // self.picam2.sensor_resolution[0]
        preview_height -= preview_height % 2
        preview_size = (preview_width, preview_height)
        # We also want a full FoV raw mode, this gives us the 2x2 binned mode.
        raw_size = tuple([v // 2 for v in self.picam2.camera_properties['PixelArraySize']])
        self.preview_config = self.picam2.create_preview_configuration(main={"size": preview_size})
        self.capture_config = self.picam2.create_still_configuration()
        if 'AfMode' in self.picam2.camera_controls:
            self.picam2.set_controls({"AfMode": controls.AfModeEnum.Auto})
        else:
            print("Attached camera does not support autofocus")
        
        self.picam2.configure(self.preview_config)
        self.picam2.start()

        cameraView = QGlPicamera2(
            self.picam2,
            width=preview_size[0],
            height=preview_size[1],
            keep_ar=False
        )

        layout.addWidget(cameraView)

        self.setLayout(layout)

class SlideshowWindowSignals(QObject):
    closed = pyqtSignal()

class SlideshowWindowView(QWidget):
    def __init__(self):
        super().__init__()
        layout = QVBoxLayout()
        layout.setContentsMargins(0, 0, 0, 0)
        layout.setSpacing(0)

        # layout.addWidget(CoverWidget())
        layout.addWidget(CameraWidget())

        self.setLayout(layout)

class SlideshowWindow(QMainWindow):
    def __init__(self):
        super().__init__()
        # Setup the window
        self.setWindowTitle("Slideshow Window")
        palette = self.palette()
        palette.setColor(self.backgroundRole(), Qt.black)
        self.setPalette(palette)
        self.setAutoFillBackground(True)
        desktop = QDesktopWidget()
        screen = desktop.screen(0)
        screen_geometry = screen.geometry()
        if screen_geometry.width() > 1366 or screen_geometry.height() > 768:
            self.setFixedSize(QSize(1366, 768))
        else:
            self.setGeometry(screen_geometry)
            self.showFullScreen()
        # Setup signals
        self.signals = SlideshowWindowSignals()
        # Setup the view
        self.setCentralWidget(SlideshowWindowView())
    
    def keyPressEvent(self, e):
        if e.key() == Qt.Key_F5:
            self.close()
    
    def close(self):
        super().close()
        self.signals.closed.emit()

class NotificationsArea(QWidget):
    def __init__(self):
        super().__init__()
        # Pick the default font for the icons
        font = QFont("tabler-icons", 30)
        # Build layout
        layout = QHBoxLayout()
        layout.setContentsMargins(0, 0, 0, 0)
        layout.setSpacing(8)
        # Configuration Icon
        self.iconConfig = QLabel("A", self)
        self.iconConfig.setFont(font)
        self.iconConfig.setAlignment(Qt.AlignCenter)
        p2 = self.iconConfig.palette()
        p2.setColor(self.iconConfig.foregroundRole(), Qt.white)
        self.iconConfig.setPalette(p2)
        layout.addWidget(self.iconConfig)
        # Sync Icon
        self.iconSync = QLabel("B", self)
        self.iconSync.setFont(font)
        self.iconSync.setAlignment(Qt.AlignCenter)
        p2 = self.iconSync.palette()
        p2.setColor(self.iconSync.foregroundRole(), Qt.white)
        self.iconSync.setPalette(p2)
        self.iconSync.hide()
        layout.addWidget(self.iconSync)
        # Signal Strenght Icon
        self.iconCellSignalQuality = QLabel("C", self)
        self.iconCellSignalQuality.setFont(font)
        self.iconCellSignalQuality.setAlignment(Qt.AlignCenter)
        p2 = self.iconCellSignalQuality.palette()
        p2.setColor(self.iconCellSignalQuality.foregroundRole(), Qt.white)
        self.iconCellSignalQuality.setPalette(p2)
        layout.addWidget(self.iconCellSignalQuality)
        # Set the element layout
        self.setLayout(layout)

class Toolbar(QWidget):
    def __init__(self):
        super().__init__()
        # Build layout
        layout = QHBoxLayout()
        # Icon and title
        title = QLabel("My Application")
        palette = title.palette()
        palette.setColor(title.foregroundRole(), Qt.white)
        title.setPalette(palette)
        font = QFont("Work Sans Regular", 30)
        title.setFont(font)
        layout.addWidget(title)
        # Add a filler to split the widgets
        layout.addStretch()
        # Notifications area
        layout.addWidget(
            NotificationsArea()
        )
        # Set the element layout
        self.setLayout(layout)

class MainWindowView(QWidget):
    def __init__(self):
        super().__init__()

        # Initialize the view
        layout = QVBoxLayout()
        layout.setContentsMargins(0, 0, 0, 0)
        layout.setSpacing(4)
        toolbar = Toolbar()
        layout.addWidget(toolbar)

        layout.addStretch()

        self.setLayout(layout)

class MainWindow(QMainWindow):
    def __init__(self):
        super().__init__()
        # Initialize window
        self.setWindowTitle("Control Window")
        palette = self.palette()
        palette.setColor(self.backgroundRole(), Qt.black)
        self.setPalette(palette)
        self.setAutoFillBackground(True)
        desktop = QDesktopWidget()
        screen = desktop.screen(1)
        screen_geometry = screen.geometry()
        if screen_geometry.width() > 800 or screen_geometry.height() > 480:
            self.setFixedSize(QSize(800, 480))
        else:
            self.setGeometry(screen_geometry)
            self.showFullScreen()

        # Initialize Slideshow window
        self.slideshow = SlideshowWindow()
        self.slideshow.signals.closed.connect(self.close)

        # Set the main widget
        self.setCentralWidget(
            MainWindowView()
        )
    
    def keyPressEvent(self, e):
        if e.key() == Qt.Key_F5:
            self.close()
    
    def close(self):
        if self.slideshow.isVisible():
            self.slideshow.close()
        super().close()
    
    def show(self):
        super().show()
        self.slideshow.show()

if __name__ == "__main__":    
    app = QApplication(sys.argv)

    window = MainWindow()
    window.show()
    
    exitCode = app.exec_()

    sys.exit(exitCode)

```

This the output log:

```
[0:01:33.805355788] [1422]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:01:33.951072920] [1425]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media2
[0:01:33.963356333] [1422]  INFO Camera camera.cpp:1028 configuring streams: (0) 800x450-XBGR8888
[0:01:33.963809804] [1425]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[0:01:54.590277504] [1425] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:02:12.565979952] [1425] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:02:23.036807858] [1425] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:02:23.278268622] [1425] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
```

davidplowman, 2023-04-11:
Hi, I think the problem is that when you call `picam2.start()` it creates an event loop because there isn't one running, and then the QGlPicamera2 widget creates another one.  Simply move `picam2.start()` to just below the creation of the widget. I guess we could help there by detecting that something bad is happening and provoking an error, so we'll look into that.

To get a smoother framerate, maybe consider pegging the max framerate to, say, 30fps, and possibly increase the buffer_count slightly too, depending on how it seems to run.

bud-mo, 2023-04-23:
I refactored the code a little bit to move the start() after the widget initialization.
Now the frame rate is good for the preview.

Errors appears half of the time.

Many time appears during the call switch_mode_and_capture_array(), it says “camera not started properly” and the app crashes.

davidplowman, 2023-04-24:
Obviously including an error message or the console output would be helpful here.

If I had to take a guess, then maybe you're running out of CMA memory? You can alleviate this by finding the line `dtoverlay=vc4-kms-v3d` and replacing it by `dtoverlay=vc4-kms-v3d,cma-384` or even `dtoverlay=vc4-kms-v3d,cma-512` (do not add any extra spaces or capitalisation). The reboot. You can check how much CMA you have with `grep Cma /proc/meminfo`.

bud-mo, 2023-04-24:
I checked the CMA Memory and it is already set to 512 (more than 300 are free)
```
pi@raspberrypi:~ $ grep Cma /proc/meminfo
CmaTotal:         524288 kB
CmaFree:          350268 kB
```

The cycle is controlled by timers and act this way:
- Normally the finder is not shown, the `QGlPicamera2`
- Once the user hit a button a `QTimer` starts the countdown
- While the timer runs the related label is updated
- At Countdown 0 an AF Cycle is performed (if supported by the camera)
- The picture is took with `switch_mode_and_capture_array`

In the attached log, it happens with the second picture, the first works great.

This is the error I get into the console:
```
# The finder is not shown
[0:07:06.940145902] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:07:08.060265816] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:07:12.604154357] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:07:14.524267140] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:07:15.580132619] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:07:15.774217508] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
[0:07:19.676165010] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
# Finder is shown and the countdown start (triggered by a QTimer()
Countdown: 7
Countdown: 6
Countdown: 5
Countdown: 4
Countdown: 3
[0:07:28.252144241] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
Countdown: 2
Countdown: 1
[0:07:29.948140924] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
Countdown: 0
[0:07:31.324132512] [2150] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output
AF cycle succeeded in 10 frames
[0:07:31.683598010] [2133]  INFO Camera camera.cpp:1028 configuring streams: (0) 4608x2592-BGR888
[0:07:31.688701403] [2150]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
[0:07:31.927757035] [2150] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video1[20:cap]: Failed to start streaming: Errore di input/output
Camera did not start properly.
Traceback (most recent call last):
  File "/home/pi/wedding-buddy-photobootd/slideshow_window/finder_widget.py", line 64, in onCameraDone
    image = self.picam2.wait(job)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1152, in wait
    return job.get_result()
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 77, in get_result
    return self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 433, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 47, in execute
    done, result = self._functions[0]()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1212, in switch_mode_
    self.start_()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1003, in start_
    raise RuntimeError("Camera did not start properly.")
RuntimeError: Camera did not start properly.
Annullato
```

bud-mo, 2023-04-24:
This is the complete log of te first picture I try to take:

```
[0:23:28.083819063] [2596]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:23:28.219569727] [2610]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[0:23:28.257486882] [2596]  INFO Camera camera.cpp:1028 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[0:23:28.264959107] [2610]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
Multithreading with maximum 4 threads
[Modem thing omitted]
Countdown: 7
Countdown: 6
Countdown: 5
Countdown: 4
Countdown: 3
Countdown: 2
Countdown: 1
Countdown: 0
AF cycle succeeded in 18 frames
[0:23:54.653770794] [2596]  INFO Camera camera.cpp:1028 configuring streams: (0) 4608x2592-BGR888
[0:23:54.658432473] [2610]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
[0:23:55.278493062] [2596]  INFO Camera camera.cpp:1028 configuring streams: (0) 800x450-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[0:23:55.290454048] [2610]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[0:23:55.398715031] [2610] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video1[20:cap]: Failed to start streaming: Errore di input/output
Camera did not start properly.
Traceback (most recent call last):
  File "/home/pi/wedding-buddy-photobootd/slideshow_window/finder_widget.py", line 64, in onCameraDone
    image = self.picam2.wait(job)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1152, in wait
    return job.get_result()
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 77, in get_result
    return self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 433, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 47, in execute
    done, result = self._functions[0]()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1372, in capture_array_and_switch_back_
    self.switch_mode_(preview_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1212, in switch_mode_
    self.start_()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1003, in start_
    raise RuntimeError("Camera did not start properly.")
RuntimeError: Camera did not start properly.
Annullato
```

davidplowman, 2023-04-24:
Hi again, well that all looks very bad. Do you have the touch screen connected? It sounds like there are problems communicating with the camera, which can sometimes be caused by other devices.

bud-mo, 2023-04-24:
Yes, the 7inch touchscreen from raspberry.
Also a 4G hat and an hdmi monitor in hdmi0.

bud-mo, 2023-04-24:
Just tested without the 7in touchscreen display.
It seems to work great 5/5 pictures taken without any error, also no `ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Errore di input/output` was shown.

I there is anythng I can do to help, it will be a pleasure for me to help this project!

bud-mo, 2023-04-24:
It is using `vc4-kms-v3d`. I will check on the forum, thanks.

bud-mo, 2023-05-02:
I think I can close because the errors are a separate issue.

closed 2023-05-02 by bud-mo