raspberrypi/picamera2 #655: [OTHER] Problems video reading using OpenCV and picam2

marghelo94, 2023-04-07:
I am using a HQ camera and a raspberry pi 4. 
My code acquires 10s-video using picamera2.start_recording and picamera.stop_recording, with the suggested video_configuration and H264 encoder. Then, I convert the video from .h264 to .mp4 format.
So far so good. The problem starts when I call OpenCV videoCapture to read this mp4 video. The cap.read( ) function freezes at the first frame. 
I think there are any interference with the picam2 library, because the same code worked with picam. The acquired mp4 video is not corrupted, and if I use cap.read() on the same video but in other script it works.
So, the problem seems to be in the subsequent use of picam2 and opencv.
Any suggestion?

Thank you very much

davidplowman, 2023-04-11:
Could you include a minimal code snippet that shows how you are expecting to read frames from the file, and which is not working? Thanks!

davidplowman, 2023-04-17:
Hi, thank you for posting the code. Unfortunately I don't seem to be able to run that example so there's not too much that I can suggest.

I think I would take the `video_h264` file and see if that looks correct. For example, can you play it with `ffplay`? If that seems to play OK then the problem may be with this `convert` function? Obviously I'm guessing here, sorry that I can't be more helpful.

marghelo94, 2023-04-17:
``Hi, thank you for your reply.
I can play video_h264 and its conversion video_mp4 with VLC media player and they are both ok.
I don't understand why the problem in opening videos with OpenCV happens only if I use picamera2 in the same script. If I record a video X in a script and then I read this video X in another script, there is no problem.

I don't think the convert function is responsible of this, I will share below.

```
`def convert(file_h264, file_mp4):
    command = "MP4Box -add " + file_h264 + " " + file_mp4
    call([command], shell= True)
    print('Video converted!')`
```

Thank you for the help

davidplowman, 2023-04-17:
Hard to know what's wrong. It feels like your h264 stream is probably OK and your mp4 file may be fine too, though mp4 is a fairly complex file format so there's certainly scope for compatibility issues. Maybe you could try a different converter, such as ffmpeg? Something like `ffmpeg -i input.h264 -c:v copy output.mp4` might be close to what you want.

marghelo94, 2023-04-17:
Thank you for your suggestion.
Maybe I've just found the source of error.
If I comment `picam2.start_preview(preview=True) ` before `picam2.start_recording(encoder,videopath + video_h264)` and `picam2.stop_preview()` immediately after, I can read the recorded video with OpenCV without freezing.
It is not clear to me why the preview function makes this effect... Perhaps it isn't the most convenient way to have a preview of video recording...
Anyway, the main problem seems to be solved.

Thanks a lot for your help!

davidplowman, 2023-05-09:
I'll close this issue for the time being then. Please open up another if you have further questions. Thanks!

closed 2023-05-09 by davidplowman