raspberrypi/picamera2 #657: [BUG]

FedricoCiccio, 2023-04-09:
Camera controls,  shadows metering not working
command used:
button_swmeter.when_pressed = lambda: (picam2.set_controls({"AeConstraintMode": controls.AeConstraintModeEnum.Shadows}),print("shadow"))

error given
ERROR IPARPI raspberrypi.cpp:905 Constraint mode 2 not recognised

Hardware: Camera Module V3 on Raspberry Pi Zero

davidplowman, 2023-04-11:
Thanks for spotting this. I'll fix that and submit the change to upstream libcamera. Obviously this will take a little while to filter through, but in the meantime you could use the mode named "custom" as a workaround. You'd have to add this to the tuning file, though I note that the imx708 tuning does not have the "shadows" mode either, so you'd have to add it there too.

FedricoCiccio, 2023-04-11:
Thank you, let me just say that i really apreciate the work that you all do,
I guess I'll have to change the tuning file for the imx708 anyways (the color shading correction in the standard tuning is, as the kids say, very spicy). I guess i will have to work at that first.

DFoxinator, 2023-10-13:
Hi - does anyone know if there are any updates to this issue? Thanks!

davidplowman, 2023-10-16:
Hi, your request is for a "shadow" AE/AGC mode? Can you also say what camera you're using. As I explained above, you can certainly define such a mode for yourself in the tuning file - please ask if you need help.

I'd certainly like to add something like this, though for the moment such changes are rather stuck behind all the Pi 5 work in the queue! But please bear with us... thanks.

DFoxinator, 2023-10-29:
Hi - thanks so much for the response. I'm using Raspberry Pi Camera Module 3. Yeah, I was interested in trying out a "shadow" AE mode. I was a bit confused about the tuning file/how to set that up - is there any good documentation on that?