raspberrypi/picamera2 #658: [HOW-TO] Convert SRGGB10 into RGB frame?

saswata-paladhi, 2023-04-10:
Guys. I want to perform object detection on the raw feed from the picamera2. Its like the main feed will be zoomed and the detection algo will be run on raw feed. So when the camera zooms on an object and when the object moves out of the zooming frame, the camera frame will follow it as the detection will be ran on the zoomed out frame. The main feed changes with set_control but the raw doesn't changes. The object detection is not running well on the raw feed as its not post processed. 

How can I post process it so that so that the raw frame can be used for object detection?

davidplowman, 2023-04-11:
There's quite a lot that has to happen to a raw frame to turn it into regular RGB - after all, there's a ton of image processing hardware that is doing this for you.

Are you able to perform detection on a greyscale image? That would make things easier as you could merely pick G pixels and use those. If you need RGB, you could pick out the R and B pixels, and one of the greens (or average them). But you still have to deal with black level (image metadata will tell you what to subtract), and white balance (again, image metadata will tell you the gains you need to apply). You'll have to apply a gamma transform too. Even this won't apply any colour or lens shading correction, though on some sensors (HQ cam for example) it's not much of a problem.

I wonder slightly if it might be easier to process the whole camera image, and then use something like OpenCV to do the zoom for your "main feed" afterwards? I guess it needs some experimentation!

saswata-paladhi, 2023-04-12:
Hi. So can we take multiple picamera2 feeds from a single instance and like perform zoom on one feed and not on another?

davidplowman, 2023-04-12:
I'm afraid not. Both the "main" and "lores" images are constrained to have the same field of view.

saswata-paladhi, 2023-04-12:
Oh. So regarding that opencv zooming, is there any other way to perform digital zoom? Other than that slicing frames?

davidplowman, 2023-04-12:
I guess you can use cv2.resize, I'm not sure if there's anything better out there. Quite a lot probably depends on how you are going to render the images, whether you're using the built-in previews, or cv2.imshow, or something else.