raspberrypi/picamera2 #660: Remote Desktop Connection - Preview window unavailable

Bra1nsen, 2023-04-12:
```
No protocol specified
Preview window unavailable

```

any idea how to solve that?

Bra1nsen, 2023-04-12:
[image]
what is going on -.-

davidplowman, 2023-04-12:
What's in your DISPLAY environment variable?

tvoverbeek, 2023-04-12:
From your log:
qt.qpa.xcb: could not connect to display :10.0

It seems you are connecting via SSH with X-Forwarding.
To display the preview window you need to be running an X-server on your local machine (not on the machine with the camera and libcamera).
Can you run a simple X based program on SkyCamDLR (i.e. lxterminal) to check if X-forwarding is working?

Bra1nsen, 2023-04-13:
thanks for your help: 

[image]

Bra1nsen, 2023-04-13:
anythingelse I could try?

Bra1nsen, 2023-04-17:
```
#!/usr/bin/python3

# Capture a JPEG while still running in the preview mode. When you
# capture to a file, the return value is the metadata for that image.

import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()

preview_config = picam2.create_preview_configuration(main={"size": (800, 600)})
picam2.configure(preview_config)

picam2.start_preview(Preview.QTGL)

picam2.start()
time.sleep(2)

metadata = picam2.capture_file("test.jpg")
print(metadata)

picam2.close()
```

```
dlr@SkyCamDLR:~/Capturing $ sudo python cap.py
[sudo] password for dlr: 
[0:35:19.336544939] [2865]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:35:19.375896495] [2866]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:35:19.381172754] [2865]  INFO Camera camera.cpp:1028 configuring streams: (0) 800x600-XBGR8888
[0:35:19.381528069] [2866]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
No protocol specified
qt.qpa.xcb: could not connect to display :10.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted
```

```
dlr@SkyCamDLR:~ $ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyQt5
>>> 
>>> print(PyQt5.__path__)
['/usr/lib/python3/dist-packages/PyQt5']
```

What the hell is going on?

tvoverbeek, 2023-04-17:
Why are you running your script as root?
Probably a problem with xauth (X-Authorization). root has to have the same xauth to access your display as your normal user (dlr, I assume).

Bra1nsen, 2023-04-17:
I use the Pi headless, but I connect via remote desktop connection to watch preview. there is a stupid issue with remote desktop (xrdp) and "pi" because of the groups "render" and "video" that is somehow auto login and cant screen twice or something like that ([only shows a blue screen after connecting](https://forums.raspberrypi.com/viewtopic.php?t=323471&hilit=xrdp&sid=4b40dd5ac0f90de62c6055be41c841af&start=25)) anyway if I dont use sudo I end up with permission denied.

tvoverbeek, 2023-04-17:
Do you have "Raspberry Pi OS light" installed on SkyCamDLR (since you are running headless)?
In that case there is no X support and X-forwarding will not work (qt.qpa.xcb: could not connect to display :10.0).
I am still not clear how your setup works.
I assume SkyCamDLR is a remote RPi and you are logged in with ssh from your local computer (also a RPi or something else)

Bra1nsen, 2023-04-17:
Would I be able to show you this screenshot, if I installed light version? 

> thanks for your help:
> 
> [image]

Bra1nsen, 2023-04-18:
to get a preview window i have to add my user to:

`sudo usermod -a -G video, render <your_username>`

since this blocks the remote desktop access, its not possible. wtf

closed 2023-04-18 by Bra1nsen

somsubhro88, 2023-08-10:
Hello @Bra1nsen,

How did you fix this? I seem to be getting the same issue.

Bra1nsen, 2023-08-10:
yes iam aware of this problem, i use currently remote desktop connection with an added user. but you cant open any preview windows since you need to turn off ` sudo usermod -a -G video, render <your_username>` to make the remote connection work.., hope that get fixed in future