raspberrypi/picamera2 #663: [HOW-TO]How to synchronize data to a USB drive immediately after capture and recording？

yingliuzhizhuzed, 2023-04-13:
I am using 4B and module3 cameras to take photos and record videos, and I am saving the data on a USB flash drive. Sometimes I forget to eject the USB drive and just unplug it, and then the data is lost. I am a newcomer to Linux, and I believe the reason for data loss is that the system has not yet had time to write data from the cache to the USB drive, right? If that's the case, can I immediately perform cache synchronization after each capture and recording?

davidplowman, 2023-04-13:
It depends a bit how your software works, but maybe you could call the `sync` command from your code every time you finish saving a file?

yingliuzhizhuzed, 2023-04-13:
Thank you, as you said, I  call 'sync' every time I save a file“ now.

davidplowman, 2023-04-14:
I'll close this one for now, then. Please open up another issue if anything else comes up. Thank you!

closed 2023-04-14 by davidplowman