raspberrypi/picamera2 #666: [HOW-TO] Blocking versus non-blocking?

carterw, 2023-04-14:
I'm reading through the Picamera2 Library Manual. On page 56 there's a list of functions that can be called 'directly' and those that can be called in a 'non-blocking manner'. I don't understand what the ramifications of this are in terms of what to do in the code. 

In the list, picam2.start() is 'Safe to call directly" and picam2.capture_array() is 'Call in a non-blocking manner'. In the examples I see several instances of code where they are simply called one after the other with no particular special treatment. Can someone please explain?

`picam2 = Picamera2()
config = picam2.create_still_configuration()
picam2.configure(config)

picam2.start()

np_array = picam2.capture_array()`

davidplowman, 2023-04-17:
Hi, thank you for the questions.

The import thing to understand is that this section of the documentation is about writing Qt applications. When you're _not_ writing Qt applications, for  example you're just using a script like the code you listed above, you can call all these functions directly and without thinking about it. That's because your script is running in Python's main interpreter thread, and there's _another_ thread running in the background which does stuff, and your script can sit back and block while it waits for that other thread to do stuff.

But things are very different when you're writing a Qt app. In this case, your code (most likely) runs in the Qt event loop, which is also where the camera does stuff. There is no separate camera event loop now that you can sit back and wait for. So you can no longer safely call something like `capture_array()` because it may need to wait for the camera to do something which will now never happen because the entire event loop is stuck. Does that make sense?

carterw, 2023-04-17:
That does make sense, thanks for the good explanation.