raspberrypi/picamera2 #667: [HOW-TO] signal_done, done_signal, and capture_done

carterw, 2023-04-15:
On page 57 of the manual I am seeing some example code that attempts to show how these functions are used but it is confusing.

`picam2.switch_mode_and_capture_file(cfg, "test.jpg", signal_function=qpicamera2.signal_done)`

What does qpicamera2.signal_done() look like and what does it do? There's code that supposedly hooks it to a 'done' method;
`qpicamera2.done_signal.connect(capture_done)`

Here is capture_done(), do we have to wait for some reason? And what does the 'result' tell us? 
"if there is a result that we need", what could it do for me?

```
def capture_done(job):
  result = picam2.wait(job)
  button.setEnabled(True)
```

I'd like to see example code in the repository that fleshes this out and explains it better.

carterw, 2023-04-17:
Can you please have a look at the following code;
[https://github.com/pageauc/MoTrack-Picam2-Demo/blob/master/streampilibcam.py](url)

This is experimental code where the author has instantiated a thread which calls picam2.capture_array() in a loop. A consumer would call the read() method to obtain the most recent frame. Now, based on your explanation of blocking versus non-blocking behavior, maybe it isn't necessary to go to all this trouble if it isn't a Qt application? Would be simpler and equally effective to merely call picam2.capture_array() directly instead of having this thread intermediary?

Also I am wondering if it would be possible to eliminate the thread and have a more event-driven model for capturing the arrays at the camera's frame rate. Is there a mechanism where I could invoke picam2.capture_array() with a callback or a signal function that would run at completion of the capture, do something with the frame data, and kick off another capture?

davidplowman, 2023-04-17:
Having an external thread like that is certainly one way to implement things, it really all depends on your use case. You probably want to be clear whether you want to write a Qt application or not. If you do, then you need to be very sure that you never call anything that might wait for the camera system - because doing that in the Qt event loop will deadlock everything.

In other cases, an external thread may be more complexity than it's worth. Can you explain anything more about what you really want to accomplish? I think that's where the answer will lie.

carterw, 2023-04-17:
The use case is detecting the motion of an object horizontally across the view area of the camera and using that to infer the speed of the object. And in this case the 'object' will be cars whizzing past on the street, how fast are they going? It may be that processing frames takes longer than the frame rate of the camera or less time, but in any case it would be nice to have the most recent frame sitting there available when the main loop is ready to evaluate one.

carterw, 2023-04-17:
Here's one implementation I'm looking at, it was done with the older camera library;
[https://github.com/pageauc/speed-camera/blob/master/speed-cam.py](url)

See the speed_camera() method on line 1285, that's the main evaluation loop. There are some "if gui_window_on" statements, but in normal running circumstances with no human being present the gui would be off. Now, it may well be that picam2.capture_array() has so little overhead that it is fine to just call it directly and not worry about events or threads?

davidplowman, 2023-04-18:
> No requirement for a GUI other than to debug the application, I don't think Qt is necessary in this case.
> 
> Is there any performance benefit to using picam2.capture_array() to grab frames in a thread rather than calling it directly?

Probably not. In the end, it's the rate at which you can do the analysis that is presumably important? If that doesn't keep up on average with the camera then you're going to drop frames somewhere, either in a separate thread or in the camera thread. Choice of image formats, resolutions and framerates, whether you need to do image format conversions in software, and the general speed of your algorithms, may be most critical. Best advice (unsurprisingly) is probably to start with the simplest thing and see what happens...!

carterw, 2023-04-18:
Thanks for the very helpful advice!

closed 2023-04-18 by carterw