raspberrypi/picamera2 #669: [HOW-TO]How do I output video as mp4?

JanGordon, 2023-04-16:
**Please only ask one question per issue!**

I am trying to record video as mp4 but ffmpeg seems to throw an error. 

I used the example code in the mp4_capture file but this is the error:

      libavutil      56. 51.100 / 56. 51.100
      libavcodec     58. 91.100 / 58. 91.100
      libavformat    58. 45.100 / 58. 45.100
      libavdevice    58. 10.100 / 58. 10.100
      libavfilter     7. 85.100 /  7. 85.100
      libavresample   4.  0.  0 /  4.  0.  0
      libswscale      5.  7.100 /  5.  7.100
      libswresample   3.  7.100 /  3.  7.100
      libpostproc    55.  7.100 / 55.  7.100
    Hyper fast Audio and Video encoder
    usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

    Use -h to get full help or, even better, run 'man ffmpeg'
    Exception in thread Thread-2:
    Traceback (most recent call last):
      File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
        self.run()
      File "/usr/lib/python3.9/threading.py", line 892, in run
        self._target(*self._args, **self._kwargs)
      File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 200, in thread_poll
        self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
      File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 238, in outputframe
        out.outputframe(frame, keyframe, timestamp)
      File "/usr/lib/python3/dist-packages/picamera2/outputs/ffmpegoutput.py", line 85, in outputframe
        self.ffmpeg.stdin.flush()  # forces every frame to get timestamped individually
    BrokenPipeError: [Errno 32] Broken pipe

I tried using audio=False in FfmpegOutput as I saw suggested in a different issue but that didn't change anything. Also, ffmpeg is definitely installed and the cli works but I haven't installed any pip package for it.

I'm using an arducam 16mp camera with imx519 sensor.

davidplowman, 2023-04-17:
Hi, so it looks like the FFmpeg process is unhappy and has quit by the time we try to send it stuff. Just to be clear, do you get this error when you run `mp4_capture.py` without making any changes to it? If so, can you also print out the entire console log when you run that example. Thanks!

JanGordon, 2023-04-18:
No actaully I was being stupid.. I was in the wrong directory but now it works. The mp4_capture doesn't hwoever.

JanGordon, 2023-04-18:
I reinstalled it but it stilll appears to have the same issue

davidplowman, 2023-04-18:
Do you have a spare SD card on which to burn a clean image and try that? If that works, then it's a case of finding out how your installation differs. If it doesn't... well that would be strange, but at least we'd know to look somewhere else.

JanGordon, 2023-04-18:
I only flashed this image a few days ago and I haven't changed much but I can give it a go anyway.