raspberrypi/picamera2 #670: [BUG] autofocus doesn't complete

carterw, 2023-04-16:
I'm running bullseye with a camera module 3. I wanted to experiment with the autofocus and was looking at page 28 of the PiCamera2 Library Manual. The example code from there is shown below. 

The call to picam2.autofocus_cycle() never returns. I am not seeing any examples in the Git repository here that use autofocus_cycle().

```
from picamera2 import Picamera2
from libcamera import controls
picam2 = Picamera2()
picam2.start(show_preview=True)
success = picam2.autofocus_cycle()
```

davidplowman, 2023-04-17:
Hi, I'm not able to reproduce that symptom, so I'm not sure what's happening. Can you perhaps run the following:
```
from picamera2 import Picamera2

def print_af_state(request):
    md = request.get_metadata()
    print(("Idle", "Scanning", "Success", "Fail")[md['AfState']], md.get('LensPosition'))

picam2 = Picamera2()
picam2.pre_callback = print_af_state
picam2.start(show_preview=True)
success = picam2.autofocus_cycle()
picam2.pre_callback = None
```
and report what it prints out (obviously just the first few dozen lines if it doesn't stop). For reference, I see the following
```
Idle None
Idle None
Idle None
Idle None
Scanning None
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.0
Scanning 1.3267180919647217
Scanning 1.3267180919647217
Scanning 1.3667539358139038
Scanning 1.3667539358139038
Scanning 1.3667539358139038
Scanning 1.3667539358139038
Scanning 1.3667539358139038
Scanning 1.3667539358139038
Scanning 1.3667539358139038
Success 1.3667539358139038
```
after which it stops.

closed 2023-04-17 by carterw

davidplowman, 2023-04-17:
No problem - that's exactly right. You need to start the camera system before doing anything that waits for it to produce images. To be fair, I've sometimes wondered about putting in an explicit catch for this, maybe I should consider that again.

LeifSec, 2024-04-11:
I have got a similar problem with a ìmx708`- `autofocus_cycle()` does not complete.  It also cannot be stop by
 running with `wait=False`and doing 
* `picam2.stop()` or 
* `picam2.set_controls({'AfTrigger': libcamera.controls.AfTriggerEnum.Cancel})`  (from #884)

Running the code from [davidplowman](https://github.com/davidplowman) above I got:
``` 
>>> picam2.start(show_preview=True)
[8:53:04.070405711] [18238]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 1536x864-SBGGR10_CSI2P
[8:53:04.071946033] [18244]  INFO RPI vc4.cpp:611 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
>>> Idle 1.0
Exception in thread Thread-2 (thread_func):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 29, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 83, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1278, in process_requests
    display.render_request(display_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 74, in render_request
    self.render_drm(self.picam2, completed_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 153, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
```

quitmeyer, 2024-06-12:
> Hi, I'm not able to reproduce that symptom, so I'm not sure what's happening. Can you perhaps run the following:
>
> ```
> from picamera2 import Picamera2
>
> def print_af_state(request):
>     md = [...]

Thanks for sharing this, I couldn't find a good example of how to scan if my autofocusing was working! Appreciate it!

quitmeyer, 2024-06-12:
> picam2.start(show_preview=True)

A weird challenge I have been having is that in my setup, picamera2 will not FULLY focus unless I am visually previewing the output (which goes WAYYY slower)

for instance if I use the code from @davidplowman above, it works great. it just takes 90 seconds to focus, and i get a value like
7.685 that looks great for the lensposition

but if i switch "show_preview" to false
the whole thing runs way quicker, in just a couple seconds, BUT it claims success at a lensposition of like 6.89, and it is all blurry looking.

i guess this is maybe a difference between the "preview" and "still" config or something? but im not sure what's wrong or how to make it to focus well without previewing it.

davidplowman, 2024-06-13:
> A weird challenge I have been having is that in my setup, picamera2 will not FULLY focus unless I am visually previewing the output (which goes WAYYY slower)
>
> for instance if I use the code from [...]

Can you say what camera you're using? Can you say exactly how you're configuring the camera and, if it's not a standard Pi camera, what camera mode is being chosen? In fact it would be helpful to post the entire script (as much as is required just to start the camera and run the autofocus cycle) too, just to be sure we know what's running.

Displaying the preview on the screen _shouldn't_ make a difference, though depending on the camera configuration I guess it might cause frame drops which may affect the behaviour. In turn, maybe that suggests that the AF algorithm hasn't been tuned properly. If it's a third party camera, maybe they never did that Or if it's one of ours, then we should be able to investigate.

Adding @njhollinghurst in case he has any suggestions. Might there be some parameters to tweak or debug to enable?

quitmeyer, 2024-06-13:
> Can you say what camera you're using? Can you say exactly how you're configuring the camera and, if it's not a standard Pi camera, what camera mode is being chosen? In fact it would be helpful to [...]

It's the Arducam64Mp Hawkeye. And I'm running Bookworm (last updated in March i believe) on a Pi4

and I have the problem in a big long script about taking and processing the moth photos, BUT, it's replicable on your code you posted above or my modified version (that just turns a bright light on and off)

```
from picamera2 import Picamera2

import RPi.GPIO as GPIO

Relay_Ch1 = 26
Relay_Ch2 = 20
Relay_Ch3 = 21

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(Relay_Ch1,GPIO.OUT)
GPIO.setup(Relay_Ch2,GPIO.OUT)
GPIO.setup(Relay_Ch3,GPIO.OUT)

print("Setup The Relay Module is [success]")

def flashOn():
    #GPIO.output(Relay_Ch3,GPIO.LOW)
    GPIO.output(Relay_Ch2,GPIO.LOW)
    print("Flash On\n")
    
def flashOff():
    GPIO.output(Relay_Ch2,GPIO.HIGH)
    print("Flash Off\n")
    

def print_af_state(request):
    md = request.get_metadata()
    print(("Idle", "Scanning", "Success", "Fail")[md['AfState']], md.get('LensPosition'))

flashOn()

picam2 = Picamera2()

picam2.pre_callback = print_af_state
picam2.start(show_preview=True)
success = picam2.autofocus_cycle()
picam2.pre_callback = None
flashOff()

```

If i leave the line 
`picam2.start(show_preview=True)`

then it fully autofocuses (but takes 90 seconds)

if i change it to 

`picam2.start(show_preview=False)`

the focusing takes about 1-2 seconds, but is generally off by like a whole dioptre

my current thought is maybe without the preview its somehow doing the focusing at a really low resolution? So it thinks it's focused, but it can't tell?

BONUS challenge!
I used that code above since it was reliable and repeatable to get my camera to auto focus (even though a full autofocus round took about 2 mins). And it worked when i manually launched the scripts. but when i set it out last night and cron ran the scripts, the auto-focus script would always fail with the error:

```
Exception in thread Thread-2 (thread_func):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 29, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 83, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1278, in process_requests
    display.render_request(display_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 74, in render_request
    self.render_drm(self.picam2, completed_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 153, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane
Terminated
```

Which seems like for some reason it can't run a preview if it is launched from cron?
I can also confirm if i change the part of the script to "show_preview=False" cron can run it just fine (we just run into the problem where it then can't fully autofocus)

quitmeyer, 2024-06-13:
Also I don't think I mentioned, but like a regular libcamera-hello command like:
```
hello_command = ["libcamera-hello","--analoggain", "1", "--info-text", "'lens %lp' 'shutter %exp' 'analogue gain %ag", "-t", "0","--vflip"       ]
```
that I call from python will do what i consider "expected behavior" for this camera. It will take maybe 5-8 seconds and get a nice focus. That could totally be a workable solution for me, but I don't think I can get the focus value from this (even though i can write it to the GUI).

[image]

quitmeyer, 2024-06-17:
have found the solution in the form of an Arducam user posting his own corrected version of Arducams install files in the past day or so. Looks like they had something janky going on in the past couple updates, and the user JakeBrowning's fix fixed it.
(cmon arducam, this feels pretty sloppy with little typos in your code!)
https://forum.arducam.com/t/arducam-64mpx-hawkeye-b0399-no-autofocus/7031/8