raspberrypi/picamera2 #672: [HOW-TO] Readout HQ Camera has a built-in temperature sensor without capturing images

Bra1nsen, 2023-04-18:
I would like to catch the temperature of the camera without capturing images, is that possible? And if yes, how?

davidplowman, 2023-04-18:
I think you should find `'SensorTemperature'` in the image metadata... let me know if it isn't there!

Bra1nsen, 2023-04-23:
yes sure, but withouth capturing images?

davidplowman, 2023-04-24:
I wasn't quite sure whether you meant "with the camera running and delivering frames, but without the application code reading them" or "without the camera streaming frames at all".

In the first case, you can use `picam2.capture_metadata()`, though of course the camera is delivering frames, and libcamera is receiving them but then they are being discarded without being passed to the application.

In the second case I'm not sure. I think in libcamera you're not going to be able to get those results without running the camera (as described just above). Maybe you could bypass libcamera completely and go straight to V4L2? Something like `v4l2-ctl -d /dev/v4l-subdev0 --list-ctrls` should list all the camera settings and values. Though if the camera isn't streaming frames then presumably it won't get so hot either.

Have I understood the question? Thanks!

Bra1nsen, 2023-04-24:
Yes the idea was to monitor the sensor temperature without using libcamera, but acutally its not that important. Thanks for the hint with V4L2, I will give it a try.

PS: Iam still working on adding tags to exif metadata, hope I can show results soon.

closed 2023-04-24 by Bra1nsen

Oxize, 2023-08-22:
Found a way to read out the temp? I need it for something else.