raspberrypi/picamera2 #673: [OTHER] Variable Framerates Multiexposure Capturing

Bra1nsen, 2023-04-18:
[image]

[image]

> One slight caveat is that we have a sensor/driver bug related to the imx477 which means that for this to work correctly all the time, you need to use a fixed framerate, but if I recall correctly, I think you may be doing this anyway.

Iam currently thinking about this, is it possible to change/vary framerates now without issues?

davidplowman, 2023-04-18:
Unfortunately not. We have no idea why the driver and sensor are behaving like this. There's no problem varying the exposure time, just the framerate - and of course the framerate will limit the maximum exposure.

Bra1nsen, 2023-04-24:
which framerate will be set?

```
def exposure_bracket():
    with picamera2.Picamera2(tuning=tuning) as camera:
        config = camera.create_still_configuration(main={"size": (1014, 760), "format": "RGB888"},
                                                   raw={"format": "SRGGB12", "size": (2032, 1520)})
        camera.configure(config)

        for exp in exposures:
            camera.set_controls({"ExposureTime": exp, "AnalogueGain": 1, "ColourGains": (1.0, 1.0)})
            camera.start()
            metadata = camera.capture_file(f"{exp}.tga")
            camera.stop()
```

Since I work with very small exposure times `exposures = [50, 100, 150, 250, 400, 650, 1050, 1700, 2750]` there would be no benefit of beeing able to change framerate or?

It only becomes necessary if one wants to work with long exposures for nigh sky images or do you see different use cases for beeing able to change framerate variable?

closed 2023-05-04 by Bra1nsen