raspberrypi/picamera2 #674: [HOW-TO] Metadata Expansion

Bra1nsen, 2023-04-18:
```
#!/usr/bin/python3

import time
from datetime import timezone, timedelta, datetime
from picamera2 import *

tuning = Picamera2.load_tuning_file("imx477.json")

t1 = time.perf_counter()
exposures = [50, 150, 250, 400, 650, 1050, 1700, 2750, 5250]

def exposure_bracket():
    with picamera2.Picamera2(tuning=tuning) as camera:
        config = camera.create_still_configuration(main={"size": (1014, 760), "format": "RGB888"},
                                                   raw={"format": "SRGGB12", "size": (2032, 1520)})
        camera.configure(config)

        start = time.monotonic()
        for exp in exposures:
            camera.set_controls({"ExposureTime": exp, "AnalogueGain": 1, "ColourGains": (1.0, 1.0)})
            camera.start()

            metadata = camera.capture_file(f"{exp}.png")

            camera.stop()

        end = time.monotonic()
        print("Time:", end - start)

exposure_bracket()

t2 = time.perf_counter()
print(f"Finished in {t2 - t1} seconds")

```

using exiftool for showing metadata: 

[image]
I would like to add "lux", "sensor_temperature" and some histogram values. any idea how this is possible?

davidplowman, 2023-04-18:
You'd need DNG/Exif to support the tags that you want to add. Probably track the DNG spec down online and see if that gives you sufficient information. If there are supported tags that you want to add, it would then be a case either of editing PiDNG (more work), or you could possibly just use exiftool to update those tags once the file has been written (less work).

Bra1nsen, 2023-04-27:
with your help (thanks!) and this [documentation](https://piexif.readthedocs.io/en/latest/functions.html#dump) i got it.

example script:

```
from picamera2 import *

exposures = [50, 150, 250, 400, 650, 1050, 1700, 2750, 5250]

def is_daytime(s):
    now = datetime.now(pytz.timezone(city.timezone))
    return s["sunrise"] <= now <= s["sunset"]

def save_image_with_custom_exif(image_path, metadata):
    # Load the Exif data
    exif_dict = piexif.load(image_path)

    # Add custom Exif tags using the Exiftool format
    exif_dict["0th"][piexif.ImageIFD.Make] = "Raspberry Pi"
    exif_dict["0th"][piexif.ImageIFD.Model] = "Raspberry Pi High Quality Camera"
    exif_dict["Exif"][piexif.ExifIFD.DateTimeOriginal] = datetime.now().strftime("%Y:%m:%d %H:%M:%S")

    # Add custom metadata to the Exif UserComment
    exif_dict["Exif"][piexif.ExifIFD.UserComment] = json.dumps(metadata).encode('utf-8')

    # Dump the modified Exif data
    exif_bytes = piexif.dump(exif_dict)

    # Save the image with the new Exif data
    piexif.insert(exif_bytes, image_path)

def exposure_bracket():
    while True:
        now = datetime.now()
        s = sun(city.observer, date=now)  # Update s within the while True loop

        while not is_daytime(s):
            time.sleep(60)  # Wait 1 minute before checking again
            s = sun(city.observer, date=datetime.now())  # Update s after waiting

        with picamera2.Picamera2(tuning=tuning) as camera:
            config = camera.create_still_configuration(main={"size": (1014, 760), "format": "RGB888"},
                                                       raw={"format": "SRGGB12", "size": (2032, 1520)})
            camera.configure(config)

            while is_daytime(s):
                save_path = create_save_path()
                start = time.monotonic()
                for exp in exposures:
                    camera.set_controls({"ExposureTime": exp, "AnalogueGain": 1, "ColourGains": (1.0, 1.0)})
                    camera.start()

                    timestamp = datetime.now().strftime("%Y%m%d%H%M%S")
                    file_name = f"{timestamp}_{exp}.jpg"
                    full_path = os.path.join(save_path, file_name)
                    metadata = camera.capture_file(full_path)
                    lux = metadata["Lux"]
                    print(lux)
                    save_image_with_custom_exif(full_path, metadata)

                    end = time.monotonic()
                    execution_time = end - start
                    print(f"Exposure Time: {exp} us")
                    print(f"Execution Time: {execution_time:.2f} seconds")
                    print(f"Metadata: {metadata}")
                    print(f"Saved to: {full_path}\n")

                    camera.stop()

                time.sleep(capture_interval)
                s = sun(city.observer, date=datetime.now())  # Update s after capturing images

if __name__ == "__main__":
    exposure_bracket()
```

example output exiftool:

```
ExifTool Version Number         : 12.16
File Name                       : 20230427120358_5250.jpg
Directory                       : /media/dlr/usbstick/Dataset/2023/Cloud_Cam_Eagle/04/27/12
File Size                       : 82 KiB
File Modification Date/Time     : 2023:04:27 12:03:58+01:00
File Access Date/Time           : 2023:04:27 01:00:00+01:00
File Inode Change Date/Time     : 2023:04:27 12:03:58+01:00
File Permissions                : rwxrwxr-x
File Type                       : JPEG
File Type Extension             : jpg
MIME Type                       : image/jpeg
Exif Byte Order                 : Big-endian (Motorola, MM)
Make                            : Raspberry Pi
Camera Model Name               : Raspberry Pi High Quality Camera
Software                        : Picamera2
Modify Date                     : 2023:04:27 12:03:58
Exposure Time                   : 1/191
ISO                             : 100
Date/Time Original              : 2023:04:27 12:03:58
Warning                         : Invalid EXIF text encoding for UserComment
User Comment                    : {"FocusFoM": 151, "ExposureTime": 5247, "SensorTemperature": 0.0, "SensorTimestamp": 71576501403000, "ScalerCrop": [0, 0, 4056, 3040], "ColourTemperature": 5932, "DigitalGain": 1.0004446506500244, "AnalogueGain": 1.0, "ColourGains": [1.0, 1.0], "AeLocked": false, "SensorBlackLevels": [4096, 4096, 4096, 4096], "Lux": 11786.9912109375, "FrameDuration": 24994, "ColourCorrectionMatrix": [1.9874064922332764, -0.8473411798477173, -0.1400553286075592, -0.2653316855430603, 1.7090944051742554, -0.44377270340919495, -0.096389040350914, -0.47202178835868835, 1.568410873413086]}
Image Width                     : 1014
Image Height                    : 760
Encoding Process                : Baseline DCT, Huffman coding
Bits Per Sample                 : 8
Color Components                : 3
Y Cb Cr Sub Sampling            : YCbCr4:2:0 (2 2)
Image Size                      : 1014x760
Megapixels                      : 0.771
Shutter Speed                   : 1/191
```

example to get "Lux" :

```
import subprocess
import json

image_path = "/media/usbstick/20230427130308_5250.jpg"

exiftool_path = "/usr/bin/exiftool"

# Run the exiftool command and capture the output as a JSON string
output = subprocess.check_output([exiftool_path, '-j', image_path])
metadata = json.loads(output.decode('utf-8'))[0]

# Extract the "User Comment" field and parse the JSON string inside it
user_comment = metadata.get('UserComment')
if user_comment:
    user_comment_dict = json.loads(user_comment)

    # Extract the "Lux" value from the parsed JSON dictionary
    lux = user_comment_dict.get('Lux')
    print("Lux:", lux)
else:
    print("No user comment found in metadata")
```

`Lux: 11848.8828125`

closed 2023-04-27 by Bra1nsen