raspberrypi/picamera2 #683: [HOW-TO] SONY IMX219 Camera Frame size and Crop Limits

ekef, 2023-04-24:
Hello, I am using 2 stereo SONY IMX219 sensors on raspberry pi cm4. I want to take quality photos with 1240x375 resolution from these cameras. When I set this resolution with create_still_configuration, the camera shoots close as if zoomed in. I solved this problem by changing sensor mode but only for create_preview_configuration.

The sensor modes of my camera are as follows:

`[{'format': SRGGB10_CSI2P, 'unpacked': 'SRGGB10', 'bit_depth': 10, 'size': (640, 480), 'fps': 103.33, 'crop_limits': (1000, 752, 1280, 960), 'exposure_limits': (75, 11766829, None)}, 
{'format': SRGGB10_CSI2P, 'unpacked': 'SRGGB10', 'bit_depth': 10, 'size': (1640, 1232), 'fps': 41.85, 'crop_limits': (0, 0, 3280, 2464), 'exposure_limits': (75, 11766829, None)}, 
{'format': SRGGB10_CSI2P, 'unpacked': 'SRGGB10', 'bit_depth': 10, 'size': (1920, 1080), 'fps': 47.57, 'crop_limits': (680, 692, 1920, 1080), 'exposure_limits': (75, 11766829, None)},
 {'format': SRGGB10_CSI2P, 'unpacked': 'SRGGB10', 'bit_depth': 10, 'size': (3280, 2464), 'fps': 21.19, 'crop_limits': (0, 0, 3280, 2464), 'exposure_limits': (75, 11766829, None)}, 
 {'format': SRGGB8, 'unpacked': 'SRGGB8', 'bit_depth': 8, 'size': (640, 480), 'fps': 103.33, 'crop_limits': (1000, 752, 1280, 960), 'exposure_limits': (75, 11766829, None)}, 
 {'format': SRGGB8, 'unpacked': 'SRGGB8', 'bit_depth': 8, 'size': (1640, 1232), 'fps': 41.85, 'crop_limits': (0, 0, 3280, 2464), 'exposure_limits': (75, 11766829, None)}, 
 {'format': SRGGB8, 'unpacked': 'SRGGB8', 'bit_depth': 8, 'size': (1920, 1080), 'fps': 47.57, 'crop_limits': (680, 692, 1920, 1080), 'exposure_limits': (75, 11766829, None)},
 {'format': SRGGB8, 'unpacked': 'SRGGB8', 'bit_depth': 8, 'size': (3280, 2464), 'fps': 21.19, 'crop_limits': (0, 0, 3280, 2464), 'exposure_limits': (75, 11766829, None)}]`

I `{'format': SRGGB8, 'unpacked': 'SRGGB8', 'bit_depth': 8, 'size': (1640, 1232), 'fps': 41.85, 'crop_limits': (0, 0, 3280, 2464) ), 'exposure_limits': (75, 11766829, None)}]`, I want to take a photo with the resolution I want in the best quality, using the this sensor mode, that is, with full frame size.
When I set create_preview_configuration({"size": (1242, 375)}, raw=picam2a.sensor_modes[5]) I get the image I want, but it is not of good quality.
When I try the same with create_still_configuration({"size": (1242, 375)}, raw=picam2a.sensor_modes[5]) I get an error. Is there a way to this?

davidplowman, 2023-04-24:
Yes, unfortunately there's a bug which means that you can't simply supply a mode from the mode list in some configurations. This is already fixed and will be in the next release. In the meantime you can work round it simply by supplying the size of the mode that you want. It will default to 10 over 8 bit modes, but I would expect that this would be what you want.

For imx219 I would probably use
```
config = picam2.create_preview_configuration({'size': (1242, 375)}, raw={'size': (1640, 1232)})
```
if I wanted to hit 30fps (or more) by using a 2x2 binning mode, or
```
config = picam2.create_still_configuration({'size': (1242, 375)}, raw={'size': (3280, 2464)})
```
if the framerate was less important but I want better quality. 

If things are still not working well enough after this then it would obviously be helpful to supply more explanation, or an example.

Thanks!

ekef, 2023-04-28:
Thank you very much for your reply. I'm also using a stereo camera for streaming video and when taking pictures I need to make sure it's taken at exactly the same time. Currently there is a delay between the 2 cameras when taking pictures and this causes moving objects to be different in both images. How can I achieve this problem?

davidplowman, 2023-05-02:
Unfortunately libcamera itself doesn't handle synchronisation of multiple cameras  so you either have to synchronise them as well as you can in software (and hope it's good enough), or some cameras have a strobe line which you can control, at least for still image captures. So far as I recall that includes the imx477 and imx708 sensors (HQ and v3 cam), but not the imx219. If you want to know more about that, maybe ask on the forum where you'll find people who know more.

How are you synchronising things currently using software? There are obviously a range of different strategies there, some of which will be better than others. If you were able to post a very minimal script which shows what you're doing, that would be helpful. Thanks!