raspberrypi/picamera2 #694: [HOW-TO] Install picamera2 along with libcamera inside arm64 docker container for use on Raspberry Pi cm4

habjoel, 2023-05-08:
Hi there, 

I am trying to install libcamera and picamera2 inside my docker container based on arm64 debian bullseye for use on my raspberry pi cm4 fleet. I cannot install any of these things through apt (`apt-get inastall libcamera-apps libcamera-tools python3-picamera2`). It's always unable to locate the package (why?).

So I installed libcamera from source following [this guide](https://libcamera.org/getting-started.html). I am now able to import libcamera into my python  and it works. I then installed picamera2 through `pip3 install picamera2`. The version of picamera is 0.3.9

However, as I try to import picamera2 into python, it throws the following error:
```
ImportError: cannot import name 'convert_from_libcamera_type' from 'picamera2.utils' (/usr/local/lib/python3.9/site-packages/picamera2/utils/__init__.py)
```
Where does this error come from? Could there be some kind of version mismatch between libcamera and picamera2?

I posted the same question on this [libcamera issue page](https://github.com/kbingham/libcamera/issues/74).

Thank you for any help!!

davidplowman, 2023-05-09:
Hi, Picamera2 is a supported by Raspberry Pi, not by the libcamera project, so this is probably the place to ask.

Also, I'd recommend building libcamera from https://github.com/raspberrypi/libcamera.git as we only update to match the mainline repo once we've tested that it works properly (and occasionally add a temporary patch or two to achieve this). Though I don't expect this has any relation to your problem.

I don't know anything specifically about Docker, but when I install Picamera2 from pip (this is what you're doing?) I get a `python3.9/site-packages/picamera2` folder (as you have, though mine is in `/home/pi/.local`), and in this folder:

* I _don't_ have a `utils` folder. I think Picamera2 used to have one, but it doesn't any more (AFAIK). So I don't understand why you would be seeing one.
* But I _do_ have a `utils.py` file, and this contains the `convert_from_libcamera_type` function.

The version of `picamera2.py` in that folder contains the line `import picamera2.utils as utils`, does yours look the same? It might be worth double-checking what's in your `sys.path` variable, and that you're picking up the version of Picamera2 that you really want, I get a feeling that there might be something older lurking on your system.

Sorry if this isn't a complete answer, but hopefully gives you some further things to look at.

habjoel, 2023-05-10:
Thanks a lot for your answer. I don't know why, but somehow it creates a utils folder if I install picamera2 through pip. I then installed picamera from source by cloning this repo and using the setup.py and everything succeeds. I have no idea what the problem with the pip installation might be but I now got it working :)

davidplowman, 2023-05-10:
Well, I'm glad it's working, though I remain a bit baffled as to why you get that utils folder...!

dsidynalogic, 2023-07-06:
Same problem here, a utils folder is created and seems to be (the ?) reason it was not working as deleting it directly (not a good durable solution) makes it work.

In my particular case, my program runs in a pipenv environment installed with `pipenv --site-packages install` to get the picamera2 from the os (as it was installed with `apt install python3-picamera2` ). Its version is `python3-picamera2/stable,stable,now 0.3.9-1 all [installed]` (command : `apt list --installed | grep python3-picamera2`)

Removed the folder with `rm -rf /home/pi/.local/share/virtualenvs/ui-T3VwWgYX/lib/python3.9/site-packages/picamera2/utils`which makes it work.
python 3.9.2 (both os and pipenv)
Tested as not working with both pip versions identical too ( `pip 23.1.2 from /home/pi/.local/share/virtualenvs/ui-T3VwWgYX/lib/python3.9/site-packages/pip (python 3.9)` and `pip 23.1.2 from /home/pi/.local/lib/python3.9/site-packages/pip (python 3.9)` ).

_some other info : 
kernel_release : 6.1.21-v8+
vcgencmd version: 
Mar 21 2023 17:18:16 
Copyright (c) 2012 Broadcom
version 3cc1c2dfc5460da9e1a0a4f48b48ab508c48bfe5 (clean) (release) (start)_

davidplowman, 2023-07-06:
Hi, thanks for the report. It's obviously related to the fact that there _used_ to be a utils folder, but which got removed. There must be something stuck in some of the build artefacts so I'll go and have another dig to see if I can find it.

dsidynalogic, 2023-07-06:
Fixed it for my particular case (pipenv and site-packages), my error was I still had 
`[packages]
picamera2 = "0.3.9"` in my Pipfile wich is at least useless since I want to use the os version and here is even buggy. Deleted the line and `pipenv --rm` and `pipenv --site-packages install` fixed it. 
However @davidplowman while this fixes it for me since I'm not using the "pip version of picamera2" it might still  mean that the pip version needs fixing (not sure, maybe). 
Thank you,

lurch, 2023-12-18:
See https://github.com/raspberrypi/picamera2/issues/503#issuecomment-1861034147

Pepslee, 2024-01-02:
> Thanks a lot for your answer. I don't know why, but somehow it creates a utils folder if I install picamera2 through pip. I then installed picamera from source by cloning this repo and using the [...]

Can you share the example of your Dockerfile pls ?

ChetanKukreja, 2024-05-04:
@davidplowman 
> Thanks a lot for your answer. I don't know why, but somehow it creates a utils folder if I install picamera2 through pip. I then installed picamera from source by cloning this repo and using the [...]
can you give me a guide how did you do this?

davidplowman, 2024-05-20:
Can you say what version of Picamera2 you are installing? I know there was a time when folks were getting a utils folder, but I though this had been resolved.