raspberrypi/picamera2 #696: [HOW-TO] Is there an exception hook to catch exceptions during capture

Elaschatz, 2023-05-09:
# Is there an exception hook to catch exceptions during capture?

I have a program, which starts a recording via a method like this:
```
def startVideo(self):
         self.isRecording=True
         #if the outdirectory does not exist, create it
         if not os.path.exists(self.out_directory):
             os.makedirs(self.out_directory)
         self.currentFilename=self.generateFilename()
         self.encoder.output = FileOutput(self.out_directory+"/"+self.currentFilename+".h264")
         self.picam2.start_encoder() 
```

When I want to stop the recording, I use the following function
```
def stopVideo(self):
         self.isRecording=False
         self.picam2.stop_encoder()
         self.currentFilename= None 
```

**This all works fine.** 
As I write the videos to a USB stick, sometimes there is no space left on this device and an exception is raised during recording:

 ```Exception in thread Thread-6721:
Traceback (most recent call last):
   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
     self.run()
   File "/usr/lib/python3.9/threading.py", line 892, in run
     self._target(*self._args, **self._kwargs)
   File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 188, in thread_poll
     self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
   File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 214, in outputframe
     out.outputframe(frame, keyframe, timestamp)
   File "/usr/lib/python3/dist-packages/picamera2/outputs/fileoutput.py", line 88, in outputframe
     self._write(frame, timestamp)
   File "/usr/lib/python3/dist-packages/picamera2/outputs/fileoutput.py", line 118, in _write
     self._fileoutput.flush()
OSError: [Errno 28] No space left on device
```

Now my Question.
## Is there any option to catch this exception?
At the moment the exception is printed to the console and I have no possibility to catch it with a `try ...except`. The python script hangs up, and this is not good, as it is a longtime surveillance and the console is not monitored.

I read about `excepthook` in the `threading` module. Is there anything similar which I can use?

Josee-xav, 2025-05-22:
I can confirm that threading.excepthook works but this method sucks as theirs no proper way of handling the issue thats causing the exception as u cannot pass the exception down to the calling/main thread.  Unless their is a way then i would love to know.

davidplowman, 2025-05-23:
Hi, and thanks for the comment. I'm not really too sure what the right answer is here (or if there is one). Can you explain a bit more about what you think should happen?

AIUI, one of the background threads, that's writing out the encoded video frames, is throwing an exception. Might there be a solution where you could provide a callback to the output object which is invoked whenever an error occurs? There's already something like that in the `FfmpegOutput` which lets the user set an `error_callback`, but maybe we could extend it to other outputs? Which one are you using, by the way?

Or if you have any other suggestions, I'd be interested in those too. Thanks!