raspberrypi/picamera2 #698: [HOW-TO] Capture frame directly as grayscale image

simon-staal, 2023-05-10:
Is there some way to capture images directly in black and white / grayscale using `Picamera2`, to avoid having to later convert it to gray using `cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)`? I'm assuming this would be more efficient to perform this operation in-place on the buffer, since it would presumably reduce the amount of data needing to be copied, and it would also be convenient to register this as some kind of `pre_callback`, since I would like all my images to be in grayscale.

Unfortunately, registering a callback to do this, similar to what is shown in the documentation, is not possible, as `cv2.cvtColor` returns a copy of the image being manipulated, rather than modifying it in-place:
```python
def convert_to_grayscale(request):
        with MappedArray(request, "main") as m:
            cv2.cvtColor(m.array, cv2.COLOR_BGR2GRAY) # Does nothing

picam.pre_callback = convert_to_grayscale
```
I've tried writing back to `m.array`, but get an `AttributeError: can't set attribute`. Is there anything I can do to accomplish this?

davidplowman, 2023-05-11:
Hi, I would probably capture a YUV420 format image and then grab the Y plane, so something like this:
```
from picamera2 import Picamera2

picam2 = Picamera2()
WIDTH = 1024
HEIGHT = 768
config = picam2.create_preview_configuration({'format': 'YUV420', 'size': (WIDTH, HEIGHT)})
picam2.configure(config)
picam2.start()
yuv = picam2.capture_array()
grey = yuv[:HEIGHT, :WIDTH]
```
It saves you doing any slow software conversions, so that's a bonus. If you want the preview to show it as greyscale as well, you could set the saturation control to zero too:
```
picam2.set_controls({'Saturation': 0})
```
Does that help?

simon-staal, 2023-05-11:
Perfect, that's exactly what I was looking for! For anyone interested, I'll post the performance details below (the method you suggested is around 2x faster). These results were obtained on a Raspberry Pi Model 3B+ using a High-Quality Camera module. In both cases the time taken to capture an image and convert it to grayscale was considered. The results of the 2 conversions were visually indistinguishable, and both appeared to work fine for further processing with `cv2`.

### Capturing in BGR then converting using openCV

```python
IMG_DIMS = (1280, 720)

def main():
    picam = setup_camera_gray()
while True:
        # Hot loop (measured)
        img = picam.capture_array()
        img_preproc = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)
        # End of hot loop

def setup_camera():
    picam = Picamera2()
    config = picam.create_preview_configuration()
    config['main']['size'] = IMG_DIMS
    config['main']['format'] = "RGB888"
    picam.align_configuration(config)
    print(config)
    picam.configure(config)
    picam.start()

    return picam
```
This implementation averaged 44.16 FPS over 219 frames.

### Capturing in YUV and truncating to convert to grayscale
```python
IMG_DIMS = (1280, 720)

def main():
    picam = setup_camera_gray()
while True:
        # Hot loop (measured)
        img = picam.capture_array()
        img_preproc = img[:IMG_DIMS[1], :IMG_DIMS[0]]
        # End of hot loop

def setup_camera():
    picam = Picamera2()
    config = picam.create_preview_configuration()
    config['main']['size'] = IMG_DIMS
    config['main']['format'] = "YUV420"
    picam.align_configuration(config)
    print(config)
    picam.configure(config)
    picam.start()

    return picam
```
This implementation averaged 104.64 FPS over 211 frames.

maizy3434, 2024-02-11:
5x speed improvement for PiCam V3 still image capture 2 grey at full resolution

dusanlamos, 2024-10-02:
Not sure about performance, but a simple way how to get grayscale output out of picamera2 is to set the saturation to 0.0:

picam2.set_controls({"Saturation": 0.0})

umarinvocation, 2025-02-04:
I found the missing piece

https://lnkr.fun/391ac23