raspberrypi/picamera2 #703: [HOW-TO] Obtain precise Linux time and other metadata for every frame during video recording

prouast, 2023-05-20:
I am trying to obtain a precise Linux timestamp and other metadata for every frame of video recorded using `picamera2`. Is this possible?

**Background**:
I have a Python app that records video with the V2 camera module and other sensor data. Precise Linux timestamps are used to synchronize the various sensor data. Now I am considering to upgrade to the V3 camera module, but I hear that it isn't compatible with my `picamera` code.

**Problem**:
I am currently investigating whether I can replace my existing `picamera` code, which relies on `mmal` to record video to disk and additionally writes frame metadata (including Linux timestamp, gain, and frame type) to `csv`. The reason I'm using `mmal` is to be able to achieve as high recording framerate as possible, and get access to the metadata for each frame.
In my existing `picamera` code, I am realizing  by accessing properties of `buf` and the instance of `mo.MMALCamera` in a callback:

- For timestamps, observing `buf.pts` and comparing with `camera.control.params[mmal.MMAL_PARAMETER_SYSTEM_TIME]` and `time.time()` [as outlined by 6by9](https://forums.raspberrypi.com/viewtopic.php?t=169081#p1086515) 
- For other metadata, observing `buf.flags` and `camera.control.params[mmal.MMAL_PARAMETER_CAMERA_SETTINGS].analog_gain`.

**Thoughts**:
- I have found that it is possible in `picamera2` to get a `pts` csv as an additional output, but this is not enough for my use case
- Am I correct in assuming that using `picam2.post_callback` would be the way to go? I see that the callback gets a `req` passed. Is there anywhere to look up what info this `req` contains?
- Once I have access to the frame `pts`, `gain` etc. I will still need access to the GPU system clock to derive the Linux timestamp corresponding to `pts`. Is this possible with `picamera2`?

Thanks in advance!

davidplowman, 2023-05-22:
Hi, thanks for the question.

All the frames that are produced by libcamera have a timestamp associated with them. You can find this in the metadata associated with that frame (or request), for example
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start()
picam2.capture_metadata()['SensorTimestamp']
```
Using the callbacks to fetch these timestamps sounds reasonable. Another option to consider might be to subclass the Output object, and then you could override its outputtimestamp method.

I'm not quite sure what you mean by "GPU system clock". The timestamp that is associated with the frame _is_ the Linux timestamp for that frame, in nanoseconds, since the system started. Do you need something different?

prouast, 2023-05-22:
Thanks for taking the time to look at this.

Regarding the "GPU system clock", I suspect this is the clock that generates the frame timestamp you mentioned. I was relying on the explanation in these two forum posts: [1](https://forums.raspberrypi.com//viewtopic.php?t=134635#p896490) [2](https://forums.raspberrypi.com/viewtopic.php?t=169081#p1086515):

> SYSTEM_TIME gets the SoC STC (System Time Clock) - effectively time since boot. No direct link between that and Linux time. [...]
> camera.frame.timestamp is the GPU system clock timestamp when it receives the frame start from the camera, so that is the end of exposure for the first line. [...]

I have other sensor data that is timestamped using `time.time()` which is Linux (unix) time, i.e., time since 00:00:00 UTC on 1 January 1970. My practical issue would then be to translate the frame timestamp (measured as time since boot) to Linux time (measured as time since 1970).

In my existing code I am relying on calculating the current delta between `camera.control.params[mmal.MMAL_PARAMETER_SYSTEM_TIME]` and `time.time()` to realise this.
I'm now searching for a way to realise this in `picamera2` - is there still a way to retrieve the current time from the clock that generates the frame timestamps?

davidplowman, 2023-05-22:
Maybe try `time.monotonic()`. So far as I know, this is measuring the same system clock (but reporting in seconds, not nanoseconds).

prouast, 2023-05-22:
Thanks, I will try this. If this doesn't work or isn't accurate enough, it seems to me like the takeaway is that I will need to stick with the older camera module because libcamera does not expose as much information as MMAL did.

Pinging @6by9 : Is that correct or am I missing something?

prouast, 2023-05-22:
Thanks very much for taking the time to clarify how things work with the new stack, it indeed appears `time.monotonic_ns()` is exactly what i need.

Another quick question: Is the new stack compatible with the older camera modules, i.e., will I be able to test this with my camera module 2 before buying the new one?

6by9, 2023-05-23:
Yes, libcamera is compatible with all Raspberry Pi camera modules, and a number of 3rd party ones too.

closed 2023-05-23 by prouast