raspberrypi/picamera2 #708: [HOW-TO] Optimize Frame Rate, Encoding and Serving a MJPEG Stream Simultaneously

jevchance, 2023-05-31:
Greetings, thanks in advance for any assistance you can provide.

I'm working with a Raspberry Pi Zero W and trying to optimize my code to generate h264 files at 30fps without any dropped frames. The code I have assembled below functions fine, but my output files, recorded at 30fps, are generally only 24-26 seconds in length and play back faster than real time. I have also tried the code at 24fps and I'm having the same issue.

The purpose of the code is to generate a web page using an MJPEG stream, and in parallel save footage to disk with a time overlay. As I said this code is functional, but the frame rates are not accurate.

The basis of the MJPEG code is written by @davidplowman (thank you very much) and was taken from this thread: https://github.com/raspberrypi/picamera2/issues/255
Unfortunately all of the code from that thread has been redacted, except David's.

Here's a sample video. The code is written to take a 30 second clip at 30fps, which is confirmed by the time stamp. However, the recorded video is 26 seconds long: https://youtu.be/FjITjgecVn8

Please find the code below, and thanks again for your attention!

```
#!/usr/bin/python3

import logging
import time, cv2, math
import socketserver
from http import server
from threading import Condition, Thread
import simplejpeg
import cv2
from picamera2 import Picamera2, Preview, MappedArray
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput
from datetime import datetime
import numpy as np
from libcamera import controls

SENSORWIDTH, SENSORHEIGHT, SENSORFPS = 2304, 1296, 30 
FRAMETIME = round(1000000/SENSORFPS)
ENCODEWIDTH, ENCODEHEIGHT = 640, 360
CLIPLENGTH = 30
TXTCOLOR = (255, 255, 255, 255)
TXTORIGIN = (50, 50)
TXTFONT = cv2.FONT_HERSHEY_DUPLEX
TXTSCALE = 1
TXTTHICKNESS = 1
OUTPUTPATH = "/dev/shm/"
PAGE = """\
<html>
<head>
<title></title>
</head>
<body>
<img src="stream.mjpg" width="320" height="180" />
</body>
</html>
"""

def mjpeg_encode():
    global mjpeg_frame
    while not mjpeg_abort:
        yuv = picam2.capture_array("lores")
        rgb = cv2.cvtColor(yuv, cv2.COLOR_YUV420p2RGB)
        buf = simplejpeg.encode_jpeg(rgb, quality=80, colorspace='BGR', colorsubsampling='420')
        with mjpeg_condition:
            mjpeg_frame = buf
            mjpeg_condition.notify_all()
            time.sleep(1)

class StreamingHandler(server.BaseHTTPRequestHandler):
    def do_GET(self):
        if self.path == '/':
            self.send_response(301)
            self.send_header('Location', '/index.html')
            self.end_headers()
        elif self.path == '/index.html':
            content = PAGE.encode('utf-8')
            self.send_response(200)
            self.send_header('Content-Type', 'text/html')
            self.send_header('Content-Length', len(content))
            self.end_headers()
            self.wfile.write(content)
        elif self.path == '/stream.mjpg':
            self.send_response(200)
            self.send_header('Age', 0)
            self.send_header('Cache-Control', 'no-cache, private')
            self.send_header('Pragma', 'no-cache')
            self.send_header('Content-Type', 'multipart/x-mixed-replace; boundary=FRAME')
            self.end_headers()
            try:
                while True:
                    with mjpeg_condition:
                        mjpeg_condition.wait()
                        frame = mjpeg_frame
                    self.wfile.write(b'--FRAME\r\n')
                    self.send_header('Content-Type', 'image/jpeg')
                    self.send_header('Content-Length', len(frame))
                    self.end_headers()
                    self.wfile.write(frame)
                    self.wfile.write(b'\r\n')
            except Exception as e:
                logging.warning(
                    'Removed streaming client %s: %s',
                    self.client_address, str(e))
        else:
            self.send_error(404)
            self.end_headers()

class StreamingServer(socketserver.ThreadingMixIn, server.HTTPServer):
    allow_reuse_address = True
    daemon_threads = True

def text_overlay(request):
    overlay_text = time.strftime("%c")
    with MappedArray(request, "main") as m:
        cv2.putText(m.array, overlay_text, TXTORIGIN, TXTFONT, TXTSCALE, TXTCOLOR, TXTTHICKNESS)

def h264_encode():
    while not encode_abort:
        file_name = OUTPUTPATH+time.strftime("%Y-%m-%d_%H-%M-%S")+".h264"
        encoder = H264Encoder(bitrate=1500000)
        encoder.output = FileOutput(file_name)
        picam2.start_encoder(encoder)
        time.sleep(CLIPLENGTH)
        picam2.stop_encoder()

print("\nFrame duration is set to: " + str(FRAMETIME) + "\n")
picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(raw={"size": (SENSORWIDTH, SENSORHEIGHT)}, main={"size": (ENCODEWIDTH, ENCODEHEIGHT)}, lores={"size": (320, 180)}, encode="main", display="main"))
picam2.set_controls({"FrameDurationLimits": (FRAMETIME, FRAMETIME)})
picam2.set_controls({"NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Fast})
picam2.pre_callback = text_overlay

try:
    picam2.start()

    mjpeg_abort = False
    mjpeg_frame = None
    mjpeg_condition = Condition()
    mjpeg_thread = Thread(target=mjpeg_encode, daemon=True)
    mjpeg_thread.start()

    encode_abort = False
    encode_thread = Thread(target=h264_encode, daemon=False)
    encode_thread.start()

    address = ('', 8000)
    server = StreamingServer(address, StreamingHandler)
    server.serve_forever()
    mjpeg_thread.join()
    encode_thread.join()
finally:
    picam2.stop()
```

davidplowman, 2023-05-31:
Hi, I'd be tempted to wind back a little bit and check firstly that you can do 30fps with nothing else going on. Then maybe see what effect the text overlay has, and what happens when you start the jpeg encode. I guess the jpeg encode might be sufficiently expensive to push things over the edge?

If so you might want to try the hardware jpeg codec instead of using the software one. There's an example [here](https://github.com/raspberrypi/picamera2/blob/main/examples/dual_encode.py) showing how you can drive it.

jevchance, 2023-05-31:
Hi David, thanks for the feedback.

I did verify I am able to achieve 30fps with the text overlay, but without the MJPEG stream encoding. I'll dig into the JPEG example you provided, thanks very much.