raspberrypi/picamera2 #712: [HOW-TO] transform raw to grayscale

KruemelmonsterK, 2023-06-09:
Hey guys,

I am currently working with the [Raspberry Pi Camera Module 3 NoIR, 12MP](https://www.berrybase.de/raspberry-pi-camera-module-3-noir-12mp). 

I want to create a video of the complete FOV, but in video mode the FOV is croppend. I guess I can't change that.
My idea is, that I can acquire the complete FOV with the highest frame rate by using frames in raw. Afterwards I will combine the frames, such that I'll have a video. 

I use this code so save the raw frame in an array:
######
# Configure a raw stream and capture an image from it.
from picamera2 import Picamera2, Preview
import time
import numpy as np
from PIL import Image

### Image acquisition in raw
picam2 = Picamera2()
#picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration(raw={"size": picam2.sensor_resolution})
print(preview_config)
picam2.configure(preview_config)

picam2.start()
time.sleep(2)

raw = picam2.capture_array("raw")
print(raw.shape)
print(picam2.stream_configuration("raw"))

picam2.stop()

### Image transformation
array = np.array(raw)

array = array.astype(np.uint8)

image = Image.fromarray(array)

image.save("image.jpg")
######

Saving the array works well, but afterwards, I am not able to transform the array to a "normal looking" grayscale image. 
Besides I am confused. Why is the size of the array (2592, 5760), when it is supposed to have the size of the sensor (4608, 2592)?

The image will look like this: 
[image]

So my question is: Why can't transform the array to a normal looking grayscale image?
Does that have to to with the 'format': 'SBGGR10_CSI2P'?

Thanks for you help! :)

davidplowman, 2023-06-12:
Hi, thanks for the questions. There are lots of things going on here!

First of all, a "raw" image, in the context of cameras, normally means a "raw Bayer" image. This needs lots of processing to turn it into a normal viewable image. In the context of video encoding, people often use "raw" to mean "uncompressed", and this is a common source of confusion (as you have discovered)!

Secondly, you can force the camera to use the full field of view by forcing the camera into a specific mode. We can do this by requesting a size for the raw (= "raw Bayer") stream, which is actually what the image sensor sends us.

Thirdly, probably you'll get the best framerates by encoding as h.264 because there's hardware support for that. Any software encoder is likely to be much slower. You could _avoid_ compressing the images but the catch is that you'll be writing much more data to the disk, which may well end up worse than using the h.264 encoder and writing smaller frames.

Finally, you'll probably also get better framerates by disabling some of the noise processing, and turning off the preview.  Also specify the highest framerate that the camera mode can support. Something like this:
```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder

picam2 = Picamera2()
config = picam2.create_video_configuration({'size': (1280, 720), 'format': 'YUV420'},
                                           raw={'size': (2304, 1296)},
                                           controls={'NoiseReductionMode': 0, 'FrameRate': 56})
picam2.configure(config)
encoder = H264Encoder(bitrate=5000000)
picam2.start_recording(encoder, "output.h264")

time.sleep(10)

picam2.stop_recording()
```

KruemelmonsterK, 2023-06-12:
Well, you code works. Thank you.
While waiting for your answer I tried a few different things. For example libcamera. I get similar results with this command: 
libcamera-vid --framerate 30 --width 1920 --height 1080 -t 10000 -- output video.h264
I got the information for the command from this website (chapter: High Frame Rate Mode): https://www.waveshare.com/wiki/Template:RPi_Camera_Libcamera_Guide

The problem (with your solution and the solution from libcamera) is, that both seem to skip frames. At least, when I play the h264 files via vlc player.

These guys seem to have similar problem: https://github.com/raspberrypi/libcamera-apps/issues/157

How can I avoid frame skipping?

davidplowman, 2023-06-13:
Don't trust vlc - recent versions of it can't play raw h.264 streams properly. Maybe try something like ffplay instead and see how that behaves.

Both Picamera2 and libcamera-vid allow you to write out timestamp files. This is a good way to check whether there are dropped frames. (In Picamera add `pts="timestamps.txt"` into the `start_recording()` call; in libcamera-vid I add `--save-pts timestamps.txt`. In general I would expect libcamera-vid to be less prone to frame drops as it's not all running through the Python interpreter.)