raspberrypi/picamera2 #718: [BUG]

cambridgeBlueMan, 2023-06-13:
**Describe the bug**

QGlPicamera2 causes pyinstaller to fail, i.e. executables produced by pyinstaller from code which utilise the QGlPicamera2 class produce the following terminal output:

```
lea@raspberrypi:~/Code/pkg4/dist/app_full $ sudo ./app_full
OpenGL will not be available
Traceback (most recent call last):
  File "app_full.py", line 13, in <module>
ImportError: cannot import name 'QGlPicamera2' from 'picamera2.previews.qt' (/home/lea/Code/pkg4/dist/app_full/picamera2/previews/qt.pyc)
[7283] Failed to execute script 'app_full' due to unhandled exception!
lea@raspberrypi:~/Code/pkg4/dist/app_full $
```

 It should be noted that if QPicamera2 is substituted for QGlPicamera2 then the rendering is successful. 

**To Reproduce**
Picamera2/apps contains the example file app_full.py, which makes use of the QGlPicamera2 class. If we use the following code to make an executable:

`pyinstaller --windowed full_app.py
`

then attempting to execute the output file app_full produces output from the terminal as shown above

```
lea@raspberrypi:~/Code/pkg4/dist/app_full $ sudo ./app_full
OpenGL will not be available
Traceback (most recent call last):
  File "app_full.py", line 13, in <module>
ImportError: cannot import name 'QGlPicamera2' from 'picamera2.previews.qt' (/home/lea/Code/pkg4/dist/app_full/picamera2/previews/qt.pyc)
[7283] Failed to execute script 'app_full' due to unhandled exception!
lea@raspberrypi:~/Code/pkg4/dist/app_full $
```

**Hardware :**
 Raspberry pi 4b, bullseye, HQ camera.

**Additional context**

from the manual:
Picamera2 provides two Qt widgets: 
• QGlPicamera2 - this is a Qt widget that renders the camera images using hardware acceleration through the Pi’s GPU. 
• QPicamera2 - a software rendered but otherwise equivalent widget. This version is much slower and the QGlPicamera2 should be preferred in nearly all circumstances except those where it does not work (for example, the application has to operate with a remote window through X forwarding).

Is this a circumstance "where it does not work" or can it be fixed? Because it is not an ideal situation if one cannot package an application which contains instances of the QGlPicamera2 class.

davidplowman, 2023-06-14:
Hi, thanks for the report. I did try this with the same results as you. As far as I can tell, it gets unhappy when you import anything from OpenGL. For example, I tried using this "application":
```
#!/usr/bin/python3

import os

os.environ["PYOPENGL_PLATFORM"] = "egl"

from OpenGL.EGL.EXT.image_dma_buf_import import *

print("hello")
```
It runs fine "normally", but after running it through pyinstaller we get import failures.

I'm afraid I don't know anything about pyinstaller and OpenGL, but if anyone else reading this does, or has time to investigate, then that would obviously be useful. It seems like getting the demo app above to work would be a good first step. Sorry that I can't be more helpful at the moment.

ZixuanLiu4869, 2023-10-30:
> Hi, thanks for the report. I did try this with the same results as you. As far as I can tell, it gets unhappy when you import anything from OpenGL. For example, I tried using this "application": [...]

After further exploring the OpenGL module. I found the error actually comes from the OpenGL/plugins.py:
        def importByName( fullName ):
              """Import a class by name"""
              name = fullName.split(".")
              moduleName = name[:-1]
              className = name[-1]
              module = __import__( ".".join(moduleName), {}, {}, moduleName)
              return getattr( module, className )
when it tries to load "OpenGL.platform.egl", it has the error "No module named "OpenGL.platform.egl"" and this cause all the returned plugin being None and then cause the problem. In order to fix that, before running pyinstaller, ensure that export PYOPENGL_PLATFORM=egl, which means we want the egl platform. And add the missing module by "pyinstaller --hidden-import=OpenGL.platform.egl -F your_script.py". Then everything should work and also the app code.

davidplowman, 2023-11-06:
Thanks very much for this update. Does that resolve this problem (at least as far as we can resolve it), or is there anything more to do? Thanks.

cambridgeBlueMan, 2023-11-07:
Hi David, thanks very much for this, and sorry for the delay in getting back to you. I can confirm that the small test file I was using for this now builds successfully. 

Perhaps more importantly, the app I am building which is relatively speaking much more complex and diverse also built successfully.

Very pleased! I needed a win today, even if it came from someone else :)

closed 2023-11-07 by cambridgeBlueMan