raspberrypi/picamera2 #720: [BUG] picam2.start_recording(encoder, 'test.h264', Quality.HIGH)

KruemelmonsterK, 2023-06-17:
from picamera2.encoders import H264Encoder, Quality
from picamera2 import Picamera2
import time
picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration())
encoder = H264Encoder()
picam2.start_recording(encoder, 'test.h264', Quality.HIGH)
time.sleep(10)
picam2.stop_recording()

This Code from your Documentation doesnt work.

davidplowman, 2023-06-19:
Thanks for pointing that out. It should of course be
```
picam2.start_recording(encoder, 'test.h264', Quality.HIGH)
```
I'll get the documentation updated.

KruemelmonsterK, 2023-06-21:
No worries, I am happy to help.

Sorry, but I don't see any difference between the correction and the original command.

davidplowman, 2023-06-21:
Sorry, slip of the mouse. Have corrected it above. Thanks!

KruemelmonsterK, 2023-06-21:
Thank you! Its closed now.