raspberrypi/picamera2 #721: [HOW-TO] Adjust frame rate in response to required exposure time

captainjack64, 2023-06-18:
Hello

I currently have a simple setup of mjpeg web streaming with motion detection and recording of bird feeder, thanks to numerous examples here. My framerate is set to 30fps.

The issue is that the exposure time is limited due to this. If I wanted a higher exposure time, say when it's darker, I need to (currently manually) adjust framerate to a lower value in order for auto-exposure feature to go as high as it can. I found that I don't get much more benefit going below 10fps.

I was wondering if there's a way to adjust the frame rate based on the required exposure time. Say I want exposure time to be 40ms, the frame rate should be 25fps. If I want exposure to be 33ms or lower, I can set it to 30fps.

I've not seen any examples or documentation that would allow this to happen  on the fly. 

My current train of thought is to set the required fps during motion detection but before the recording starts. For example:

1. Set fps to 10
2. Run through motion detect function until one is detected
3. Get camera metadata to get the current exposure time
4. Set the maximum possible fps based on that
5. Start recording
6. Finish recording
7. Goto 1

While I think this will work, it doesn't look like the most elegant way to do this.

Any other thoughts or ideas?

captainjack64, 2023-06-19:
Yes, I am an idiot - all I needed to do was set `FrameDurationLimits` and that would allow variable frame rate based on exposure required.

davidplowman, 2023-06-19:
Well done, I was just starting to type exactly that!!