raspberrypi/picamera2 #722: [BUG] Unable to use Picamera2 as a systemdservice with FfmepgOutput

opoupel, 2023-06-23:
**Describe the bug**
Using the example file audio_video_capture.py, and defining the Python file to run as a systemd service lead to error and no video recorded.

**To Reproduce**

- Create the audio_video_capture.py file using the example
- Test it. It records a test.mp4 file
- Create the file /etc/systemd/system/capture_video_mp4.service

```
[Service]
ExecStart=/usr/bin/python3 /home/olivier/tests/capture_video_MP4.py
WorkingDirectory=/home/olivier/tests/
User=olivier
Group=olivier
Type=simple 
[Install]
WantedBy=multi-user.target
```
- Launch the service
`$ sudo systemctl start capture_video_mp4.service
`

- No video is created
- Watch service status
```
$ sudo systemctl status capture_video_mp4.service 
● capture_video_mp4.service
     Loaded: loaded (/etc/systemd/system/capture_video_mp4.service; disabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-06-23 18:03:21 CEST; 11s ago
   Main PID: 1233 (python3)
      Tasks: 8 (limit: 779)
        CPU: 1.838s
     CGroup: /system.slice/capture_video_mp4.service
             └─1233 /usr/bin/python3 /home/olivier/tests/capture_video_MP4.py

Jun 23 18:03:22 raspberrypi python3[1233]:     self.run()
Jun 23 18:03:22 raspberrypi python3[1233]:   File "/usr/lib/python3.9/threading.py", line 892, in run
Jun 23 18:03:22 raspberrypi python3[1233]:     self._target(*self._args, **self._kwargs)
Jun 23 18:03:22 raspberrypi python3[1233]:   File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 200, in thread_poll
Jun 23 18:03:22 raspberrypi python3[1233]:     self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
Jun 23 18:03:22 raspberrypi python3[1233]:   File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 238, in outputframe
Jun 23 18:03:22 raspberrypi python3[1233]:     out.outputframe(frame, keyframe, timestamp)
Jun 23 18:03:22 raspberrypi python3[1233]:   File "/usr/lib/python3/dist-packages/picamera2/outputs/ffmpegoutput.py", line 84, in outputframe
Jun 23 18:03:22 raspberrypi python3[1233]:     self.ffmpeg.stdin.write(frame)
Jun 23 18:03:22 raspberrypi python3[1233]: BrokenPipeError: [Errno 32] Broken pipe
```

**Expected behaviour**
The mp4 video should be recorded as expected

**Hardware :**
```
$ libcamera-hello --list-cameras 
Available cameras
-----------------
0 : imx708_wide_noir [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

```
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```
```
$ cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
```

**Additional context**
The same test with the the file capture_video.py that records an h264 file and does not use FfmpegOuput, work properly.

youmustfight, 2023-12-12:
Also noticing this. When I ssh to my pi and run my python file it's fine but can't do via systemd

davidplowman, 2023-12-13:
I'm afraid I don't know anything about systemd. I would start by figuring out how to get something to run successfully using just Ffmpeg (ignoring cameras and anything else). After that, move on to a camera application. You might find people with systemd expertise on some of the other Raspberry Pi forums. Sorry not to be more helpful.

signag, 2024-01-16:
Hi, seing you issue, I initially thought that I had a similar problem with my Pi camera web server when recording videos in a service  ([https://github.com/signag/raspi-cam-srv](https://github.com/signag/raspi-cam-srv)). 
 But this finally turned out to be a path issue.

When I tried your test case, that worked fine when run as service in my environment. 
However, I had to set audio=False in the output. 

I tried this on a Debian 11 bullseye and a Debian 12 bookworm system and it worked on both.

With audio=True, as in the original example code, no .mp4 was written and I got
- the same exception as you when run as service under bullseye
- no exception when run as service under bookworm

davidplowman, 2024-01-17:
> Hi, seing you issue, I initially thought that I had a similar problem with my Pi camera web server when recording videos in a service (https://github.com/signag/raspi-cam-srv). But this finally [...]

Hi, thanks for your message. After reading it I wasn't totally sure whether you were reporting a problem, or something that you have got to work? If there is a problem, please include a precise description so that we can look at it (include the type of Pi, camera, OS, and a minimal snippet of code that demostrates the problem). Thanks!

signag, 2024-01-19:
Sorry for being not clear enough. Just wanted to share my experience while trying to get Picamera2 Apps to run as service.

In the meantime, I found a solution which worked at least on a **Pi 5** with **Bookworm** and an attached USB microphone:

In order to record audio together with the video (audio=True), Picamera2 requires access to the audio infrastructure, which is achieved through [PulseAudio](https://wiki.archlinux.org/title/PulseAudio).
However, the relevant PulseAudio service (pulseaudio.socket) is not running as a **system** service but as a **user** service and system services cannot access user services. Assumingly, this is the reason for the BrokenPipeError.

The solution is, therefore, to run the Picamera2 app as **user** service and not as **system** service:

- Stage the .service file under `~/.config/systemd/user`
- Remove the "User" and "Group" entries under the [Service] section
- Eventually add "Restart=no" to avoid automatic restart
- Then start the service with `systemctl --user start audio_video_capture.service`

I used the following `audio_video_capture.service` as user "sn":
```
[Unit]
Description=audio_video_capture
After=network.target

[Service]
ExecStart=/usr/bin/python3 /home/sn/prg/tesPicamera2/audio_video_capture.py
WorkingDirectory=/home/sn/prg/tesPicamera2/
Restart=no

[Install]
WantedBy=multi-user.target
```

User services are normally killed when the user session ends. However, the intended behavior that a service starts with system boot and does not require a user session, can also be achieved: see [systemd/User](https://wiki.archlinux.org/title/Systemd/User)

As said, this worked well under Bookworm on a Pi 5.

Trying the same on a Pi Zero 2 W under Bullseye led to the result that .mp4 files were created but these were corrupt or had  video/audio synchronization issues.
Possibly, fixing pulseaudio ([Audio Not Working? How To Enable Audio](https://www.vuepilot.com/support/article/fixing-raspberry-pi-audio-issues/)) might help. I'll try.

signag, 2024-01-19:
OK, I need to correct myself:
The issues with Bullseye systems, which I had observed on Pi Zero 2 W as well as Pi 4, were not synchronization issues.
Instead, audio recording continued after video recording had finished.

Looking at #902, I saw that such a behavior has also been discussed there.

I tested on a Pi 4 with Bullseye (Debian version 11.8)

Checking the picamera2 version with `apt list python3-picamera2` 
revealed `python3-picamera2/oldstable,now 0.3.12-2 all [installed]`

So, I checked out the **0_3_12_next** branch from this repository.

With PYTHONPATH pointing to this directory, `python audio_video_capture.service.py` stopped correctly after 10 sec.

For testing the user service as described previously, I added `Environment="PYTHONPATH=/home/sn/prg/picamera2"` to the [Service] section.
Starting the user service with this service definition resulted in a correct `test.mp4` with audio and video in sync.

The journal entry was
```
Jan 19 19:10:39 raspi03 systemd[718]: Started audio_video_capture.
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.501829645] [5662]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.538586462] [5663]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@>
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.538675943] [5663]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/l>
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.541068542] [5662]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.575798293] [5666]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@>
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.575887718] [5666]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/l>
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.583513437] [5662]  INFO Camera camera.cpp:1033 configuring streams: (0) 1280x720-XBGR8888>
Jan 19 19:10:39 raspi03 python3[5662]: [4:34:51.584113911] [5666]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selec>
Jan 19 19:10:40 raspi03 python3[5670]: Guessed Channel Layout for Input Stream #0.0 : stereo
Jan 19 19:10:51 raspi03 systemd[718]: audio_video_capture.service: Succeeded.
Jan 19 19:10:51 raspi03 systemd[718]: audio_video_capture.service: Consumed 5.313s CPU time.
```

signag, 2024-01-23:
[Here](https://github.com/signag/raspi-cam-srv#service-configuration-for-audio-support) is a step by step description for configuring a Picamera2 app as service to support audio recording along with video.    
The description includes variants for Bookworm as well as for Bullseye systems.    
It can be easily modified to handle own apps, different from raspiCamSrv for which it has been specified.