raspberrypi/picamera2 #723: [BUG]

zahigeni, 2023-06-25:
while using Picamera2 to take picture every 10s in loop from time to time pictures are half grey
**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.
running 
simple loop that runs:
`picam2.capture_file(image_location)`
every 10 seconds as part of my app 
**Console Output, Screenshots**
[image]
blur is by me the grey is from the camera

**Hardware :**
raspberrypi 3b
raspberrypi camera v2 
pi os lite64  6.1.21-v8

davidplowman, 2023-06-26:
Hi, thanks for the report. I must confess I've not seen anything quite like this before. We should probably start by checking a few basic things.

* Do you have any alternative cameras/ribbon cables/SD cards/Pis that you could swap, just to rule out any defective devices? Also, just to check, how long is the ribbon cable? (Though it doesn't look like a typical transmission error to me.)
* If you don't save the file, but merely view it, does the same effect occur? I ask because  it looks a bit like the JPEG encoder has given up, or truncated the file.

zahigeni, 2023-06-26:
hey,
I have a few pi's running, its happened to all of them, I'm using the ribbon that came with the pi camera v2, did not noticed it when trying only to view the images, I agree that is looks like jpeg compression issue, but there's no exception raised while this half grey image captured