raspberrypi/picamera2 #725: [HOW-TO] Is there any possibility to use the IMX178M sensor together with picamera2 ?

felixwanitschke, 2023-06-29:
Hello, 
I am currently using the Raspberry Pi HQ-camera with a Rasperry Pi 4b and picamera2, which works really nice. 

I wanted to compare the HQ camera to another sensor, the IMX178M (from veye; https://www.veye.cc/en/product/mv-mipi-imx178m/). Just plug and play did not work for that camera module and while reading the instructions from veye (https://wiki.veye.cc/index.php/Raspberry_Pi_Application_Note_Index), I read about V4L2....

Seemingly there should be some compatibility between libcamera (which is the basis of picamera2 ???) and V4L2 --> https://libcamera.org/docs.html

**So: Is there any possibility to use the IMX178M sensor together with picamera2 ?**

davidplowman, 2023-06-29:
Hi, the main requirement is an open source Linux driver for this sensor. If the vendor can supply one then it should be possible to make it work. A number of 3rd parties have successfully added support for the own sensors to the Raspberry Pi.

felixwanitschke, 2023-06-29:
Hello, thanks for the quick response.

Is it something like this: https://github.com/veyeimaging/raspberrypi_v4l2

If so, what are the steps I have to do, to make it work?

1. install the driver as described in the repository 
....?

Should it then already work or are the further steps to do. Excuse my incompetence...

Regards
Felix