raspberrypi/picamera2 #727: [BUG] calling "pprint.pprint(picam2.sensor_modes)" decreases requests throughput

tturpin, 2023-07-01:
Running the following simple code, I get a different number of processed frames per second by just commenting out the "pprint" line:
```
from picamera2 import Picamera2
import time
import pprint

picam2 = Picamera2()
config = picam2.create_video_configuration({"size": (1920, 1080)},controls={"FrameRate": 56.03})
pprint.pprint(picam2.sensor_modes)
picam2.align_configuration(config)
picam2.configure(config)
picam2.start()

last_time=time.time()
last_count=0
count=0
while True:
    t=time.time()
    count=count+1
    if t >= last_time + 1 :
        print(count-last_count)
        last_count=count
        last_time=t
    request = picam2.capture_request()
    request.release()

```
I have roughly 38FPS with the pprint line, and 56 without.

**Console Output**

With pprint, the output looks like:

> [0:03:47.068729903] [1534] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:299 [0mlibcamera v0.0.4+22-923f5d70
> [0:03:47.209151222] [1535] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:1476 [0mRegistered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media1
> [0:03:47.233792207] [1534] [1;32m INFO [1;37mCamera [1;34mcamera.cpp:1028 [0mconfiguring streams: (0) 640x480-XBGR8888 (1) 1536x864-SBGGR10_CSI2P
> [0:03:47.234324687] [1535] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:851 [0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
> [0:03:47.257666027] [1534] [1;32m INFO [1;37mCamera [1;34mcamera.cpp:1028 [0mconfiguring streams: (0) 640x480-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
> [0:03:47.258201732] [1535] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:851 [0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
> [0:03:47.287993801] [1534] [1;32m INFO [1;37mCamera [1;34mcamera.cpp:1028 [0mconfiguring streams: (0) 640x480-XBGR8888 (1) 4608x2592-SBGGR10_CSI2P
> [0:03:47.288526957] [1535] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:851 [0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
> [0:03:47.358557843] [1534] [1;32m INFO [1;37mCamera [1;34mcamera.cpp:1028 [0mconfiguring streams: (0) 1920x1080-XBGR8888
> [0:03:47.359133296] [1535] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:851 [0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
> [{'bit_depth': 10,
>   'crop_limits': (768, 432, 3072, 1728),
>   'exposure_limits': (9, 77193582, None),
>   'format': SRGGB10_CSI2P,
>   'fps': 120.13,
>   'size': (1536, 864),
>   'unpacked': 'SRGGB10'},
>  {'bit_depth': 10,
>   'crop_limits': (0, 0, 4608, 2592),
>   'exposure_limits': (13, 112015443, None),
>   'format': SRGGB10_CSI2P,
>   'fps': 56.03,
>   'size': (2304, 1296),
>   'unpacked': 'SRGGB10'},
>  {'bit_depth': 10,
>   'crop_limits': (0, 0, 4608, 2592),
>   'exposure_limits': (26, 220417486, None),
>   'format': SRGGB10_CSI2P,
>   'fps': 14.35,
>   'size': (4608, 2592),
>   'unpacked': 'SRGGB10'}]
> 32
> 38
> 37
> 38
> 38
> 38
> 38
> 38
> 38
> 38
> 38
> 38
> 

and without pprint:

> [0:04:45.635203098] [1555] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:299 [0mlibcamera v0.0.4+22-923f5d70
> [0:04:45.769170307] [1556] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:1476 [0mRegistered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media1
> [0:04:45.789984038] [1555] [1;32m INFO [1;37mCamera [1;34mcamera.cpp:1028 [0mconfiguring streams: (0) 1920x1080-XBGR8888
> [0:04:45.790584875] [1556] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:851 [0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
> 49
> 56
> 57
> 57
> 57
> 57
> 56
> 57
> 57
> 57
> 56
> 56
> 56
> 56
> 56
> 56
> 56
> 57
> 

**To Reproduce**
Install picamera2, save the above code in a file and run python on it. I ran the example in console mode, after stopping lightdm to improve repeatability.

**Expected behaviour**
Clearly, querying the supported modes is not supposed to impact performance !

**Hardware :**
 Raspberry Pi 3B, Pi camera module 3 NoIR

**Software :**
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
python version: 3.9.2
libcamera version: v0.0.4+22-923f5d70
picamera2 version: 0.3.9-1

**Additional context**
I encountered this issue while experimenting with high framerate, high resolution processing.

davidplowman, 2023-07-03:
Thanks for reporting this. Indeed I'm seeing the same thing on a Pi 4 though it's more intermittent, i.e. sometimes it outputs numbers around 55, and then for a bit you get numbers near 38. I found I could make it go away by adding `raw={'size': (2304, 1296)}` into the call to `create_video_configuration()`, so it's evidently related in some way to the use of raw image buffers with the sensor. So I guess that's a reasonable workaround for now, but I have to confess I'm a bit baffled as to why.

tturpin, 2023-07-03:
Thanks for looking into this.
I tried your workaround, and when I specify `raw={'size': (2304, 1296)}` in `create_video_configuration()` The framerate is consistently on the low side: 38 FPS independently of whether I print the sensor_modes or not. In a sense, this makes things more deterministic, but from the performance point of view this is the contrary of what I want...

tturpin, 2023-07-03:
Additional information: I just tested on the 64-bit OS and I get exactly the same issue with the same numbers (38 or 56FPS).  I checked because I have found that the 64-bit version improves the performance when I start doing something with the request.

davidplowman, 2023-07-03:
All very confusing. Another workaround to try might be this:
```
...
with Picamera2() as picam2:
    sensor_modes = picam2.sensor_modes
picam2 = Picamera2()
# You can still use the sensor_modes from earlier.
...
```
So basically get the list of sensor modes and throw away the Picamera2 object immediately. I will try and get round to investigating this further at some point.

tturpin, 2023-07-03:
This one works !