raspberrypi/picamera2 #728: [HOW-TO] increase fps and rsolution of mjpeg_server_2.p

KruemelmonsterK, 2023-07-01:
Hey guys, 

I am using a code example that is provided by you guys. Its called mjpeg_server_2.py.

I am taking the frames from the server an create videos with a length of 10 seconds. This is the code, which I use to create the videos from stream:

import subprocess as sp
import os

#mjpeg
command = ['ffmpeg',
           '-f', 'mjpeg',  # Specify input format
           '-i', 'http://169.254.102.139:8000/stream.mjpg',  # Replace with your stream URL
           '-c:v', 'copy',  # Copy codec settings from input
           #'-r', '50',  # Set the output frame rate to 50 fps
           #'-s', '2304x1296',  # Set the output resolution to 2304x1296
           '-f', 'segment',  # Segment output
           '-segment_time', '10',  # Duration of each segment
           '-reset_timestamps', '1',  # Reset timestamps at the start of each segment
           '-strftime', '1',  # Enable strftime
           'C:/Users/gewis/Downloads/pythonProject/videos/%Y_%m_%d_%H_%M_%S.mp4']  # Output file name pattern

pipe = sp.Popen(command)

# Keep the program running
try:
    while True:
        pass
except KeyboardInterrupt:
    pipe.terminate()

The problem is, that no matter what I try, the video and stream don't have the input that I specify on the server.

When I set in the config file the fps to 50 and the resolution to 2304 × 1296 the generated videos simply dont have these specs. They always have 25 fps and 1920x1080. The same happens, when I take the H264 Encoder. The only thing that changes, is that the fps are 30 and not 25 anymore.

Vincent-Stragier, 2023-07-03:
Hi @KruemelmonsterK,

Sorry, but can you edit your previous comment using a code block like so:

``````````
```python
Your code with the correct indentations and syntaxt highlighting.
``` 
``````````

To reply to your question, I think you need to set the desired frame rate to the camera stream before starting the capture (you can also change it later on, but that's not what you are asking for). In the server code, you would have to change:

https://github.com/raspberrypi/picamera2/blob/dd984341d3fda13f1c08c41dc93e25625853bcfe/examples/mjpeg_server_2.py#L84-L87

To something like:
```python
picam2 = Picamera2()
# Select the framerate
camera_stream_framerate = 50

def make_frame_duration_limits(framerate: int = 50):
    """Return the frame duration limits for the given framerate.

    Args:
        framerate (int, optional): The framerate. Defaults to 50.
    
    Returns:
        tuple: The frame duration limits.
    """
    return (1000000 // framerate, 1000000 // framerate)

camera_frame_duration_limits = make_frame_duration_limits(
    camera_stream_framerate)

# Appy video configuration hereunder
picam2.configure(picam2.create_video_configuration(
    main={"size": (640, 480)}),
    controls={"FrameDurationLimits": camera_frame_duration_limits}
)
output = StreamingOutput()
picam2.start_recording(MJPEGEncoder(), FileOutput(output)) 
```

davidplowman, 2023-07-03:
Hi, thanks for the question, though I'm not sure I understand everything!

So mjpeg_server_2.py creates a web page containing an mjpeg video. The video is 640x480 and runs at 30fps. Have you changed this file?

I didn't know you could feed an HTML page straight to FFmpeg, does that actually work? I'm wondering what happens to the HTML headers and page boundaries! 

Or is this more a question about what FFmpeg is doing? Clearly nothing you do in the ffmpeg command has any effect on the camera system that is running remotely.  I'm sorry not to understand better, perhaps you could fill in a few more details? A very minimal self-contained example is often helpful. Thanks!