raspberrypi/picamera2 #735: Best settings for Pi Camera V3 Module noIR for daylight/indoor capture

NewAmazingPVP, 2023-07-04:
Hello there, I tried adjusting AWB mode setting to get rid of the purple tint in noIR camera photographs and captures but had no luck. Any suggestions on what settings I should change to make my captures look as similar to what human eyes see?

davidplowman, 2023-07-05:
Hi, there should be an imx708_noir.json tuning file that you can use, something like
```
tuning = Picamera2.load_tuning_file("imx708_noir.json")

picam2 = Picamera2(tuning=tuning)
```
Does that make a difference?

NewAmazingPVP, 2023-07-05:
No, sadly, still doesn't make a difference; the images are still purple.

NewAmazingPVP, 2023-07-05:
This is my initializing code 

``python
tuning = Picamera2.load_tuning_file("imx708_noir.json")
picam2 = Picamera2(tuning=tuning)
picam2.configure(picam2.create_preview_configuration(main={"format": 'XRGB8888', "size": (1920, 1080)}))
picam2.start()
picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})
picam2.set_controls({"AwbMode": controls.AwbModeEnum.Indoor})
picam2.set_controls({"AeConstraintMode": controls.AeConstraintModeEnum.Normal})
picam2.set_controls({"AeExposureMode": controls.AeExposureModeEnum.Normal})
picam2.set_controls({"AeMeteringMode": controls.AeMeteringModeEnum.Matrix})
``

AyrtonRicardo, 2023-07-06:
@NewAmazingPVP check one of the directories below to find different filters, in my case I have a ov5647 sensor, so I applied this filter and it removed the purple color.

- /home/pi/libcamera/src/ipa/raspberrypi/data
- /usr/local/share/libcamera/ipa/raspberrypi
- /usr/share/libcamera/ipa/raspberrypi

Check via `ls -lah <directory>` inside your pi, and use one filter from there to see if any helps you.

NewAmazingPVP, 2023-07-06:
Hey there, I did find 4 files for imx708 sensor (imx708_wide_noir.json, imx708_wide.json, imx708_noir.json, imx708.json) in /usr/share/libcamera/ipa/raspberrypi and used all of them, but none helped. When I used imx708.json and imx708_wide.json my captures came out pink and when I tried to use the noIR ones (imx708_wide_noir.json and, imx708_noir.json) my images came out with purple color. Any suggestions what I can do? Are there any more sensor files that I'm missing.

davidplowman, 2023-07-10:
I don't really understand what's going on here. When you use the "noir" version of the tuning files it runs simple greyworld AWB, so in theory there should be no way you can get purple images. Is it possible for you to capture a DNG file that shows the problem and include a link to it here? Thanks.

NewAmazingPVP, 2023-07-11:
Ok, thanks for the help. Here is the dng file: https://www.dropbox.com/s/b6lq0kht9hb7d70/full.dng?dl=0 I took this at night; the purple tint only shows up during the daytime. Does that make a difference or not? If so, I can take the picture in daylight. Sorry, I have no experience with this lol.

davidplowman, 2023-07-11:
That file looks perfectly fine to me. It's pretty grey which is what I'd expect. The "as shot neutral" field in the EXIF data reports 1.02817191 1 0.4331254331 suggesting that the red/blue colour gains are about 0.97 and 2.31. These are nowhere near the normal values for a regular sensor (with the IR filter), suggesting that the right things are happening.

I'd be interested to see a DNG that shows the purple problem. It might be worth looking what the colour gains are (`picam2.capture_metadata()['ColourGains']` while the preview is running). Are you able to set the colour gains to get the kind of image you expect (`picam2.set_controls({'ColourGains': (1.0, 2.3)})` for example)?

But overall, I'm pretty baffled by all this.

davidplowman, 2023-07-11:
Thanks for the picture. Mostly this looks in line with my expectations. Most of the image is slightly green, and a couple of things slightly purple, but taken as a whole it looks pretty much grey "on average" - which is what it tries to achieve for NoIR cameras.

The problem with IR is that it's very difficult to calibrate for it. The red, green and blue pixels on the sensor are all sensitive to it, and the amount of IR will vary with the illumination, type of glass in the windows, whether you're using any IR lamps and so on. You may find that the IR is swamping everything a bit, so things will look a bit faded and strange whatever you do.

You should be able to adjust those colour gains to achieve the "best possible result", but even this may not give you true to life colours. For reference, the colour gains for this image look to be about (0.8, 1.4), so that would be a place from which to start adjusting them. One reasonable approach would be to find something you know to be grey and adjust the gains until it is indeed grey. Out of interest, what colour should the walls be?

NewAmazingPVP, 2023-07-11:
Oh, I understand now about the challenges with IR callibration. Thank you for explaining. The walls are white, so they should look and be white. I will definitely experiment with different color gains to see if I can achieve the image I want.

NewAmazingPVP, 2023-07-12:
I will close the issue now. Once again, thank you for your guys help and time. I really appreciate it.

closed 2023-07-12 by NewAmazingPVP