raspberrypi/picamera2 #736: [SUGGESTION]

Naderschah, 2023-07-05:
**Please only include one suggestion or features request per issue!**
Inclusion of image capturing in dtype uint10 or uint12
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.
My camera (arducam B0333) only supports uint10 or 12 output, im not sure if your code accounts for it, I did not find a mention of correcting the buffer stream for this, but may very well be very wrong here. But the main reason I want this is to increase the sensitivity of my device as I made a spectroscope out of it and only noticed after calibration that all saved data is uint8 not 10 or 12.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to have the option to specify the requested and returned dtype, for my particular interest this would only involve having some extra variable that can be passed down into request.py to change the datatype the stream accepts, and changing the dtype the buffer returns, allhough I wouldnt know what changes need to be made here for this. 
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I could just use the libcamera implementation, but I already wrote to much code, so im not going to do that, I will just stick with uint8 and change my processing to use uint8 exclusively
**Additional context**
Add any other context or screenshots about the feature request here.

davidplowman, 2023-07-05:
Hi, thanks for the question. You should be able to request a 10 or 12 bit mode by asking for a raw stream with a 10 or 12 bit format. For example, try asking for a raw stream with format `SBGGR12`. This should give you unpacked 12-bit raw output. Convert it to actual 16-bit samples using `numpy.view`.

Naderschah, 2023-07-06:
Ah thank you,
I completely fergot about this feature!

Naderschah, 2023-07-06:
Hey
I have been trying to configure it in an interactive session here is the output
>>> from picamera2 import Picamera2
>>> cam = Picamera2()
[0:06:28.168762349] [723]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4226-8689dd6b
[0:06:28.290708104] [724]  WARN CameraSensorProperties camera_sensor_properties.cpp:243 No static properties available for 'arducam-pivariety'
[0:06:28.291891072] [724]  WARN CameraSensorProperties camera_sensor_properties.cpp:245 Please consider updating the camera sensor properties database
[0:06:28.342702720] [724] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'arducam-pivariety.json' not found for IPA module 'raspberrypi'
[0:06:28.697265511] [724]  INFO RPI raspberrypi.cpp:1485 Registered camera /base/soc/i2c0mux/i2c@1/arducam_pivariety@c to Unicam device /dev/media3 and ISP device /dev/media0
>>> still = cam.create_still_configuration(raw={'size':(1920, 1080), 'format':'SRGGB10'})
>>> cam.configure(still)
[0:06:39.680762719] [723]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-BGR888 (1) 1920x1080-SRGGB10
[0:06:39.782021821] [724]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/arducam_pivariety@c - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected unicam format: 1920x1080-RG10
>>> cam.start()
>>> r = cam.capture_request()
>>> r.make_array('raw')
array([[ 75,   0,  77, ...,   1, 157,   0],
       [ 75,   0,  72, ...,   0, 129,   0],
       [ 63,   0,  70, ...,   1, 156,   0],
       ...,
       [ 67,   0,  67, ...,   0,  65,   0],
       [ 70,   0,  72, ...,   0,  72,   0],
       [ 72,   0,  65, ...,   0,  56,   0]], dtype=uint8)
>>> r.release()
>>> arr = cam.capture_array(raw)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'raw' is not defined
>>> arr = cam.capture_array('raw')
>>> arr
array([[ 72,   0,  68, ...,   1, 179,   0],
       [ 79,   0,  66, ...,   0,  98,   0],
       [ 69,   0,  63, ...,   1, 180,   0],
       ...,
       [ 64,   0,  69, ...,   0,  65,   0],
       [ 71,   0,  73, ...,   0,  74,   0],
       [ 65,   0,  65, ...,   0,  67,   0]], dtype=uint8)

The sensor modes available are
[{'format': SRGGB10_CSI2P, 'unpacked': 'SRGGB10', 'bit_depth': 10, 'size': (1920, 1080), 'fps': 60.0, 'crop_limits': (0, 0, 1920, 1080), 'exposure_limits': (14, 15534385, None)}]

I tried the above also with format SRGGB10_CSI2P

If i understand correctly all methods to capture images will at some point use the request method make_buffer which only returns uint8 

I cant use the Helpers capture buffer method as SRGGB10 is not supported for PIL

And I believe this exhausts the underlying methods for capturing implemented

Could you help me out here as now im also given to thing that the raw stream supplied is uint10 but I have been saving uint8 images and im not sure how that modified the data as request.save_dng also uses data returned from request.make_buffer which always returns uint8 regardless of bit_depth

Naderschah, 2023-07-06:
Hey,
So regardless of how I capture the image at some point it will be scaled to uint8 and then i can just upscale to uint10.
Is this correct?
Because what I want is to get the returned uint10 image and not to downscale it as I will loose precision by downscaling and then upscaling again.
If the above is correct I could request SBGGR10_CSI2P and then unpack the bytes manually to be uint10? Or is the returning stream already uint8?

davidplowman, 2023-07-06:
Not quite. If you capture a raw 10-bit image, the first two bytes might (for example) be 10 and 2. This represents the 10-bit value 2 * 256 + 10. Using `np.view` will do this conversion automatically for you.

Naderschah, 2023-07-06:
Alright, so turns out uint10 doesnt exist and seems to be a pain to make a workaround with numpy, I will attempt this at some later time. 
So when I get the np.uint8 image, does the array simply interpret the bytestream as uint8 or is the bytestream converted from uint10 to 8 and then read by numpy?

davidplowman, 2023-07-06:
The raw image is, at the most basic level, just a buffer of bytes. It is exactly how the camera peripheral deposits the bytestream from the camera in memory. The camera peripheral can write it to memory either in its packed for ("CSI2P") which is a bit more efficient, but hard to use for other purposes. It can also unpack the data and write it straight to memory where each 10-bit pixel occupies 2 bytes. I don't know if that's any clearer...

`np.view` is useful here, because it gives you a "uint16" view of the same buffer without copying it.

Naderschah, 2023-07-06:
Oh ok, I think I start to understand. Sorry this is taking so long.
But if you could just answer one more question, the reason I didnt attempt uint16 is that I assume that 16 bytes will be then read per value so that if i have a uint 10 returned and pretend its encoded like such 1010101000 the uint 16 would read the 10 bytes but also the first 6 of the following number so that the array will be the wrong sice and the datapoints loose all meaning.
From your last reply I take that this is not how it works, but it understands to read it as uint10 and then simply zero pads the beginning 6 bits?

davidplowman, 2023-07-06:
Correct. If you use an unpacked raw format, than those 6 bits in each 16-bit word (4 bits if it's 12-bit raw) are filled with zeros. (The packed format gives you 4 bytes where each contains the 8 MSBs of a pixel, and then the 5th byte is the LSBs of those pixels all packed together. My advice is to use the unpacked formats!!)

Naderschah, 2023-07-06:
Ah beautiful,
Thank you for all your help, i understand now how to work with this
Now if you dont mind one more question arose for me, if i were to end up using just the 8bit representation, does this 8 bit representation then correspond to the 10 bit values read as 8bits so that the max 255 value is reached earlier relative to digital count or is it that the uint10 is scaled to uint8 by numpy automatically?

davidplowman, 2023-07-06:
Not really. It's like I said earlier - you get the buffer as 8-bit values, so the first bytes (for example) might be 10 and 2. This is actually the 10-bit value 2 * 256 + 10. It's probably worth getting your raw array (as 8 bits) and comparing it with what you get after `np.view`. You should find the conversion done "magically" for you. You'll also notice that the "shape" now has half the width.

Naderschah, 2023-07-06:
Alright, thank you very much for all the information.
I think I understand now.