raspberrypi/picamera2 #737: [HOW-TO] Customize the FPS of a recorded video using H264Encoder

azsde, 2023-07-05:
Hello,

I am trying to record a video at the highest framerate supported by my camera V3 which is supposed to be capable of doing 1080p@50FPS according to the documentation.

But the FPS I'm trying to apply doesn't seem to work and I always get an output file that is 30FPS:

[image]

Here is my code:

```
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder, Quality

from libcamera import controls

picam2 = Picamera2()
picam2.video_configuration.controls.FrameRate = 50.0
video_config = picam2.create_video_configuration(main={"size": (1920, 1080)})
picam2.configure(video_config)
picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})
h264encoder = H264Encoder()
picam2.start_recording(h264encoder, "test.h264", quality=Quality.VERY_HIGH)
sleep(10)
picam2.stop_recording()
```

What am I doing wrong ?

I thank you in advance for your help.

Regards,

Azsde.

davidplowman, 2023-07-05:
Hi, please try
```
video_config = picam2.create_video_configuration(main={"size": (1920, 1080)}, controls={'FrameRate': 50})
```
(and you probably don't need the line above.) Does that make a difference?

azsde, 2023-07-10:
It doesn't seem to affect the reported framerate in MediaInfo (which is still at 30.000 FPS).

azsde, 2023-07-10:
Ok, so now the .h264 file correctly is reporting 50 FPS:

[image]

But either when using MP4Box or ffmpeg to make it into a playable .mp4 file, the duration is not correct and the footage is sped up (it should be around 10s and is recognized to be 3 seconds long by VLC).

Am I missing something here ?

davidplowman, 2023-07-10:
Maybe try saving a timestamp file alongside the h.264 file so that we can check the video really is running at 50fps? (Add `pts="timestamps.txt"` to the `start_recording()` call.)

davidplowman, 2023-07-10:
Looking at the timestamp file (which contains frame timestamps in milliseconds), the camera does seem to be delivering frames every 20ms (actually very marginally less). But there are lots of jumps when a whole second or so of frames go missing. So it seems to be that the problem is that there are large numbers of frame drops. What kind of Pi is this?

I would expect the correct duration if you give the timestamp file to mkverge, only there will presumably be numerous small pauses where the missing frames should be.

azsde, 2023-07-10:
This a Raspberry Pi 3b+, is it too weak to handle such a video stream ?

tvoverbeek, 2023-07-10:
Isn't it just a memory issue? 3B+ has 'only' 1GB. Pi4 probably has  more (2, 4, or 8GB).
Try the Pi4. My guess is it will behave better.

azsde, 2023-07-10:
I just tried with a Pi 4 8Gb, I still have issues to "convert" the .h264 file to a .mp4 file using ffmpeg / mp4box, the output videos are indeed 60FPS but they play way too fast.

When using mkvmerge, the playback speed is correct, however the output file is not 60 FPS but rather 38ish FPS.

davidplowman, 2023-07-11:
One thing you have to do at high framerates is turn off the extra denoise processing, which is relatively slow (the `libcamera-vid` command you were using does this). In Python, add `'NoiseReductionMode': 1` to the controls when you create the configuration.

I would say the first thing to check is that `libcamera-vid` will run and give you the correct framerate reliably with no drops, judged by checking the timestamp file. Also be sure to run without a preview window. Then let's move on to Python. Maybe use format "YUV420", and perhaps start with lower resolutions/bitrates to begin with. Anyway, so I think your configuration would be something like this:
```
video_config = picam2.create_video_configuration(main={"size": (1920, 1080), 'format': 'YUV420'},
                                                 controls={'FrameRate': 50, 'NoiseReductionMode': 1})
```

azsde, 2023-08-01:
Hello,

I had time to work on the subject again,the issue doesn't seem to come from picamera as libcamera-vid produces the same result:

`libcamera-vid --level 4.2 --framerate 50 --width 1920 --height 1080 --save-pts timestamp-libcam.pts -o video-libcam.264 -t 60000 --denoise cdn_off -n`

Using MediaInfo on the "raw" h264 file shows that the produced file has the expected FPS.

[image]

However, when trying to play the h264 file using VLC, the output video doesn't have any duration and play faster than it should (like if the video was sped up).

"Converting" the h264 file using ffmpeg also result in a 50FPS video, but with an incorrect duration and it play faster that it should as well.

"Converting" the h264 file using mvkmerge result in a 32ish FPS video with an correct duration and it plays at a normal rate.

Anyway, I'll try to get support on the Raspberry Pi forums as it is not an issue related to picamera2.

kodonnell, 2024-01-19:
I'd try recording at a much lower resolution and/or quality. If you get the FPS you're expecting with that, then it's likely a performance issue, which seems quite possible. If not, then you can dig deeper.

Generating a timestamps.txt is about the only reliable way I've found to validate the framerate. Each represents the timestamp for each frame, so you can figure out the FPS. (As in your example above, they're 20ms apart, apart from the frame drops, which timestamps.txt is also useful for detecting.)

davidplowman, 2024-01-19:
Remember that ".h264" is not a proper file format. It contains video frames but without timestamps, so a player can choose to play it at any rate that it wants. Also note that for the last few years, vlc has been unable to play streams without timestamps properly. It skips stuff, produces garbled output, all kinds of nonsense happens. If you want to play ".h264" files, I recommend a player other than vlc (e.g. ffplay).

PythoErgo, 2025-04-15:
Hello there, I'm having the exact same problem as @azsde, here's my simple testing code:
`
from picamera2 import Picamera2, encoders
from picamera2.encoders import H264Encoder
import signal
import sys

picam2 = Picamera2()

# Define 480p at 48fps
cfg = {"size": (640, 480), "fps": 48}
filename = "480p48_test.h264"

# Timeout handler
def timeout_handler(signum, frame):
    print("\n[INFO] Timeout reached. Stopping recording...")
    picam2.stop_recording()
    print("[INFO] Recording complete. Saved as 480p48_test.h264")
    sys.exit(0)

# Set the timeout signal
signal.signal(signal.SIGALRM, timeout_handler)
signal.alarm(10)  # 10-second timeout

# Create and apply configuration
video_config = picam2.create_video_configuration(
    main={"size": cfg["size"], "format": "YUV420"}
)
picam2.configure(video_config)
picam2.set_controls({"FrameRate": cfg["fps"]})

# Encoder with SPS frame rate signaling
h264encoder = H264Encoder(framerate=cfg["fps"], enable_sps_framerate=True)

# Begin recording
print("[INFO] Recording for 10 seconds at 480p 48fps...")
picam2.start()
picam2.start_recording(h264encoder, filename)

# Wait indefinitely; timeout handler will trigger exit
signal.pause()
`

Very basic, common code here. When trying to record for 10 seconds I checked the actual duration using ffmpeg, which says the video is 343 frames and 7.12 seconds long (343/7.12=48.17fps, correct framerate but wrong duration). I've tried two other similar methods like `picam2.video_configuration.controls.FrameRate = 48.0` and `picam2.set_controls({"FrameRate": 48})`, both of which do not actually set the fps to 48 but instead to the default 30fps.

A quick workaround to this would be to use subProcess and simply run the console command (libcamera-vid) with parameters from input(), but I'm trying to make one big Python script for some smart glasses (needs code for display, Bluetooth, video recording) I am making, and I'm not sure if that would work.

davidplowman, 2025-04-15:
@PythoErgo Thanks for the question. I would certainly agree that running `rpicam-vid` in a sub-process sounds much clunkier and gives you less control, so I would definitely try and avoid that.

I might start by confirming a couple of basic things:

* What framerate are you really getting? Maybe ignore the recording for now and try this:
```
# ...
picam2.start()
picam2.capture_metadata()  # don't time very first frame, it can be a bit slow

count = 0
start_time = time.monotonic()
while time.monotonic() - start_time < 10:
    picam2.capture_metadata()
    count += 1

print(count / 10, "fps")
```

* Check how many frames the encoder is seeing. When the recording finishes, I think you can look at `encoder.frames_encoded`.

You might also want to set the framerate as part of the configuration, like so:
```
video_config = picam2.create_video_configuration(
    main={"size": cfg["size"], "format": "YUV420"},
    controls={'FrameRate': cfg["fps"]}
)
```
as this will take effect as soon as the camera starts (setting it later will cause it to wait a few frames before it changes).

davidplowman, 2025-04-15:
Hi again, you don't say what kind of a Pi and what kind of a camera you have. Pasting the output of `rpicam-hello --list-cameras` would be helpful, just to see what camera modes you have available.

If you have a non-Pi 5, then it may well be worth adding `"NoiseReductionMode": 3` to the controls that you set at the start. Really, you want that reporting the expected framerate before you do anything else. Might be worth printing out `picam2.camera_configuration()` once you've configured the camera as well, just to check what mode it selected.

Under some circumstances, increasing the number of buffers in the camera queue can help (add `buffer_count=16` when you create the configuration).

In the final example, I suspect that suspending/waking the main Python thread 100 times per second may not be helping, but in any case, let's make the other example work first.

(PS. for a block of code, use a line with just three  consecutive backquotes to start/end it)

PythoErgo, 2025-04-15:
I am using a Raspberry Pi Zero W. After running `rpicam-hello --list-cameras` I got this:
[image]
As for your other suggestion I simply added `print(picam2.camera_configuration())` before starting the camera (not sure if this is wrong or not) and got this:
[image]

davidplowman, 2025-04-15:
Thanks for that. The camera mode selection and everything looks fine. So I'd make that noise reduction and buffer_count change that I suggested, and then see what fps you get running the first test case (so no recording yet).

PythoErgo, 2025-04-15:
So something like this?
'''
video_config = picam2.create_video_configuration(
    main={"size": cfg["size"], "format": "YUV420"},
    buffer_count=16
)
picam2.configure(video_config)
picam2.set_controls({"FrameRate": cfg["fps"], "NoiseReductionMode": 3})
'''

davidplowman, 2025-04-15:
Hmm, that doesn't seem great. Maybe I need to try and find a Zero W to try it. I tend not to have them lying around because they are pretty slow at running Python code, but I'll have a look round.

Just to check - you're running Raspberry Pi OS Lite (no desktop), Bookworm 32-bit, right?

PythoErgo, 2025-04-15:
I am running Raspberry Pi OS Desktop 32bit, but I'm on console boot right now and CPU, memory, etc levels are healthy. With desktop boot I'd have zero chance of video recording, haha.

PythoErgo, 2025-04-24:
Well, I just got the Raspberry Pi Zero 2 W. Question still remains, how to control FPS in Python? If the mystery is still unsolved then I will have to simply work with what I'm given.

davidplowman, 2025-04-24:
I ran the following on my Pi Zero 2 and v1 camera
```
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import Output
import time

picam2 = Picamera2()
config = picam2.create_video_configuration(
    {'size': (640, 480), 'format': 'YUV420'},
    controls={'FrameRate': 60, 'NoiseReductionMode': 3}
)
picam2.configure(config)
encoder = H264Encoder()
output = Output()

picam2.start_recording(encoder, output)
time.sleep(10)
picam2.stop_recording()

print("Frames encoded:", encoder.frames_encoded)
```
and it reports `Frames encoded: 614` when it finishes.

PythoErgo, 2025-04-24:
Will try AS SOON AS I CAN. (this comment will be deleted after trying it)

PythoErgo, 2025-04-24:
Just to confirm, you're using this camera module right?
[image]

I got 319 frames with your exact code.  I'm using the Pi Zero W though, want me to try switching to the Pi Zero 2 W?

PythoErgo, 2025-05-01:
Sorry for the late reply, I've been working on other parts of my project that isn't the camera. Well I reran your code on my new Raspberry Pi Zero 2 W, and got 614 frames! Success! One more thing though, I must check whether the footage looks sped up or not. If I record for 10 seconds the video shouldn't be shorter than that like before.

davidplowman, 2025-05-01:
Glad that it's running better now! The original Pi Zero is pretty old and slow by today's standards, so hopefully the Zero 2 will be much less painful.

Don't forget that if you record H.264 frames straight to a flat file, then it will contain no timestamps - so the rate at which it gets played back can be rather variable, depending on the software. And VLC doesn't play H.264 files properly at all. You may have more luck recording to something like an mp4 file which will have correct timestamps (use `PyavOutput("test.mp4")`).

PythoErgo, 2025-05-16:
Hello David. I have returned, once again...

I've gone down some rabbit holes, looked at the Picamera2 docs many times and am very confused. Here's some code that is supposed to record both video and audio from a MEMS microphone and mux them together to create an MP4 file after I'm done recording:

```
audio_process = None
video_filename = "video.h264"
audio_filename = "audio.wav"
final_filename = ""

def start_video(config):
    global is_recording, picam, audio_process, final_filename

    try:
        size = config["size"]
        framerate = config["framerate"]

        print(f"Starting video at {size} @ {framerate}fps.")
        picam.stop()

        # Proper video configuration per your spec
        main = {"size": size, "format": "YUV420"}
        controls = {"FrameRate": float(framerate)}
        video_config = picam.create_video_configuration(main, controls=controls)

        picam.configure(video_config)

        encoder = H264Encoder()
        picam.start_recording(encoder, video_filename)
        is_recording = True

        # Start audio recording
        audio_process = subprocess.Popen([
            "arecord",
            "-D", "dmic_sv",
            "-c1", "-r", "16000", "-f", "S16_LE",
            "-t", "wav", audio_filename
        ])

        # Display status
        img = Image.new("RGB", (160, 80), "black")
        draw = ImageDraw.Draw(img)
        status_msg = "Recording..."
        msg_width = draw.textlength(status_msg, font=font)
        draw.text(((160 - msg_width) // 2, 35), status_msg, font=font, fill="red")
        display.display(img)

    except Exception as e:
        print(f"Video start failed: {e}")

def stop_video():
    global audio_process, final_filename

    try:
        picam.stop_recording()
        timestamp = time.strftime("%Y%m%d_%H%M%S")
        final_filename = f"/home/PythoErgo/Videos/recording_{timestamp}.mp4"

        if audio_process:
            audio_process.terminate()
            audio_process.wait()

        # Rotate 90° clockwise at mux time
        subprocess.run([
            "ffmpeg", "-y",
            "-i", video_filename,
            "-i", audio_filename,
            "-vf", "transpose=1",  # Clockwise rotation
            "-c:v", "libx264",
            "-c:a", "aac",
            "-strict", "experimental",
            final_filename
        ])

        print(f"Recording saved as {final_filename}.")

    except Exception as e:
        print(f"Video stop failed: {e}")
```

I can't get the framerate parameter to listen at all, it always defaults to 30fps and I haven't found a solution yet. I'm temporarily using my Raspberry Pi Zero W instead of the 2 W, because my camera module, display, microphone, and other things are already wired to it for testing, so I just want to test 640x480 and 10fps for now, just to see if I can get audio input on a video recording. I haven't gotten any sound to appear yet, but I'm quite sure that's because the video framerate is all messed up, so it can't mux properly.

Although it seemed like your frames-encoded counter works, trying to control framerate on a video still doesn't work I guess? I haven't figured it out? The footage is sped up for roughly 3 seconds and then it just freezes, maybe it freezes at the same time I start speaking into the microphone, I don't know!!

[image]
[image]
[image]

PythoErgo, 2025-06-16:
> > [@davidplowman](https://github.com/davidplowman) I have made a big breakthrough on all framerate issues with the PiCamera2 library, would you mind adding me on Discord (pythoergo) or emailing me ([PythoErgo@gmail.com](mailto:PythoErgo@gmail.com))? That would be great. In my working code, I had to specifically set the framerate in 4 different areas of the program, but I'm sure less are truly required. I get the right metadata, right FPS, right speed, everything works flawlessly. Maybe you could add my ideas into the documentation.
> 
> I'd love to get your inputs on this too.

Alright, I will just share my solution here as soon as I can.

PythoErgo, 2025-06-16:
I've tried setting the framerate in the FFMPEG config before, but it never worked right, since the H264 file was roughly always 30fps, the final MP4 would always be sped up or slowed down if I wasn't already recording at 30FPS, in that case it worked fine. For a long time, I added -r and -framerate parameters into the FFMPEG command before my discovery, after which I realized it didn't really need it if I set the framerate in the encoder's parameters.

By the way, I have no idea what PyavOutput is, pretty embarrassing I know, haha. Maybe you could explain that? Is it simpler than what I did? Also, I think my solution would work fine if the video output was MP4 in the first place, I just heard that recording H264 format is faster and easier on the Raspberry Pi and camera.

davidplowman, 2025-06-16:
I guess I don't have anything specific - but certainly ping me again if you do!