raspberrypi/picamera2 #738: [BUG] Multi-encoder recording issue

azsde, 2023-07-05:
Hello,

When using two encoders and triggering a recording,when stopping the recording it will stop every encoders:

```
picam2 = Picamera2()
video_config = picam2.create_video_configuration()
picam2.configure(video_config)

mjpegEncoder = MJPEGEncoder()
h264encoder = H264Encoder()

picam2.start_encoder(mjpegEncoder)
picam2.start()

picam2.start_recording(h264encoder, "output.h264", quality=Quality.VERY_HIGH)
sleep(10)
picam2.stop_recording()

# From here, mjpegEncoder is not usable anymore as it has been stopped when calling picam2.stop_recording().
```

davidplowman, 2023-07-05:
Hi, I think it's intentional that `stop_recording()` stops everything. I would probably create the outputs explicitly and then you can stop one and not the other. Maybe section 9.3 of [the manual ](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)might help?

azsde, 2023-07-05:
Hi,

Thank you for your answer.

I based my code on the examples, which seem to have a different approach than the one in the section 9.3.

Your advice would be to use the outputs directly then ?

RedBlaze42, 2023-07-11:
Hi,
You can stop only one encoder by using
`picam2.stop_encoder(encoders=mjpegEncoder)`
or
`picam2.stop_encoder(encoders=h264encoder)`
Depending on what you want

closed 2023-08-01 by azsde