raspberrypi/picamera2 #739: [HOW-TO] Remote shutter release

Naderschah, 2023-07-07:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
A clear and concise description of what you want to happen.

Im using raspberry camera boards at the moment to make night time images of the sky, im aware that the HQ and arducam low light cameras are well suited for this, however, I would like to make it work on the other (cheaper) boards as well. From my understanding the main problem with the other camera boards is the internal clock being not capable of storing longer times, for example I have a waveshare RPi Camera B (SKU: 8193) that is limited at somewhere between 1-2 seconds exposure. So I would like to know if it is possible to do something equivalent to bulb mode on DSLR cameras, where one times the exposure not on the camera board but from the raspberry pi. 

** alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I could add images together, but with the Rpi Camera B for example 1 second is not sufficient to go past the noise levels at night so it is not really possible. I can always buy other cameras more suited to night time photography. But also with those this feature would be very useful for me. 

**Additional context**
Add any other context or screenshots about the request here.

davidplowman, 2023-07-10:
Hi, thanks for the question. I'm afraid I don't know of any support for such a feature in any of our cameras. I'm guessing there'd have to be support in the sensor and in the driver as a minimum. If you're using a 3rd party camera you'd have to reach out to them for an answer, of course.

I didn't quite understand the point about not wanting to add images. Adding ten 1 second exposures should give you very similar noise levels to a single 10 second exposure, for example. There is some example code for adding such images [here](https://github.com/raspberrypi/picamera2/blob/main/examples/stack_raw.py).

Naderschah, 2023-07-10:
Shame that it isnt a thing, thanks for verifying.
I mean that with the 1 second exposure the signal observed isnt strong enough to form any sort of image.
Its essentially like a dark frame, just thermal noise.

Naderschah, 2023-07-10:
Ah thats alright, thank you for the help though. I will post in this thread in case I find a solution. As there isnt much info online about this

Naderschah, 2023-08-27:
I was informed of two methods to increase the exposure time

One using V4L2 
First edit dtoveray and change 
dtoverlay=arducam-pivariety
to
dtoverlay=arducam-pivariety,media-controller=0
Then reboot
Open two terminals, enter into the first:
time v4l2-ctl --set-fmt-video=width=1920,height=1080,pixelformat='RG10' --stream-mmap --stream-count=-1 --set-ctrl vertical_blanking=1046415,exposure=1046415 --stream-count=2 --stream-to=imx462_frames.bin
and into the second
i2ctransfer -f -y 10 w6@0x0c 0x05 0x02 0x30 0x1c 0x00 0xff
i2ctransfer -f -y 10 w6@0x0c 0x05 0x02 0x30 0x1d 0x00 0xff
The content of the second must be entered within 15 seconds of the first command
They provide  the explanation 
The "v4l2-ctl" command captures two frames of raw data. The exposure time for the first frame is 15.5 seconds, while the exposure time for the second frame is 462 seconds. Both frames of raw data are stored in the "imx462_frames.bin" file. You need to write code to parse this data.

And a second method utilizing libcamera 

This time the dtoverlay is dtoverlay=arducam-pivariety
Again we need two terminals, in the first we enter
export LIBCAMERA_SET_TIMEOUT=4294967295
libcamera-still -t 5000 -o imx462_long_exp.jpg --immediate --shutter 15500000
and in the second 
i2ctransfer -f -y 10 w6@0x0c 0x05 0x02 0x30 0x1c 0x00 0xff
i2ctransfer -f -y 10 w6@0x0c 0x05 0x02 0x30 0x1d 0x00 0xff
This must be entered within 5 seconds of the first command 

They provide the explanation:
Due to libcamera discarding several frames before taking a photo, the shooting time becomes exceptionally long (the test took one hour)

I tried controlling exposure time with the second method but so far unsuccesful, 
If anyone figures out how to control it I would ask you add this here for people looking in the future
If I get anywhere I will also post this here