raspberrypi/picamera2 #743: [HOW-TO] Capture 10bit DNG image

RedBlaze42, 2023-07-11:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
Since I use the 3280x2464-SBGGR10 sensor mode, I thought that I would get 10bit colors, but it's not the case.
But because I use the make_buffer method, is saw that there is a cast to uint8.
https://github.com/raspberrypi/picamera2/blob/dd984341d3fda13f1c08c41dc93e25625853bcfe/picamera2/request.py#L148

**Additional context**
Here is the output of `file full.dng`
`TIFF image data, little-endian, direntries=33, height=2464, bps=10, compression=none, PhotometricIntepretation=(unknown=0x8023), manufacturer=RaspberryPi, model=PiDNG / PiCamera2, orientation=upper-left, width=3280`

davidplowman, 2023-07-11:
Hi, thanks for the question. The raw buffer is returned to you as 8-bit bytes, but it still contains 10-bit raw values. If you asked for a compressed format (e.g. "SBGGR10_CSI2P") then you'll get 4 pixels every 5 bytes. If you use the uncompressed form then you get 1 pixel every 2 bytes, with the most significant 6 bits being zero. You can "view" such a buffer as 16-bit values using `numpy.view`.

RedBlaze42, 2023-07-11:
Ok I see.
It's not a pixel array but just a data array.
Thanks for the clarification!

closed 2023-07-11 by RedBlaze42