raspberrypi/picamera2 #744: picamera2 h264enc with gstreamer

Pepslee, 2023-07-13:
Is there possibility to use picamera2 h264encoder in gstreamer pipeline ?

davidplowman, 2023-07-13:
Hi, I believe you can use the V4L2 hardware h.264 encoder in a gstreamer pipeline, though I'm afraid I don't know the details. You might find some more knowledgeable people if you were to ask on one of the Raspberry Pi forums. Sorry not to be able to help more!

Pepslee, 2023-07-13:
> Hi, I believe you can use the V4L2 hardware h.264 encoder in a gstreamer pipeline, though I'm afraid I don't know the details. You might find some more knowledgeable people if you were to ask on one of the Raspberry Pi forums. Sorry not to be able to help more!

Thank you for so quick answer.
I have already used v4l2 for encoding and it works fine, but with v4l2 I have to manage memory allocation myself and v4l2 uses really a lot of memory, so I need to decrease memory of CMA to increase memory of GPU for v4l2 and then I can`t read the high resolution raw images from camera because there is low memory for CMA. I believe that builtin picamera2 h264encoder has CMA as memory manager. But I dont know how stream video in this case without gstreamer

davidplowman, 2023-07-18:
Hi again, I'm afraid I don't know too much about gstreamer and how the video encoding works. I believe you can pass frames to the v4l2 h264 encoder element, and there's a libcamera gstreamer source element that should be able to do this. But you might have more luck asking on the camera forum where there may be people who have tried this.

Out of interest, what kind of Pi are you using, and how large have you set your CMA area? And how do you want to stream/save the encoded video?