raspberrypi/picamera2 #748: [BUG]latest picamera2/libcamera version causes timeouts for externally triggered HQ camera

phr0gger, 2023-07-18:
I had a working setup with externally triggered HQ camera (imx477.trigger_mode=2 in cmdline.txt).
Then I accepted today's (2023-07-18) suggested updates:

- python3-picamera2:arm64 (0.3.9-1, 0.3.12-1), 
- libcamera-tools:arm64 (0~git20230302+923f5d70-1, 0~git20230707+2783c8d8-1), 
- libcamera0:arm64 (0~git20230302+923f5d70-1, 0~git20230707+2783c8d8-1), 
- libcamera-apps:arm64 (0~git20230301+54a781d-1, 1.2.1-1), 
- python3-libcamera:arm64 (0~git20230302+923f5d70-1, 0~git20230707+2783c8d8-1)

And now libcamera gives me timeouts:

[0:34:54.902448592] [1945] WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[18:cap]: Dequeue timer of 1000000.00us has expired!
[0:34:54.902534591] [1945] ERROR RPI pipeline_base.cpp:1333 Camera frontend has timed out!
[0:34:54.902549369] [1945] ERROR RPI pipeline_base.cpp:1334 Please check that your camera sensor connector is attached securely.
[0:34:54.902561961] [1945] ERROR RPI pipeline_base.cpp:1335 Alternatively, try another cable and/or sensor.

I tested the camera connection with imx477.trigger_mode=0 in cmdline.txt, and it works fine.

KLRico, 2023-07-19:
I'm currently using a pip-installed package and I ran into the exact same timeout errors today, after upgrading and trying to run the mjpeg_server_2 example. 

In my case, I was able to fix it with 'apt install libcamera-apps', which replaced my existing 'libcamera-apps-lite'. Making that change is the only thing I did, and everything is working normally again.

It looks like you're already running the full libcamera-apps, so I'm not sure if my fix will help you at all, but it might be a place to start if you need to dig deeper.

Good luck!

phr0gger, 2023-07-19:
Thank you, that works. I added `"camera_timeout_value_ms": 10000,` as my trigger arrives every 2.5 s.

closed 2023-07-21 by phr0gger

nicolasaw, 2023-07-28:
Hi, I have the same problem, I am using Arducam 64MP Hawkeye, and I added `"camera_timeout_value_ms": 10000,`. It still doesn't work, any other resolution for this? Thanks

davidplowman, 2023-07-28:
Can you say in what way it doesn't work? Is it still timing out (now after 10 seconds) when you expected a frame to arrive before that? Or is it timing out earlier than you expect?

nicolasaw, 2023-07-28:
[image]
The error shows like this, I have the libcamera-apps opened, but it keeps showing black screen
[image]

I used the guide for setup here https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/Quick-Start-Guide/#software-guide-for-imx519-fixed-focus-camera

By chance do you have a solution or idea to solve this?

davidplowman, 2023-07-28:
Is it waiting 10 seconds before timing out or not? Can you say what version of Picamera2 you have? Version 0.3.12 should support the timeout parameter. If you're using any software versions from Arducam then you may need to go to them for support.

nicolasaw, 2023-07-28:
Yes it is waiting for the 10 seconds timeout, but the camera still doesn't show images after waiting for 10 seconds. And yes I'm on Version 0.3.12 as well

davidplowman, 2023-07-31:
Hi again, I'm afraid you probably need to reach out to Arducam for support. I don't know anything about their camera modules nor their software nor the modifications they may have made to our software. If you can demonstrate a problem with one of the official Raspberry Pi cameras then I can clearly investigate, but otherwise there's probably not much I can do. Sorry not to be more helpful.

VisheshSingh1, 2023-10-28:
I tried "apt install libcamera-apps" and this also "camera_timeout_value_ms": 10000 but after doing this I am getting the same error . I am using PiCamera 5MP

davidplowman, 2023-11-06:
Hi again, could you maybe file this on the[ libcamera-apps repo](https://github.com/raspberrypi/libcamera-apps)? It doesn't sound like it's actually a Picamera2 thing, and you'll get people who know more about external camera triggers there.

sundance69, 2023-12-10:
Hi
this solved my issues with frontent timeout:
sudo apt update
sudo apt upgrade -y
sudo rpi-update pulls/5691

have an ov5647 ca 
m running on anPi 5

ABlueOctopus, 2023-12-15:
> Yes, I haven't been doing anything special. All I've done is the basic guides for installation from Arducam [https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/Quick-Start-Guide/#software-guide-for-imx519-fixed-focus-camera](url)
> 
> And when I just run basic `libcamera-hello` It shows the blank screen and the error earlier
> 
> I am using Arducam 64MP Hawkeye with RasberryPi 4

Hi, have you solved the issue? I am using the same camera and having the same error...

StarRoadlu, 2024-04-08:
I encountered the same issue as well. How did you resolve it?

> ![图像](https://private-user-images.githubusercontent.com/74193885/256797060-61e9e2a2-95f1-4690-b152-8122bdc27d05.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1NjkxNTQsIm5iZiI6MTcxMjU2ODg1NCwicGF0aCI6Ii83NDE5Mzg4NS8yNTY3OTcwNjAtNjFlOWUyYTItOTVmMS00NjkwLWIxNTItODEyMmJkYzI3ZDA1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA4VDA5MzQxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxYzU0ZmJmNDBlMWVlZGE5OWMwZTY5YmIwYTU0MGY3MjMyYzNlOWRiMjY3MDU2MmExY2M4MDdiY2NkN2I3NzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.andNTlZr3XtgZg8jxUiZgraRd4bYiPjk7wWMm3eSEPI)错误显示如下，我打开了libcamera-apps，但它仍然显示黑屏 ![图像](https://private-user-images.githubusercontent.com/74193885/256797227-34059583-14fa-4b68-a4b8-7e49d59d22e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1NjkxNTQsIm5iZiI6MTcxMjU2ODg1NCwicGF0aCI6Ii83NDE5Mzg4NS8yNTY3OTcyMjctMzQwNTk1ODMtMTRmYS00YjY4LWE0YjgtN2U0OWQ1OWQyMmU2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA4VDA5MzQxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxN2RhM2FiYzFiMzg0YTIxMjY5N2I5ZWUxNDgxMmVlYThjZDYyNzNmNGFjNWUyZmFlYWExNDI2OWY1ZGQxZjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Vlj29VoelmYtfSubKlOFS-MTYAib6OvD0qx2eE_f1rs)
> 
> 我在这里使用了设置指南 https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/Quick-Start-Guide/#software-guide-for-imx519-fixed-focus-camera
> 
> 碰巧你有解决这个问题的解决方案或想法吗？

I encountered the same issue as well. How did you resolve it?

scoles10, 2024-05-17:
Same error here trying to understand what happen. My camera was working just fine and then I see this issue.

davidplowman, 2024-07-26:
@ncr292 Hi, the original post here relates to a Pi 5 experiencing timeouts when using an externally triggered camera, which was resolved and closed.

If you are having problems with a Pi Zero and an imx219 then it is not likely to be relevant to the issue here. Please file new reports for new problems. When you do so, please be sure to include:

* The kind of Pi that you are using (include amount of memory when this can change).
* The camera you are using. Please state if it is an official Raspberry Pi camera or a 3rd party camera.
* The kind of power supply that you are using. Is it an official Raspberry Pi power supply?
* The OS version (`cat /etc/os-release`). Please state if you are not using Raspberry Pi OS.
* The kernel version (`uname -a`)
* The firmware version (`vcgencmd version`)
* The camera software version (`libcamera-hello --version`)
* The Picamera2 version (`apt list python3-picamera2`)
* Please try to include the shortest script that you can that demonstrates the problem. If possible, this should be no more than half a dozen lines.
* Include any error messages that are printed to the console.
* Are there any error messages at the bottom of `dmesg`?
* Please confirm if possible that the problem persists with a fresh and fully updated install of the OS.
* The length of your ribbon cable connector to the camera.
* Is there anything in the environment that might cause electrical interference.

Thank you.

Raccoon987, 2024-10-23:
A few days ago, after enabling external trigger mode by running:

```
sudo su
echo 1 > /sys/module/imx296/parameters/trigger_mode
exit
```

my global shutter Raspberry Pi cameras stopped working:

```
iop@raspberrypi:~/Documents/WORK_DIR_summer_24 $ libcamera-hello -c 0 -t 0
[1777]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[1783]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 14-02-2024 (14:00:12)
[1783]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[1783]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 14-02-2024 (14:00:12)
[1783]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx296@1a to CFE device /dev/media1 and ISP device /dev/media3 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 728:544:12:P
    SRGGB10_CSI2P,1456x1088/0 - Score: 1318
Stream configuration adjusted
[1777]  INFO Camera camera.cpp:1183 configuring streams: (0) 728x544-YUV420 (1) 1456x1088-BGGR16_PISP_COMP1
[1783]  INFO RPI pisp.cpp:1405 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx296@1a - Selected sensor format: 1456x1088-SBGGR10_1X10 - Selected CFE format: 1456x1088-PC1B
[1783]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video12[16:cap]: Dequeue timer of 1000000.00us has expired!
[1783] ERROR RPI pipeline_base.cpp:1376 Camera frontend has timed out!
[1783] ERROR RPI pipeline_base.cpp:1377 Please check that your camera sensor connector is attached securely.
[1783] ERROR RPI pipeline_base.cpp:1378 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
```

Simply wrote '1' to the file, without running any MicroPython code in Thonny. I’ve tested this setup on two Raspberry Pi 5 boards, with four GS cameras, four cables, three Raspberry Pi Picos, and two SD cards with the OS. After reboot cameras work fine, but not it trigger mode.

```
 Static hostname: raspberrypi
       Icon name: computer
      Machine ID: fcbc6e2529de4c86972b9a522cdbb167
         Boot ID: 069fef7ee6f24ea6b3517f2a1ffacb79
Operating System: Debian GNU/Linux 12 (bookworm)  
          Kernel: Linux 6.1.0-rpi8-rpi-2712
    Architecture: arm64
```
How to solve this problem? 
Thank you.

davidplowman, 2024-10-23:
Can you just check your `camera_timeout_value_ms` hasn't been changed (by doing an update, for example). You should have a large value in there.

Raccoon987, 2024-10-24:
Apologies for the delayed response. In my Python code, I set

`os.environ["LIBCAMERA_RPI_CONFIG_FILE"] = "./timeout.yaml"`

In the timeout.yaml file, I set:  

`camera_timeout_value_ms: 60000`

Now, instead of getting error messages right away, they appear after waiting 60 seconds. The Raspberry Pico signal is fine — I verified it with an oscilloscope.

davidplowman, 2024-10-24:
And you've made the hardware documentations described here? https://www.raspberrypi.com/documentation/accessories/camera.html#external-trigger-on-the-gs-camera

Raccoon987, 2024-10-24:
The cameras stop working after changing the value from 0 to 1 in the /sys/module/imx296/parameters/trigger_mode file. The Pico may be disconnected from the Raspberry Pi 5.

Raccoon987, 2024-10-26:
Everything works great now for me. It was my mistake—I missed something. I appreciate your help.

davidplowman, 2024-10-28:
Great, glad it's working!