raspberrypi/picamera2 #749: [HOW-TO] Keep camera acquisition rate using PyQT5 (threads vs processes)

LeoMarHor, 2023-07-18:
I am developing an app which takes images from a camera, do some processing with each frame and display a GUI for the user. The hardware I am trying this on is a Raspberry Pi 4 model b, an Arducam camera module having camera sensor AR1820HS and, as recommeded, PyQT5 for the GUI. The processing involves face detection, eye landmarks and neural networks inference using Coral Accelerator.

I have already the camera adquisition code working at around 48 fps (1224x922) and image processing code takes 25 ms. I would need more resolution when this works so I can more details. This means 3840x2160px which allows 14,1 fps (YUV420) if done independently.

Well, I have looked at other similar developments and I have beeing doubting between using multiprocessing or multithreading to make all this works efficiently in a concurrent manner. I decided to implement 2 QThreads (apart from GUI thread) using a queue to share the acquired frames. One of them would be the frame producer thread and the other would process the frame and output a result.

Below is the code I wrote...

The problem is that I am getting around 20 fps for 1224x922px and image processing is just returning the average of the frame. 

Am I doing something wrong? Is this as high as it can get?
Should I try multiprocessing? 

```
import sys

import atexit

from queue import Queue

from PyQt5.QtCore import Qt, QTimer
from PyQt5.QtWidgets import (
    QApplication,
    QLabel,
    QMainWindow,
    QPushButton,
    QVBoxLayout,
    QWidget,
    QMessageBox,
    QAction,
)

from PyQt5.QtCore import QObject, QThread, pyqtSignal

from picamera2 import Picamera2
import libcamera

from picamera2.controls import Controls

import numpy as np
import cv2 as cv

ANALOG_GAIN = 4.0
EXPOSURE_TIME = 40000    
RESOLUTIONS = ["1224x922", "2448x1378", "3840x2160", "4912x3684"]
SELECTED_RES = RESOLUTIONS[0]
#FORMAT = 'RGB888' 
FORMAT = 'YUV420'

class Camera_Producer(QObject):
    frame_produced = pyqtSignal(int)

    def __init__(self, buffer_queue):
        """Initial camera configuration
        """
        super().__init__()
        atexit.register(self.cleanup)
        self.buffer_queue = buffer_queue
        
        self.picam2 = Picamera2()
        
        resH = int(SELECTED_RES.split('x')[0])
        resV = int(SELECTED_RES.split('x')[1])

        still_config = self.picam2.create_still_configuration(main={"format": FORMAT, "size": (resH, resV)})
        still_config["transform"] = libcamera.Transform(hflip=0, vflip=0)
        self.picam2.configure(still_config)
        self.picam2.start()
        
        metadata = self.picam2.capture_metadata()
        controls = {c: metadata[c] for c in ["ExposureTime", "AnalogueGain", "ColourGains"]}
        print("Camera controls: {}".format(controls))
        
        ctrls = Controls(self.picam2)
        ctrls.AnalogueGain = ANALOG_GAIN
        ctrls.ExposureTime = EXPOSURE_TIME
        self.picam2.set_controls(ctrls)
    
    def cleanup(self):
        print("Camera destroyed")
        
    def run(self):
        """Adquire images and stores in queue for later processing by the consumer thread.
        It will adquire all the time
        """
        while(1):
            if self.buffer_queue.qsize() == 0:
                frame = self.picam2.capture_array()
                
                if FORMAT == 'RGB888':
                    img_gray = cv.cvtColor(frame, cv.COLOR_BGR2GRAY)
                elif FORMAT == 'YUV420':
                    img_gray = cv.cvtColor(frame, cv.COLOR_YUV420p2GRAY)
                
                self.buffer_queue.put(img_gray)
                self.frame_produced.emit(1)
            
class Algorithm_Consumer(QObject):
    frame_processed = pyqtSignal(int)
    mean_image = pyqtSignal(float)

    def __init__(self, buffer_queue):
        """Initial algorithms configuration (ie. load models)
        """
        super().__init__()
        atexit.register(self.cleanup)
        self.buffer_queue = buffer_queue
        
    def cleanup(self):
        print("Algorithm destroyed")
        
    def run(self):
        """Grabs image, or images, from camera queue, process it and update mouse position
        """
        while(1):
            if self.buffer_queue.qsize() > 0:
                frame_gray = self.buffer_queue.get()
                self.mean_image.emit(np.mean(frame_gray))
                self.frame_processed.emit(1)

class Window(QMainWindow):
    def __init__(self, parent=None):
        super().__init__(parent)
        self.clicksCount = 0
        self.setupUi()
        
        #Creates Queue for sharing data (images)
        queue_buffer = Queue(maxsize=1)
        
        #Initializes camera and algorithm threads
        self.thread_camera = QThread()
        self.worker_camera = Camera_Producer(queue_buffer)
        self.worker_camera.moveToThread(self.thread_camera)
        self.thread_camera.started.connect(self.worker_camera.run)
        self.thread_camera.start()
        
        self.thread_algorithm = QThread()
        self.worker_algorithm = Algorithm_Consumer(queue_buffer)
        self.worker_algorithm.moveToThread(self.thread_algorithm)
        self.thread_algorithm.started.connect(self.worker_algorithm.run)
        self.thread_algorithm.start()
        
        #Way to handle proper closure of all objects
        quit = QAction("Quit", self)
        quit.triggered.connect(self.closeEvent)
        
        #10 seconds timer to update fps (debugging purposes for now)
        timer = QTimer(self)
        timer.timeout.connect(self.updateFrameRate)
        timer.start(1000) # 1 Second Refesh Rate
        self.counter = 0
        self.camera_counter = 0
        self.algorithm_counter = 0
        
        self.worker_camera.frame_produced.connect(self.updateCameraCounter)
        self.worker_algorithm.frame_processed.connect(self.updateAlgortithmCounter)
        self.worker_algorithm.mean_image.connect(self.algorithmResult)
        
        self.result = 0
    
    def updateCameraCounter(self):
        self.camera_counter += 1
        
    def updateAlgortithmCounter(self):
        self.algorithm_counter += 1
    
    def algorithmResult(self, result):
        self.result = result
        #print("Mean image result: {}".format(result))
    
    def updateFrameRate(self):
        self.counter += 1
        #print("General Counter: {}".format(self.counter))
        
        #TODO Update value according to interval time defined
        print("Frame Rate - Camera: {} fps".format(self.camera_counter))
        print("Frame Rate - Algorithm: {} fps".format(self.algorithm_counter))
        
        print("---o---")
        
        #Resets values
        self.algorithm_counter = 0
        self.camera_counter = 0
        
    def closeEvent(self, event):
        reply = QMessageBox.question(self, 'Cerrar aplicación?',
                                     'Estás seguro que quieres salir?',
                                     QMessageBox.Yes | QMessageBox.No, QMessageBox.No)

        if reply == QMessageBox.Yes:
            self.thread_camera.quit()
            self.thread_algorithm.quit()
            
            print("Destroyed all")
            if not type(event) == bool:
                event.accept()
            else:
                sys.exit()
        else:
            if not type(event) == bool:
                event.ignore()
    
    def setupUi(self):
        self.setWindowTitle("Freezing GUI")
        self.resize(300, 150)
        self.centralWidget = QWidget()
        self.setCentralWidget(self.centralWidget)
        # Create and connect widgets
        self.clicksLabel = QLabel("Counting: 0 clicks", self)
        self.clicksLabel.setAlignment(Qt.AlignHCenter | Qt.AlignVCenter)
        self.stepLabel = QLabel("Long-Running Step: 0")
        self.stepLabel.setAlignment(Qt.AlignHCenter | Qt.AlignVCenter)
        self.countBtn = QPushButton("Click me!", self)
        self.countBtn.clicked.connect(self.countClicks)
        self.longRunningBtn = QPushButton("Long-Running Task!", self)
        self.longRunningBtn.clicked.connect(self.runLongTask)
        # Set the layout
        layout = QVBoxLayout()
        layout.addWidget(self.clicksLabel)
        layout.addWidget(self.countBtn)
        layout.addStretch()
        layout.addWidget(self.stepLabel)
        layout.addWidget(self.longRunningBtn)
        self.centralWidget.setLayout(layout)

    def countClicks(self):
        self.clicksCount += 1
        self.clicksLabel.setText(f"Counting: {self.clicksCount} clicks")

    def reportProgress(self, n):
        self.stepLabel.setText(f"Long-Running Step: {n}")

    def runLongTask(self):
        self.reportProgress(1)

app = QApplication(sys.argv)
win = Window()
win.show()

sys.exit(app.exec())
```

davidplowman, 2023-07-19:
Hi, there's probably a bit more there than I can reasonably digest, but here are a few things I would check.

Firstly, I'd take the processing after `capture_array()` our of your `Camera_Producer.run()` method, just to check that we really can get frames at the rate you want.

If you want greyscale images, YUV420 is a much better format to use. I'd have thought you could just do `img_gray = frame[:resV, :resH]` which would hopefully just give you a view onto the buffer you already have, rather than copying it again (though I don't know what `cv2.cvtColor` will do in this case).

I was a bit bothered by the `Algorithm_Consumer.run()` method which, unless I've misunderstood things, is going to busy wait. I think the test on the queue size may be superfluous as I think `queue.get()` will block until there's something there.

Also `Camera_Producer.run()` might also by busy-waiting while there's something in the queue? Again the test on the queue size being zero may be unnecessary. IIRC, you can set up a queue with a max size, so that `queue.put()` would block until there's space, so that might be a way to stop the queue growing indefinitely.

Multi-processing has to be used with care. Unless you know the runes for getting image buffers across without copying them then it can be quite slow. Though I suppose if the processing is _way_ slower than buffer copying then maybe it's not a big deal.

Multi-threading works well so long as the processing you're doing either blocks for something else to happen (during which it should release the GIL), or calls into a big C/C++ library (in my experience these normally release the GIL while you're in them, though I'm sure there are some that don't...).

LeoMarHor, 2023-07-19:
> 

Hello @davidplowman, thanks for your help!
Where should I place capture_array() for your test in this multithreaded code? The result I reported (48 fps (1224x922)) is just a simple script where I configure the camera and get the mean time taken to obtain 100 frames.

Regarding the other comments:
- For some situations, I might need RGB data. I will use what you suggest for when I need YUV.
- Is there a way to tell the thread to yield? The size of the queue was set to 1 and might need to set to 2 in other cases.
- Below I leave some code that I am using to the test multiprocesing... I have 2 image processing codes that I would need to use (one or the other which is faster) and one of them takes around 250 ms per frame. I was concerned GIL was causing the issues in multithreading. Is there a way to estimate how much time it would take to share 8 MB (3840x2160, grayscale)?

`from time import perf_counter, sleep
import multiprocessing as mp
from queue import Empty
from picamera2 import Picamera2
import libcamera
from PIL import Image

from picamera2.controls import Controls

import numpy as np
import cv2 as cv

ANALOG_GAIN = 4.0
EXPOSURE_TIME = 40000    
RESOLUTIONS = ["1224x922", "2448x1378", "3840x2160", "4912x3684"]
SELECTED_RES = RESOLUTIONS[0]
FORMAT = 'RGB888' 
#FORMAT = 'YUV420'

def do_capture(queue, finished):
    picam2 = Picamera2()
        
    resH = int(SELECTED_RES.split('x')[0])
    resV = int(SELECTED_RES.split('x')[1])

    still_config = picam2.create_still_configuration(main={"format": FORMAT, "size": (resH, resV)})
    still_config["transform"] = libcamera.Transform(hflip=0, vflip=0)
    picam2.configure(still_config)
    picam2.start()

    metadata = picam2.capture_metadata()
    controls = {c: metadata[c] for c in ["ExposureTime", "AnalogueGain", "ColourGains"]}
    print("Camera controls: {}".format(controls))

    ctrls = Controls(picam2)
    ctrls.AnalogueGain = ANALOG_GAIN
    ctrls.ExposureTime = EXPOSURE_TIME
    picam2.set_controls(ctrls)   
    while(1):
        if queue.qsize() == 0:
            t1_start = perf_counter()
            
            frame = picam2.capture_array()
            
            if FORMAT == 'RGB888':
                img_gray = cv.cvtColor(frame, cv.COLOR_BGR2GRAY)
            elif FORMAT == 'YUV420':
                img_gray = cv.cvtColor(frame, cv.COLOR_YUV420p2GRAY)
                
            t1_stop = perf_counter()
            time_capture = t1_stop - t1_start
            
            print("Time using for capturing image: {:.2f}ms".format(1000*time_capture))
            
            queue.put(img_gray)

def do_processing(queue, finished):
    while(1):
        if queue.qsize() > 0:
            frame_gray = queue.get()
            print(np.mean(frame_gray))
            
            sleep(0.1)
            #print('%d: Processing image with size %dx%d' % (
            #    os.getpid(), frame_gray.size[0], frame_gray.size[1]))

if __name__ == '__main__':
    queue = mp.Queue()
    finished = mp.Event()
    
    capture_proc = mp.Process(target=do_capture, args=(queue, finished))
    processing_proc = mp.Process(target=do_processing, args=(queue, finished))
    
    processing_proc.start()
    capture_proc.start()
    
    processing_proc.join()
    capture_proc.join()`

LeoMarHor, 2023-07-19:
I looked at the documentation and there is actually a way to yield. So, I added "QThread.yieldCurrentThread()" in an else condition within each while(1) and I am getting slighter better results:
25fps instead of 18 fps.

Still not satisfied...

davidplowman, 2023-07-20:
So I understand that sometimes one finds oneself in a situation where "yielding" a thread is necessary, nonetheless it always worries me that it's just masking a different problem. Did you the problem with the busy waiting? Normally, so long as threads sleep properly when they aren't doing anything useful then I would mostly expect things to work.

Also, is your code setting the exposure time to 40000 (i.e. 40ms)? In this case the maximum achievable framerate will be 1000/40 = 25 fps.

I'm afraid I can't tell you for sure how long it will take to copy these buffers between processes. I can only really suggest that you write to code to test the performance - possibly there is even something out there on the web.

LeoMarHor, 2023-07-21:
Ok, I found the bug... It was in my code. I had to comment the "sleep(0.5)" and it **is giving the same results now as in the multithreaded version**... I have copied it below... 
The 0.5 seconds gives some time to the camera module to acquire the image... I do not know why I added it.

The weird thing now is that the YUV420 takes more time than RGB888. For example, at 2432x1840, it takes 45 ms for RGB888 and 88 ms for YUV. Is this because the data is converted to YUV in the Raspberry and it adds more time? The format for this camera, at this resolution, is 2432x1840-SGRBG10_1X10. The camera mode displayed by --list-cameras says for this resolution: 2432x1840 [20.01 fps - (0, 0)/2432x1840 crop]

`while iterator < IMAGE_COUNT:
    
        t1_start = perf_counter()
        im_captured = picam2.capture_array()
        
        #Store image in gray
        if FORMAT == 'RGB888':
            img_gray = cv.cvtColor(im_captured, cv.COLOR_BGR2GRAY)
        elif FORMAT == 'YUV420':
            img_gray = cv.cvtColor(im_captured, cv.COLOR_YUV420p2GRAY)
            
        t1_stop = perf_counter()
        time_capture = t1_stop - t1_start
        time_capture_array.append(time_capture)

        iterator += 1
        #sleep(0.5)`

davidplowman, 2023-07-21:
Glad it's working better! Using a "still" configuration will by default do extra high quality colour denoising if it thinks it can, which is actually when you ask for a YUV420 image (because for still image capture you care less about framerate, or so it thinks...). To stop it, add `controls={'NoiseReductionMode': 1}` into your call to `create_still_configuration`. Or alternatively try `create_video_configuration` instead.

LeoMarHor, 2023-07-21:
I need both quality and speed... I will have set a line at some point.

Thanks for your help!

closed 2023-07-21 by LeoMarHor