raspberrypi/picamera2 #751: [BUG] Out of focus / Blurry landscape pictures

an-toine, 2023-07-19:
Hello,

I've been using for 3 years a Raspberry Pi Camera module 2 to take pictures from a basket hanged to a Kite (aka [Kite Aerial Photography](https://en.wikipedia.org/wiki/Kite_aerial_photography).

With the help of the [picamera module](https://picamera.readthedocs.io/en/release-1.13/) I was able to take clear and sharp pictures easily, without additional tuning.

I've recently upgraded my setup to use a Pi Camera Module 3 in the hope of improving image quality. In the process, I've upgraded my code base to use the `picamera2` module.

**Describe the bug**
Pictures taken from the kite are blurry and seems to be out of focus. On some pictures, we can observe pixelized areas, and blurry portions.

**To Reproduce**
I use the following snippet to take pictures in Python :

```python
# Disable Wide Dynamic Range
subprocess.run("/usr/bin/sudo v4l2-ctl --set-ctrl wide_dynamic_range=0 -d /dev/v4l-subdev0", shell=True, check=True, capture_output=True)
picam2 = Picamera2()
picam2.options["quality"] = 95
picam2.options["compress_level"] = 0
camera_config = picam2.create_still_configuration(main={"size": (4608, 2592)})
camera_config["transform"] = libcamera.Transform(hflip=1, vflip=1)
picam2.configure(camera_config)
picam2.start(show_preview=False)
time.sleep(2)
picam2.autofocus_cycle()
time.sleep(2)
picam2.set_controls({"AfMode": libcamera.controls.AfModeEnum.Manual, "LensPosition": 0.0, "AeEnable": False})
logging.info("Camera initialized")
# Later, when a picture is requested by user
picam2.capture_file("{}/{}".format(config["pictures"]["save_dir"],
                                                     picture_name),
                                                     format="jpeg")
``` 

**Expected behaviour**
I would have expected to get sharp and clear pictures, the way I did with the camera module 2.

**Console Output, Screenshots**
Here are some pictures taken with the camera module 3 to illustrate the problem :
[image]
[image]

On the ground, without moving I get a better picture, even though quality seems inferior to the module 2 :
[image]

**Hardware :**

Raspberry Pi :
```
pi@raspberrypi:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 797.66
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000003e5ae754
Model		: Raspberry Pi Zero W Rev 1.1
```
Camera :
```
pi@raspberrypi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [30.00 fps - (0, 0)/0x0 crop]
                             2304x1296 [30.00 fps - (0, 0)/0x0 crop]
                             4608x2592 [30.00 fps - (0, 0)/0x0 crop]
``

The Raspberry Pi is powered by a basic power bank.

**Additional context**
When flying approximately 50-75 meters above ground, wind can shake quite a lot the attached camera module : I've assumed that it had something to do with these movements messing with HDR, but it could be something completely different.

I've also set the camera focus to Manual mode and the `LensPosition` to `0.0` to avoid issues with autofocus.

Usually, it takes around 3 to 5 seconds to take a picture, could this long exposure time related to the poor results I get ?

Is there a way to improve the image quality ? More specifically, I'm looking for a way to at least get similar pictures to the ones I got with the camera module 2.

Feel free to ask me for additional information if needed.

Regards,

Antoine

davidplowman, 2023-07-19:
The autofocus mechanism, as is very common in small camera modules, is basically a lens on the end of a spring. It is therefore quite susceptible to vibration, both from movement and external factors like the wind. We can clearly reduce the effect of the wind by housing the camera in some kind of casing, but movement is very difficult to deal with.

We've had some users glue the autofocus lenses into a fixed position, but this is both tricky and (of course) WARRANTY VOIDING. They report that the drying glue will move the lens, so you have to be actively driving it to infinity as the glue sets. They did eventually get some acceptable modules this way, but only after numerous failures. Sorry not to have better news.

an-toine, 2023-07-20:
Thanks a lot for your inputs, that's exactly what I was looking for !

I now have a better understanding of the problem : this is not a software issue but rather a mechanical one.

> We've had some users glue the autofocus lenses into a fixed position, but this is both tricky and (of course) WARRANTY VOIDING. They report that the drying glue will move the lens, so you have to be [...]

By any chance, are you aware of documentation or walk-through that could guide me in this uncertain process ?

We can close this issue, I will resort to Camera module 2 till I find a solution to stabilize my setup.

Antoine

davidplowman, 2023-07-20:
I can't say there is any "official" procedure for gluing these lenses. I think the procedure would go something like:

* Squirt glue into lens.
* Run "libcamera-hello -t 0" with the camera pointed at infinity (and plenty of detail so that the continuous autofocus knows what to do).
* Wait 24 hours (or however long the glue takes to dry completely).
* Test the lens and check that it no longer responds to lens movement commands.

But as I said, be prepared to destroy a number of modules on the way, and I can't even guarantee eventual success. Just in case anyone finds this comment and doesn't read the ones above: this of course voids any warranty completely.

closed 2023-07-20 by an-toine