raspberrypi/picamera2 #753: [BUG] H264Encoder fails to start for high resolution

RolandasRazma, 2023-07-19:
**Describe the bug**
`H264Encoder` fails to start for high resolution

**To Reproduce**
```python
config = camera.create_video_configuration(main={"size": (2304, 1296), "format": "RGB888"}, raw={"size": (2304, 1296)})
encoder = H264Encoder() # setting bitrate doesn't help
camera.start_recording(encoder=encoder, config=config, quality=Quality.VERY_HIGH)
```

**Expected behaviour**
Video encoder to work

**Console Output, Screenshots**
File "picamera2/encoders/encoder.py", line 234, in start
&nbsp;&nbsp;self._start()
File "picamera2/encoders/h264_encoder.py", line 71, in _start
&nbsp;&nbsp;super()._start()
File "picamera2/encoders/v4l2_encoder.py", line 137, in _start
&nbsp;&nbsp;fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)

**Hardware :**
 Picamera V3

**Additional context**
 picamera2 0.3.12 (but this is long standing)

davidplowman, 2023-07-19:
I'm afraid video encode is limited to 1920x1080. Actually you can probably get closer to 2048, and even increase the height, but you certainly won't reach 2304x1296.

RolandasRazma, 2023-07-19:
thanks, thought so

closed 2023-07-19 by RolandasRazma