raspberrypi/picamera2 #754: request.getmetadata no longer returns FocusFOM

jlprojects, 2023-07-20:
**Describe the bug**
My app has a live camera preview window that displays some metadata alongside. After the last Picamera2 update it crashes with KeyError: "FocusFOM"

**To Reproduce**
It looks like the FocusFOM value is no longer being returned by request.get_metadata(), causing the error.

**Hardware :**
Raspberry Pi 4, High Quality Camera, current Raspberry Pi OS

**Additional context**
The camera has a macro lens attached and is manually focused onto a 16mm or 8mm film frame. This was a useful metric to display, as it helps with getting critical focus - adjust the focus until the maximum value is reached. Removing the reference makes my code work, but without this useful feature.

Here's an excerpt from my previously working code:

```
    def preview_data_update_callback(self, request):
        m = request.get_metadata()
        text = (f'Exposure: {m["ExposureTime"]}\n' +
                f'Gain: {m["AnalogueGain"]:.2f}\n' +
                f'Focus: {m["FocusFoM"]//100}\n')
```

davidplowman, 2023-07-20:
Hi, thanks for reporting this. Indeed something does seem to have got lost there. I'll look into it, but somewhat suspect we're going to have to make up a new libcamera release.

jlprojects, 2023-07-20:
Thanks! I'm glad it wasn't an intentional feature change. :)

davidplowman, 2023-07-20:
It's only a small fix and I've committed it to the repo. Just need to wait while the packaging/release wheels turn. Hopefully not long.

davidplowman, 2023-07-21:
I think
```
sudo apt update
sudo apt install libcamera0
```
should fix it for you now.

jlprojects, 2023-07-22:
Verified working. That did the trick. Thanks for your work on this!

cgobat, 2023-10-14:
FWIW, I encountered a similar issue (`"FocusFoM"` keyword missing from metadata dictionary) recently, even after a `sudo apt update && sudo apt install libcamera0`. However, the problem occurs for me only in the first frame captured after a `.start()` call. Subsequent frames' metadata contains all the expected keywords, so it is possible to "fix" the issue by just throwing away the first frame, but I figured that this may also be indicative of a larger issue and/or something that is worth investigating. Note also that despite some of the other values being wonky in the first frame(s)&mdash;e.g. #824&mdash;FocusFoM seems to be the only thing that is ever *missing* entirely.

davidplowman, 2023-10-16:
Hi, thanks for reporting this. As you know, there are some difficulties particularly with the very first frame, though there are probably some other things that take a bit longer to settle as well. I'm not quite sure how it comes about that some things report default values early on, and some might be missing. It could be a bug, but we probably won't do much until we've done all Pi 5 work. Pi 5 works quite differently in some crucial respects and so we'll need to find solutions that work on both our platforms. So please bear with us... thanks!

jlprojects, 2023-10-16:
<original poster here>
Probably worked for me as the camera runs a short while before my code gets round to reading "FocusFoM".

I've a Pi 5 on order and am looking forward to migrating my cine film scanning project to it. The processing and storage speed improvements should really help as these are the main bottlenecks at the moment. 
Looks like there might be some fun and games with picamera2/libcamera? My Python (+Cython) application is processing full resolution raw bayer data (from stills captures) from the HQ camera, with the live preview for critical focus and to obtain base exposure and colour balance data. I'm willing to help out and test as and when the Pi 5 and camera cable lands.

davidplowman, 2023-10-16:
Mostly the Pi 5 should behave exactly the same for you. The one important difference which you may encounter is that when you ask for unpacked raw data (presumably this is what you do?) , the pixels will be delivered with one pixel per 2 bytes (as on a Pi 4), but the pixel values will be left-shifted so that all the zero bits are at the bottom and not at the top (as is the case on Pi 4).

jlprojects, 2023-10-16:
Thanks for the heads-up. Looks simple enough - just a change to the lookup table that expands the 12bit data to the full 16bit range used in the rest of the processing.

jlprojects, 2023-11-06:
I now have the Pi 5 and working on modifying my code for Pi5/Bookworm. The extra speed is nice.
It looks like FocusFoM takes a while to appear in the metadata after the camera starts as described by cgobat above causing my application to crash as well, and it's returning a much larger number than previously. As a workaround, I modified my metadata callback to check if it's available first:
```
    def preview_data_update_callback(self, request):
        m = request.get_metadata()
        text = (f'Exposure: {m["ExposureTime"]}\n' +
                f'Gain: {m["AnalogueGain"]:.2f}\n')
        if 'FocusFoM' in m:
                text += f'Focus: {m["FocusFoM"]//1000000}\n'
...
        self.controls['status_label'].setText(text)

closed 2023-11-06 by jlprojects

davidplowman, 2023-11-06:
It's not really "meant" to return a larger number, though I recall that during development it kept doing that. So we might want to fix that. It is, of course, entirely arbitrary, so can only meaningful be used in comparisons against other FoMs!

jlprojects, 2023-11-06:
I noticed that it reduces by about a factor of 4 when going from 4056x3040 to 2028x1520 resolution. Based on the number of pixels?  Although arbitary, it's a useful metric to see when manually focussing. Though I think it'd be better if it would return a value normalized to some range independent of image size, or 0 if the FocusFoM value isn't present in the underlying metadata at all. :)

davidplowman, 2023-11-06:
Yes, the FoM is just a summation over all the pixels (in the region or regions being counted). Normalising to the number of pixels sounds like it might be worthwhile, and it would be amusing to see which sensors seem to give best "per pixel sharpness". One for the list of things to look at!

davidplowman, 2023-11-06:
Yes, we finally got round to posting the updates that I've been alluding to in the forums for a while now! So we did fix that oversized FoM thing, then, I was always complaining about it...

I put a post on the forum about the updated software. Mostly things should work the same, but manipulating pixels within image buffers should be faster.

jlprojects, 2023-11-06:
Thanks - I'll check the forum. :)