raspberrypi/picamera2 #762: [HOW-TO] How does metering_mode, contraint_modes work and affect AGC?

Dephrilibrium, 2023-07-27:
**Describe what it is that you want to accomplish**
I want to avoid any pixel going into saturation. For that I want to adjust the exposure-time before that happens.
In the documentation I found a list of the metering modes I found, but not how they exactly work or which parameters needs to be adjusted in which way to achieve auto-adjusted SS before any overexposure occurs.

**Additional context**
The images I capturing look like small white spots, coming and going with a voltage I apply to a sample my camera is mounted on. For each sample I dont' know where exactly how much spots occur. Furthermore I need the brightness-values of the spots, nevermind where they are on the image. It's just important that there is no pixels in saturation, which is why I want to adjust the SS.

What I'v done so far is.
I've already adjusted all weights by:
```
_agc = Picamera2.find_tuning_algo(tune, "rpi.agc")
_agc["metering_modes"]["centre-weighted"]["weights"] = [1.0] * 15
_agc["metering_modes"]["spot"]           ["weights"] = [1.0] * 15
_agc["metering_modes"]["matrix"]         ["weights"] = [1.0] * 15
```
but to be honest, I'm not sure about, if I need this, what the AGC is exactly doing and how y_target comes into play.

I saw, that there exists "contraint_modes" where the luminance-histrograms can be used. I think this fulfills more the functionality I'm looking for. Therefore, I had a look in the documentation and the debugger and found in the contents of the tune-file (_tune2):

`_tune2["algorithms"][7]["rpi.agc"]["constraint_modes"]["normal"][0]`
```
'bound': 'LOWER'
'q_lo': 0.98
'q_hi': 1.0
'y_target': [0, 0.3, 1000, 0.3]
```
From the documentation it's not clear for me, if its enough to change that to "UPPER", so that the values keep below and why i have 4 different `y_target` values. Unfortunately there is also no example, where I can see how this setting is adjusted.

Kind regards,
Deph

davidplowman, 2023-07-28:
Basically, the _metering_ affects how we measure the current image brightness, for instance using a centre-weighted average, a small spot, and so forth. The AGC aims to get this measurement near a specific target level.

The _constraints_ then use the histogram to further bound the target brightness. For example
```
'bound': 'LOWER'
'q_lo': 0.98
'q_hi': 1.0
'y_target': [0, 0.3, 1000, 0.3]
```
would say that you require the top 2% of the image histogram (actually the average value of the top 2% of the histogram) to be at or above 0.3 (30%) of the full dynamic range. But it's free to be way brighter than this if it wants.

I think there's already a "highlight" constraint mode that says something like
```
'bound': 'UPPER'
'q_lo': 0.98
'q_hi': 1.0
'y_target': [0, 0.8, 1000, 0.8]
```
meaning that the average value of the top 2% cannot be higher than 80% up the dynamic range. Note that 2% of the histogram is already quite a lot of pixels, so it may or may not work for you. Making this percentage very small does make it increasingly unstable, unfortunately.  You could try pushing that 80% value down, of course.

Remember that all this is being done pre-gamma. So a 50% brightness level here will translate in to something nearer 80% in the output image. The 80% value above is probably not far off ~95%.

Dephrilibrium, 2023-07-28:
Hello david

Ah I see, so 2% would be still 250k pixels considered by the histograms.
So, for instance, if I'd try something like this:
```
'bound': 'UPPER'
'q_lo': 0.9999
'q_hi': 1.0
'y_target': [0, 1.0, 1000, 1.0]
```
it means, that i reduce the amount of considered pixels (0.9999 -> 1233Pxls) and they need to stay below 0xFFF, correct?
If this applies, I will give it a try and play around to see if its too unstable. I want to give that a try, because currently I'm using a list of 4 shutterspeeds (SS). Because of the SS-change takes 0.9s, one of my image-sets takes up to 4..5 seconds, depending on the actual image resolution. Therefore, an autoadjustment of the SS is a huge timesave. I tested this already using a defined led-stripe as a variable light source and it worked perfectly. Thereby, the image capturing took only 0.2 .. 0.4 seconds.

The problem only occurs, when I have my samples under test. Because I want to use the brightness-signal as a measure of my samples performance, which is why not one pixel should be in saturation. Below I've added an image of how my images looks like, and what I try to evaluate then using opencv:
[image]

Kind regards,
Deph

davidplowman, 2023-07-28:
Hi, sounds like an interesting application, though I see quite a few challenges here.

One thing to note is that the histograms have only 128 bins, so the resolution there is pretty limited. 2%, therefore, corresponds approximately to the top 2-and-a-bit bins, but 0.01% is about 1/100th of a bin, so it doesn't really mean what you want. I think it will apply the constraint "pro rata" to the top bin, so it will probably test if "the top bin has more than ~123k pixels".

Setting the target y to 1.0 is also problematic, in fact I think the code will just change it to a lower value. The problem is that monitoring the top end of a histogram is really hard because once pixels saturate, you have no clue how much to reduce the exposure. Are they massively over-saturated? Or just a teeny bit? The algorithm really doesn't like that.

Can you say anything more about the exposure times you want to use, how you determine them and then how you do the captures? Generally it is possible to pipeline exposure changes so as to get sequences of frames with different exposures quite quickly.

Dephrilibrium, 2023-07-28:
Hi david,

the saturation can heavily differ. Therefore I can't assume anything regarding the exposure time.

Currently I'm avoiding overexposure by combining the brightness information of a spot on different SS-images. But as I need a lot more images, that's on cost of the capture-time. Because I'm measuring the current and voltage of the my sample additionally, the time-correlation between electrical values and brightness-values gets lost more and more.
For instance:
```
t=0.0: Request electrical data and images - 4xSS x 3xImgs per SS = 12 Imgs
t=1.0: Received electrical data
       Captured SS =   3150 done         # delta t between   electrical <-> images   = 0s
t=2.0: Captured SS =  10000 done         # delta t between   electrical <-> images   = 1s
t=3.0: Captured SS =  31500 done         # delta t between   electrical <-> images   = 2s
t=4.0: Captured SS = 100000 done         # delta t between   electrical <-> images   = 3s
       Archive image as a single *.tar-file
t=4.3: Download *.tar-file
t=6.0: Next Request (beginning at t=0 again)
...
```
Because of this, the noise  can becomse higher than the acutal measured value, which makes the dataset worthless. That's basically why I give it a try to get rid of the fix SS-list. That comes on the cost of a bit dynamic range, but the time save is just huge and worth the try to implement this feature in my code.

Kind regards,
Deph

battlesnake, 2025-04-03:
@davidplowman  Thanks for explaining how the constraints work.

Does this seem reasonable if I want to avoid having too many undersaturated(black) or oversaturated pixels?

```
        "constraint_modes": {
          "normal": [
            {
              "bound": "UPPER",
              "q_lo": 0.9,
              "q_hi": 0.95,
              "y_target": [0, 0.8, 10000, 0.8]
            },
            {
              "bound": "LOWER",
              "q_lo": 0.05,
              "q_hi": 0.1,
              "y_target": [0, 0.02, 10000, 0.02]
            }
          ]
        },
```

davidplowman, 2025-04-04:
Yes, that looks reasonable. The 90-95% inter-quantile mean will be limited to 0.8 of the full range (remember that's before gamma, so that is pretty bright), and the 5-10% mean can't go below 0.02 of the range. Though remember it won't be able to achieve that lower bound in very dark scenes if the exposure/gain max out.