raspberrypi/picamera2 #766: [BUG] picam2.capture_array("lores") completely bricks any PyQt5 app its added to.

DAlexSanc, 2023-07-30:
**Please only report one bug per issue!**

**Describe the bug**
picam2.capture_array("lores") completely bricks any PyQt5 app its added to. 
It works fine with QtGl previews, but any time i attempt to add it to any PyQt5 app, including any of the official picam2 apps, it completely crashes the camera. 

**To Reproduce**
`#!/usr/bin/python3

import numpy as np
from PyQt5 import QtCore
from PyQt5.QtWidgets import (QApplication, QCheckBox, QHBoxLayout, QLabel,
                             QPushButton, QVBoxLayout, QWidget)

from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2

def request_callback(request):
    label.setText(''.join(f"{k}: {v}\n" for k, v in request.get_metadata().items()))

def cleanup():
    picam2.post_callback = None

lsize = (320, 180)
picam2 = Picamera2()
picam2.post_callback = request_callback
preview_config = picam2.create_preview_configuration(main = {"size": (1920, 1080), "format": "XRGB8888"}, lores={"size": (lsize), "format": "YUV420"}, raw=picam2.sensor_modes[2], buffer_count=2)
picam2.configure(preview_config)

app = QApplication([])

def on_button_clicked():
    button.setEnabled(False)
    cfg = picam2.create_still_configuration(main={"size": (1920,1080)}, raw=picam2.sensor_modes[2])
    picam2.switch_mode_and_capture_file(cfg, "test.jpg", signal_function=qpicamera2.signal_done)

def capture_done(job):
    picam2.wait(job)
    button.setEnabled(True)

def on_checkbox_toggled(checked):
    if checked:
        array = picam2.capture_array("lores") ##The moment you toggle this on it dies 
        picam2.set_overlay(None)  
    else:
        qpicamera2.set_overlay(None)

qpicamera2 = QGlPicamera2(picam2, width=800, height=600)
qpicamera2.done_signal.connect(capture_done)

button = QPushButton("Click to capture JPEG")
button.clicked.connect(on_button_clicked)
label = QLabel()
checkbox = QCheckBox("Set Overlay", checked=False)
checkbox.toggled.connect(on_checkbox_toggled)
window = QWidget()
window.setWindowTitle("Qt Picamera2 App")

label.setFixedWidth(400)
label.setAlignment(QtCore.Qt.AlignTop)
layout_h = QHBoxLayout()
layout_v = QVBoxLayout()
layout_v.addWidget(label)
layout_v.addWidget(checkbox)
layout_v.addWidget(button)
layout_h.addWidget(qpicamera2, 80)
layout_h.addLayout(layout_v, 20)
window.resize(1200, 600)
window.setLayout(layout_h)

picam2.start()
window.show()
app.exec()
picam2.stop()`

**Expected behaviour**
In an ideal setting, regardless of where i declare the array it should not freeze the app and image, perhaps lose a few frames, but not crash the app.
Or at least i think it should.  

**Console Output, Screenshots**
This produces no errors, no screens, no nothing, it literally only stops working/freezes up, doesn't even close the application or crash the system, just app. 

**Hardware :**
Raspberry pi 4 8gb, Hq and basic OV5647 5MP cam. 

**Additional context**
This may well be user error, but id like to think ive gotten good enough at coding to be past that, if it is user error, please kindly explain what i did wrong, i only care about getting a solution. 

This popped up while i was working on a personal project where i wanted to use the overlay feature to make filters with the use of opencv, and it works perfectly with just the simple QtGL preview, but the moment i tried to add it to a PyQt5 app to make it look nice, i lost all functionality, i tested it on the picam2 apps and same effect, regardless of where i declare the capture array using lores, and regardless of resolution i give the lores.

I am okay with losing frames,  but i did expect this to work, so is it user error, or a picam2 bug? hopefully it wont be anything worse like a dependency error or internal clash with PyQt.

DAlexSanc, 2023-07-31:
Hi, im sorry about being casual, you are correct, the application locks up and stops responding, and i think that your assessment is correct,  by implementing `picamera.capture_array("lores", signal_function=qpicamera2.signal_done)`  
it appears to regain control again, it will now require a lot of engineering to make it work, but i think its possible, i have one last follow up question;

- Is it possible to implement different done signals? To have one for stills, and one for the array, if not i think i can just forget about the still done signal, its not like those block the apps. 

Thanks you very much for the answers!!

davidplowman, 2023-08-01:
I expect there's no reason you can't put in some extra signals and use them. Though I think it would basically be equivalent to having a "state" variable that you set just before the asynchronous call, and switch on its value when it returns. So go with whatever seems to work best for you, I think!

AngelMariscal1, 2023-08-08:
Hi! I have a similar error. When 'job' is interpreted the gotten result is False. i dont know if the method 'capture_array' its been used correctly or if there's another solution to get the array.

	def process_images(self, image_path=None):
		self.picam2.configure(self.picam2.create_preview_configuration())
		self.qpicamera2 = QPicamera2(self.picam2, width=800, height=600, keep_ar=False)
		
		self.qpicamera2.setWindowTitle("Qt Picamera2 App")
		self.picam2.start()
		
		if image_path:
			base = cv2.imread(image_path)
			self.base_ppd = self.base_preparing(base)
			self.bandera = True  # Set the flag to True to skip camera capture

		while self.bandera == True:
			array = self.picam2.capture_array("lores", signal_function=self.qpicamera2.signal_done)
			job = array.execute()	
			if job:
				grey = array[:self.lsize[1], :self.lsize[0]]
				base_Y = self.base_ppd[:,:,0]
				mask, score_F = self.SSIM(base_Y, grey)
				self.picam2.set_overlay(mask)
				self.selectImage.update_score(score_F)
			else: 
				print("Doesn't work")
				break

hope you can help me, thank you for the time.

davidplowman, 2023-08-08:
Hi, thanks for the question. It's not completely clear to me what it is that you're trying to do, however, the usual pattern is to connect a function of yours to the qpicamera2's "done_signal", and that's then where control is returned to you after calling capture_array. Maybe check out the final code example in section 8.5 of [the manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) for an example.

Does that help? If not, perhaps you could explain a bit more what it is you want to do. In this case, a really small and minimal example demonstrating the problem, and which can be run on its own, is often very helpful. Thanks!

AngelMariscal1, 2023-08-08:
Thank you so much for the response. This is the simplify version of the code: 

from PyQt5 import QtCore
from PyQt5.QtWidgets import (QApplication, QHBoxLayout, QLabel, QPushButton,
                             QVBoxLayout, QWidget)

from picamera2 import Picamera2
from picamera2.previews.qt import QGlPicamera2
lsize = (640, 360)

picam2 = Picamera2()
preview_config = picam2.create_preview_configuration(main = {"size": (1920, 1080), "format": "XRGB8888"}, lores={"size": (lsize), "format": "YUV420"}, raw=picam2.sensor_modes[2])
app = QApplication([])

def on_button_clicked():
    button.setEnabled(False)
    cfg = picam2.create_still_configuration(main={"size": (1920, 1080)})
    picam2.switch_mode_and_capture_file(cfg, "test.jpg", signal_function=qpicamera2.signal_done)

def capture_done(job):
    picam2.wait(job)
    button.setEnabled(True)
    
def signal_done(job):
    picam2.wait(job)
    button.setEnabled(True)

qpicamera2 = QGlPicamera2(picam2, width=1920, height=1080, keep_ar=False)
window = QWidget()
qpicamera2.done_signal.connect(capture_done)
array = picam2.capture_array("lores", signal_function=signal_done)
grey = array[:lsize[1] ,:lsize[0]]

layout_h = QHBoxLayout()
layout_v = QVBoxLayout()
layout_h.addWidget(qpicamera2, 100)
#layout_h.addLayout(layout_v, 20)
window.setWindowTitle("Qt Picamera2 App")
window.resize(1920, 1080)
window.setLayout(layout_h)
button = QPushButton("Click to capture JPEG", parent = window) 
button.clicked.connect(on_button_clicked) 
button.setFixedSize(100,40)   
button.setGeometry(12, 12, 10, 10) 

picam2.start()
window.show()
app.exec()

The problem is when running the code it shows this error: TypeError: 'Job' object is not subscriptable
and I need to use 'capture_array' in real time.

davidplowman, 2023-08-09:
Hi, it would be useful if you could post the code in a code block, that is, with triple backquotes on the lines just before and jsut after:
```
like this
```
That should preserve the formatting including the indentation, and the folks can just copy and paste the code to try it. Nonetheless, even without trying it I'll make a few comments on what I can see:

> Thank you so much for the response. This is the simplify version of the code:
>
> from PyQt5 import QtCore from PyQt5.QtWidgets import (QApplication, QHBoxLayout, QLabel, QPushButton, QVBoxLayout, [...]

`picam2.wait(job)` gives you the delayed result of the asynchronous call that you made. So if you called `capture_array` then you should use `array = picam2.wait(job)`.

> 
> def signal_done(job): picam2.wait(job) button.setEnabled(True)

I don't think this gets used?

> 
> qpicamera2 = QGlPicamera2(picam2, width=1920, height=1080, keep_ar=False) window = QWidget() qpicamera2.done_signal.connect(capture_done) array = picam2.capture_array("lores", signal_function=signal_done) grey = array[:lsize[1] ,:lsize[0]]

The call to `capture_array` here will give you back the job you need to wait for, not the array. In fact, you haven't started the camera yet so if you try to wait for the job it will probably deadlock or something. Basically, you can't capture an image here because you need to start the Qt event loop first.

> layout_h = QHBoxLayout() layout_v = QVBoxLayout() layout_h.addWidget(qpicamera2, 100) #layout_h.addLayout(layout_v, 20) window.setWindowTitle("Qt Picamera2 App") window.resize(1920, 1080) [...]

popcornmix, 2023-08-09:
I've edited the post above to add the code quotes