raspberrypi/picamera2 #767: [BUG] Calling switch_mode_and_capture_file on lores encoder causes an Exception

azsde, 2023-08-02:
Hello,

I have encountered the following bug on the latest release of picamera2 (0.3.12).

If I try to take a picture while recording a video, if the encoder has been started on the lores stream it causes the following crash:

```
Exception during process_requests()
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/previews/null_preview.py", line 85, in handle_request
    picam2.process_requests(self)
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/picamera2.py", line 1151, in process_requests
    encoder.encode(self.stream_map[encoder.name], req)
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/encoders/encoder.py", line 219, in encode
    self._encode(stream, request)
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/encoders/v4l2_encoder.py", line 233, in _encode
    fb = request.request.buffers[stream]
KeyError: <libcamera._libcamera.Stream object at 0x7fad8d6370>
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/previews/null_preview.py", line 29, in thread_func
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/previews/null_preview.py", line 85, in handle_request
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/picamera2.py", line 1151, in process_requests
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/encoders/encoder.py", line 219, in encode
  File "/usr/local/lib/python3.9/dist-packages/picamera2-0.3.12-py3.9.egg/picamera2/encoders/v4l2_encoder.py", line 233, in _encode
KeyError: <libcamera._libcamera.Stream object at 0x7fad8d6370>
```

Here is the full code that caused this error:

```
#!/usr/bin/python3
from time import sleep

from picamera2 import Picamera2
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import FileOutput

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(main={"size": (1920, 1080)}, lores={"size": (740, 480)}))
encoder = MJPEGEncoder()
encoder.output = [FileOutput("test.mp4")]
# Here, if I start the encoder using : picam2.start_encoder(encoder) instead there is not any crash
picam2.start_encoder(encoder, name="lores") 
still_capture_config = picam2.create_still_configuration()
picam2.start()

try:
    print("Wait 5 s")
    sleep(5)
    print("Done waiting, capturing image")
    picam2.switch_mode_and_capture_file(still_capture_config, "test_full.jpg")
    sleep(5)
finally:
    picam2.stop_encoder()
    picam2.stop()
```

As explained in the code, the crash only happens if the encoder has been started on the lores stream.
If i start the encoder without specifying the lores stream, or if I start it by specifying the main stream, I get not error and everything works as expected.

davidplowman, 2023-08-02:
Hi, thanks for the report - this came up on the camera forum just recently. I expected that the system wouldn't let you reconfigure the camera (which mode switching does) when there's an encoder running, but apparently it doesn't check. However, it should because the memory buffers that the encoder may be using are going to get deallocated. I'll put this on the list of things to do!

azsde, 2023-08-02:
So the right approach would be to stop the encoder, capture the still image using **switch_mode_and_capture_file**, and then restarting the encoder ?

davidplowman, 2023-08-02:
Yes. I think you would have to create a new encoder. If you're using a simple FileOutput, I think you can open the output file yourself, at which point stopping the encoder won't close it, and you can reuse it when when start a new encoder afterwards (saving you from merging multiple files).

azsde, 2023-08-03:
For the time being I do:

```
picam2.stop_encoder(mjpegEncoder)
picam2.switch_mode_and_capture_file(still_capture_config, still_image_filename)
picam2.start_encoder(mjpegEncoder, name="lores")
```

And it seems to work as expected since I have a FileOutput set to the mjpegEncoder.

That being said, I don't fully understand why this is not required when using the main stream ?