raspberrypi/picamera2 #770: [BUG] Timestamps are not reset when doing two successive recordings using H264Encoder

azsde, 2023-08-07:
Hello,

I am using the following code:

```
#!/usr/bin/env python
from picamera2 import Picamera2

from picamera2.encoders import H264Encoder

from time import sleep

picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={"size": (1920, 1080)})
picam2.configure(video_config)
h264encoder = H264Encoder()

picam2.start()
picam2.start_recording(h264encoder, "fileA.h264", pts="fileATimestamps.txt")
sleep(5)
picam2.stop_recording()
sleep(2)
picam2.start_recording(h264encoder, "fileB.h264", pts="fileBTimestamps.txt")
sleep(5)
picam2.stop_recording()
```

The fileBTimestamps.txt are not starting at "0":

```
azsde@azsdepi:~/RCFI $ cat fileBimestamps.txt 
6941.977
6975.298
7008.620
...
```
Is it a bug ? Or am I missing something that is required to "reset" the timestamps ?

davidplowman, 2023-08-07:
Thanks for spotting that. I'm not really sure whether that's intentional, or just wasn't thought about, but I can't really think of a good reason why you'd want the existing behaviour so probably the latter then. It's a trivial enough thing to fix, but in the meantime you can presumably just create a new encoder for the second recording.

azsde, 2023-08-07:
Alright, I'll implement a workaround in my app in the meantime.

FYI, I'm working on creating an alternative to `https://elinux.org/RPi-Cam-Web-Interface` that uses picamera2.

That is why I have identified some edge use cases.

davidplowman, 2023-08-07:
Ah, cool, that sounds like a useful (and quite large!!) project, would definitely be good to upgrade some of the old legacy apps out there!